arendst/Tasmota #10283: Reboot with Exception after decreasing Heap

ChristianB78, 2020-12-28:
### PROBLEM DESCRIPTION
The Sonoff Basic Device will reboot after 22 minutes, the Heap is decreasing from 26 to 0, the device gets an exception and reboots. The Problem occours on all my 20 devices in my Network (Sonoff Basic, Sonoff 4ch Pro, Shelly 2.5).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff Basic
- [ ] Tasmota binary firmware version number used: 9.2.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Tasmotizer
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
07:55:23 SRC: WebConsole from 192.168.6.22
07:55:23 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
07:55:24 SRC: Backlog
07:55:24 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
07:55:24 MQT: stat/nachttisch/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:55:24 SRC: Backlog
07:55:24 CMD: Group 0, Index 1, Command "MODULE", Data ""
07:55:24 MQT: stat/nachttisch/RESULT = {"Module":{"1":"Sonoff Basic"}}
07:55:24 SRC: Backlog
07:55:24 CMD: Group 0, Index 1, Command "GPIO", Data "255"
07:55:24 MQT: stat/nachttisch/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
07:55:36 WIF: Checking connection...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
07:56:38 CMD: Status0
07:56:38 SRC: WebConsole from 192.168.6.22
07:56:38 CMD: Group 0, Index 0, Command "STATUS", Data ""
07:56:38 MQT: stat/nachttisch/STATUS = {"Status":{"Module":1,"DeviceName":"nachttisch","FriendlyName":["nachttisch"],"Topic":"nachttisch","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

```
- [x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00 CFG: Loaded from flash at F6, Count 70
00:00:00 QPC: Count 1
00:00:00 CFG: CR 412/699, Busy 0
00:00:00 SRC: Restart
00:00:00 Project tasmota nachttisch Version 9.2.0(tasmota)-2_7_4_9(2020-12-21T15:03:40)
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 E.I.N.S.2GHz Channel 11 BSSId 44:4E:6D:DA:39:1D in mode 11n as nachttisch-6142...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Connected
00:00:03 HTP: Web server active on nachttisch-6142 with IP address 192.168.6.218
00:00:03 HTP: Main Menu
07:19:38 RTC: UTC 2020-12-28T06:19:38, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
07:19:39 MQT: Attempting connection...
07:19:39 MQT: Connected
07:19:39 MQT: tele/nachttisch/LWT = Online (retained)
07:19:39 MQT: cmnd/nachttisch/POWER = 
07:19:39 MQT: Subscribe to cmnd/nachttisch/#
[... 128 lines cut ...]
07:42:22 MQT: tele/nachttisch/STATE = {"Time":"2020-12-28T07:42:22","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"E.I.N.S.2GHz","BSSId":"44:4E:6D:DA:39:1D","Channel":11,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:03"}}
07:42:23 APP: Boot Count 49
07:42:23 CFG: Saved to flash at F4, Count 72, Bytes 4096
07:42:27 MQT: tasmota/discovery/68C63ACDB7FE/config = {"ip":"192.168.6.218","dn":"nachttisch","fn":["nachttisch",null,null,null,null,null,null,null],"hn":"nachttisch-6142","mac":"68C63ACDB7FE","md":"Sonoff Basic","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.2.0","t":"nachttisch","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
07:42:27 MQT: tasmota/discovery/68C63ACDB7FE/sensors = {"sn":{"Time":"2020-12-28T07:42:27"},"ver":1} (retained)
07:42:36 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
There is nothing to be done, it happens by itsself.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The Heap should not decrease and the device should not reboot by itsself.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-12-28:
Pls provide output of command ``status 0``. Pay attention to the space between status and 0

ChristianB78, 2020-12-28:
OK, sorry for my mistake. I added the output of "status 0" above.

arendst, 2020-12-28:
Considering the fact it happens on 20 devices AND apparently only in your case I suspect an external wifi issue.

Can you make a config backup and send it to me via pm. See github for the address.

arendst, 2020-12-28:
I suppose you do not use Prometheus as the 9.2.0 version contains a memory leak.

ChristianB78, 2020-12-28:
Should I Update to version 9.2.1? I tried it yesterday on a device with 9.2.1 and the problem also occurred, bit I can do it again.

arendst, 2020-12-28:
Well it doesn't hurt to try the latest development version on one of the devices. Let the others stay on 9.2.0 pls until I have done some investigation of your config.

arendst, 2020-12-28:
Loaded your config on a wemos and it still works as expected (configured my router with your SSID/Password). 

From your gateway MAC address I see you are using an AVM Fritzbox. They are great fun....

I fear the Tasmota used Arduino Core/Espressif SDK TCP/IP stack is leading to anomalies with your Fritzbox. I cannot solve it here but you might need to configure your FritzBox differently. 

Perhaps someone more knowledgeable about the FritzBox config can chime in.

ChristianB78, 2020-12-28:
I'll sniff a little in the network, maybe I'll find something unusual. My Fritz Box configuration is not special, but I will now take a closer look at it. Thank you very much!

arendst, 2020-12-28:
Did you also see this on Tasmota version 9.1.0?

If not sure you might want to ota load one of your devices from http://ota.tasmota.com/tasmota/release-9.1.0/tasmota.bin.gz

There was a connection timeout change between 9.1.0 and 9.2.0

ChristianB78, 2020-12-28:
I have installed Version 9.1.0, the issue is the same.

I will try one of my Sonoff Devices in the Network of my colleague tomorow, he has the same Fritzbox and the same Internet Provider.  

I can see some Broadcast Traffic in Wireshark, is tasmota sensitive to broadcast Traffic?

arendst, 2020-12-28:
Yes if you have any emulation (Wemo or Hue) selected.

ChristianB78, 2020-12-28:
I think the problem is recognized. I have a LAN to LAN VPN connection, and Tasmota has been running stable for about 90 Minutes since I deactivated this connection. I will double check it and I will try to find out more details about the issue.

Jason2866, 2020-12-28:
Interesting, if you find which packets (traffic) is causing the problem please post your findings

stefanbode, 2020-12-29:
@ChristianB78 : I also have quite a lot of ESP8288 and ESp32 running on my fritzbox. I can cross-check if this helps. Let me guess "LAN to LAN VPN" you're using to connecting 2 Fritzbox networks?

ChristianB78, 2020-12-29:
I have connected 2 Networks "Lan to Lan" between two Fritzboxes, but disabled at the moment.

Configuration details:
- PSK (Preshared Key)
- NetBIOS oder VPN is allowed (this could be the Problem because of many Broadcasts)
- Keep connected

I will make two traces in my Network, one time with VPN and one time without established VPN connection, to compare the Network behavior. I think I will do it on the coming long weekend.

ChristianB78, 2020-12-29:
What exactly does tasmota do at this Moment "07:40:57 WIF: Checking connection..."?

arendst, 2020-12-29:
It calls SDK function ``wifi_station_get_connect_status`` and checks if there is a valid IP address.

Jason2866, 2020-12-29:
You can try this Tasmota version. It is builded from latest Tasmota development with Arduino ESP8266 core stage and latest gcc build chain
[tasmota.bin.gz](https://github.com/arendst/Tasmota/files/5751191/tasmota.bin.gz)

ascillato2, 2021-01-03:
Hi, any news on this?

Leuselator, 2021-01-03:
Only the one that I can see it on my devices to. No Fritzbox involved - ubiquity-stuff here.
Tried 9.2.0.0 downto 8.5.0.1 - all the same. I'm not sure if it has to do with ubiquiti updates but devices ar nonresponsive, rebootin and so on. Maybe the Multi-/Broadcast thing? Somewhat idealess @  the moment...

ChristianB78, 2021-01-04:
On the nice long weekend, so on 31.12. the problem occurred again after a single reboot, so I did a lot, disabled guest WLAN, Openhab server disabled, WPA3 enabled and disabled again. So the cause of the problem was not the LAN 2 LAN VPN alone, as it was still disabled. At some point the Tasmotas ran stable again for a few days. Today I had to enable the Lan to Lan connection again, immediately the problem was back. So tonight I tried things again, many reboots, wifi disabled and reboot and much more. In a trace I noticed today that extremely much NTP traffic comes from the Tasmotas. I have the default NTP server set up, so pool.ntp.org, but it seems like the Fritzbox is accepting the requests. Three quarters of the WLAN traffic is NTP traffic, so I disabled the NTP server on the Fritzbox. After a reboot the Tasmotas are stable again, but I won't do a reboot until tomorrow at the earliest, otherwise I won't be able to get off the computer today.

[image]

Jason2866, 2021-01-05:
@ChristianB78  Mhh, it seems the NTP Server in the Fritzbox is not working always reliable. You are not the first having trouble with the Fritzbox NTP Server. Imho it is worth a try to run a extra NTP Server in your network and point the Tasmota devices to this NTP Server. A Tasmota device can act as NTP Server. You just need to connect a supported cheap GPS receiver.

arendst, 2021-01-05:
In addition I'll have a look at my ntp code if there is something that could cause a memory leak.....

arendst, 2021-01-05:
I see no cause for memory leak in the NTP code.

On restart Tasmota will try to sync with NTP within the first minute every second. After that it re-syncs once a minute until it syncs. Then it re-syncs once every hour.

So indeed you can see a lot of NTP traffic if no response from a NTP server is received.

ascillato, 2021-01-05:
If the NTP response is incorrect or incomplete, the Arduino Core code is the one that produces the exception. The problem is not in Tasmota but rather in the Arduino core.

I found this when all my devices starts rebooting because my crappy ISP don't know how to set time in its fiber concentrators (yes, it sound very silly but that is what they told me and what I have verified). When my devices tried to sync to pool.ntp.org or any internet ntp server, the response failed because the time of one of the jumps between the NTP server and my devices had an incorrect time. As soon as I set the time manually, my devices stop rebooting. Then I add a GPS to one of my Tasmota devices and point all my devices and cameras to it and have been working flawlessly since then. Zero reboots. 
The time problem I saw not only in Tasmotas, but also in cameras and PCs. After setting my own NTP server I haven't had any restarts nor time issues.

ChristianB78, 2021-01-05:
One of my Tasmota Devices is running for 16h now. The timesync was successfull at boot of the device.

[image]

I can see the following ntp traffic in my trace:
[image]

Time synchronization is not important to me, so in doubt I will set the time manually. What surprises me is that the trace says that the NTP requests go to 192.168.6.1, which is my Fritzbox. But as NTP Server is a completely different host entered.

The problem with the restarts is solved since I switched off the ntp server of the Fritzbox, but I will continue to try if it is still stable after a few restarts of the Fritzbox. Then I will activate my Lan to Lan VPN again and see if it remains stable. Here there are me too many strange effects, I do not trust the thing yet.

ascillato, 2021-01-05:
> What surprises me is that the trace says that the NTP requests go to 192.168.6.1, which is my Fritzbox

Yes, when you use DHCP, your NTP settings are ignored and the Arduino core will use the NTP provided by your DHCP server (if the DHCP server have the NTP enabled)

The NTP settings that you put in Tasmota are used only when you set a fixed IP in Tasmota. (Arduino core limitation)

ascillato, 2021-01-05:
So, this is the main issue with Fritzbox. It responds with an unsynced NTP on DHCP requests and the Arduino core produces an exception on that.

ChristianB78, 2021-01-05:
You are right, after I disabled the NTP server of the Fritzbox, the DHCP option was also changed from the local address of the Fritzbox to the NTP server of my provider, I have seen it in a new trace.  After a reboot of the Tasmota device it also has a new lease incl. the external NTP server. I will reboot all my Tasmota devices now. :-)

Leuselator, 2021-01-05:
I'm facing this issue as well - some devices such as Sonof Basic run into the exception every 10-40s others need a bit longer (GoSond 111 and WeMo D1 Mini)
I have a network exclusively run by Ubiquity-Components (USG-Pro, APAC nano HD, APAC Pro, several Switches). I've deactivated DHCP NTP-Option for the network. Set the ipaddress manualy on the tasmota-devices as well as the ntp-servers1-3 (lGateway, QNAP, PTBTIME3.PTB.DE - all as IP-Address not FQDN)
Tryed to RESET 5 and again to set IP and NTPservers. GoSunds do accept, Sonof-Basic not...
stepwise downgraded back to Version 7.1.0.6 - still get
00:16:09 MQT: stat/tasmota_14CFFE/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4023aa64","00000000","00000000"],"EXCVADDR":"00000004","DEPC":"00000000","CallChain":["4023aa60","4023aaec","4025591e","40101b32","40259fb1","40101afb","4026bc98","40253045","40252e6b","40276152","40276163","4026b8c3","4027196b","40000f49","40000f49","40000e19","401059bd","401059c3","4010000d","4026a7a8","4026a759","401046bd","40105805","40105805","401053ea","40104d39","401047af","401047ac","4010370f","401046bd","401055c5"]}}
Any suggestions?

s-hadinger, 2021-01-05:
Do you have the map file for your firmware? Which version is it?

ChristianB78, 2021-01-05:
For me personally the problem seems to be solved since the NTP server of the FritzBox is switched off and the NTP option points to the public NTP server of my provider. Also my LAN to LAN VPN connection is active again and everything runs stable. Also four reboots of the router did not cause any problems. Nevertheless, I continue to keep an eye on the matter. 

Many thanks to all who supported me here, you are a great team!

ChristianB78, 2021-01-05:
@Leuselator 
The difference with my problem is not only the network components, but also the amount of time before the problem occurs, with me it was always about 22 minutes.

Can you recreate the problem on another network? Simply put a device in the neighbor's network?

If the problem does not occur there, sniff a round with Wireshark in your own network and look for abnormalities. 

I love the exclusion method to isolate problems.

Leuselator, 2021-01-06:
Upgraded to 9.2.0.3 - did not help. 
Then I've deactivated the option "DHCP Snooping" for the site in the Unificontroller - looks as if this caused the most of this exceptions. 
I still see it on OBI-Plugs and Sonoff-Basics - by far less often then before, on GoSund111 and WeMo D1 Minis no more so far.
Wirdo in a way...

ascillato2, 2021-01-20:
> For me personally the problem seems to be solved since the NTP server of the FritzBox is switched off and the NTP option points to the public NTP server of my provider. Also my LAN to LAN VPN [...]

> Many thanks to all who supported me here, you are a great team!

Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-20 by ascillato2