arendst/Tasmota #10296: Build fails for target "tasmota32" when feature "#define USE_MQTT_TLS" is activated

EliasGabrielsson, 2020-12-28:
### PROBLEM DESCRIPTION
Build fails for target "tasmota32" when feature [#define USE_MQTT_TLS](https://github.com/arendst/Tasmota/blob/6058cf014ef57d5fdd53c7c383cbc3684e0630a5/tasmota/my_user_config.h#L395) is activated.

Build error:
```
In file included from /home/elias/git/Tasmota/tasmota/xdrv_02_mqtt.ino:29:0:
tasmota/WiFiClientSecureLightBearSSL.h:46:12: error: 'size_t BearSSL::WiFiClientSecure_light::write_P(const char*, size_t)' marked 'override', but does not override
     size_t write_P(PGM_P buf, size_t size) override;
            ^
tasmota/WiFiClientSecureLightBearSSL.h:58:12: error: 'size_t BearSSL::WiFiClientSecure_light::peekBytes(uint8_t*, size_t)' marked 'override', but does not override
     size_t peekBytes(uint8_t *buffer, size_t length) override;
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): NOT APPLICABLE
- [ ] Tasmota binary firmware version number used: NOT APPLICABLE
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: NOT APPLICABLE
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
1. Checkout clean `development` branch
2. Verify working buildchain by building the target: "env:tasmota32"
3. Enable [#define USE_MQTT_TLS](https://github.com/arendst/Tasmota/blob/6058cf014ef57d5fdd53c7c383cbc3684e0630a5/tasmota/my_user_config.h#L395)
4. Build the target  "env:tasmota32"

### EXPECTED BEHAVIOUR
A compiled binary.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-12-28:
For the ESP32 beta, TLS is not supported currently.

stefanbode, 2020-12-29:
Yes, I'm also waiting for this one to make a wider rollout of the ESP32. Currently no workaround possible. It just do not work with the bearssl that the pubsubclient relies on. No workaround possible currently

ortegafernando, 2020-12-29:
Hi, does this help? https://github.com/256dpi/arduino-mqtt, it is also compatible with ESP32

s-hadinger, 2020-12-29:
I need to find time. There is no fundamental blocker for using BearSSL with ESP32. But the Arduino SDK is different so it needs adaptation.

closed 2020-12-29 by ascillato2