arendst/Tasmota #10298: Getting IP loops while Wifi is available but no router is behind

scheilch, 2020-12-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The Device seems to loop when Wifi is available and the device is connected but cannot get an IP from a router. This happens when a Wifi Repeater is used with a LAN bridge and the cable is broken or something like that. 
The only issue is when getting an IP from the connected Wifi the device doesn't do anything else. There are rules triggered by a sensor but nothing is triggered when the device wants to get an IP.

When the repeater is reconnected to the router everything works fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____ NodeMCU v3
- [ ] Tasmota binary firmware version number used: _____ latest 9.2.0
  - [x ] Pre-compiled
  - [ x] Self-compiled
- [x ] Flashing tools used: _____ VScode with platform.io and Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-29:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

scheilch, 2020-12-29:
I will do, but I have disassmble the whole device to get it back to my computer and let it read the serial log.

ascillato2, 2021-01-03:
Please, when you have time, complete the troubleshooting template. Thanks.

closed 2021-01-03 by ascillato2