arendst/Tasmota #10302: Latency increased at switch

samuelalexmclean, 2020-12-29:
### PROBLEM DESCRIPTION
I've recently added Tasmota to [this  SRL switch](https://templates.blakadder.com/srl_3-4WW.html). Everything is working nicely with my Home Assistant server, but I've noticed the latency at the switch has increased (perceptibly, I would estimate >100ms difference). Previously I would brush against any (or multiple) of the capacitive switch points, and the LED would instantly illuminate, the relay would instantly click and the light would instantly come on (or close enough to instant that I couldn't see any delay). Now there's a solid delay and (since there's no tactile feedback on the switch) it makes things feel much jankier than you'd expect.

Is this working as intended? I get that MQTT and stuff might require more time, but a) I'd have expected that to happen after toggling things locally, and b) I'd have expected local toggling to be ~instananeous, since it's such a low level operation. Any tips? Thanks for a great bit of software!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used: https://templates.blakadder.com/srl_3-4WW.html
- [x] Tasmota binary firmware version number used: 9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tuya-convert followed by OTA upgrade
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:37:25 CMD: Backlog Template; Module; GPIO 255
10:37:25 MQT: foyer/switch/frontdoor/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
10:37:25 MQT: foyer/switch/frontdoor/stat/RESULT = {"Module":{"18":"Generic"}}
10:37:25 MQT: foyer/switch/frontdoor/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"34":"Button3"},"GPIO4":{"226":"Relay3"},"GPIO5":{"33":"Button2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"32":"Button1"},"GPIO13":{"224":"Relay1"},"GPIO14":{"227":"Relay4"},"GPIO15":{"225":"Relay2"},"GPIO16":{"35":"Button4"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

10:38:04 CMD: Status 0
10:38:04 MQT: foyer/switch/frontdoor/stat/STATUS = {"Status":{"Module":18,"DeviceName":"Foyer Light","FriendlyName":["Foyer Light","Front Door Light","Driveway Light","Garbage Bin Light"],"Topic":"foyer/switch/frontdoor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:38:04 MQT: foyer/switch/frontdoor/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:18","StartupUTC":"2020-12-29T09:34:46","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2020-12-29T10:34:53","SaveCount":59,"SaveAddress":"F4000"}}
10:38:04 MQT: foyer/switch/frontdoor/stat/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"462/699"}}
10:38:04 MQT: foyer/switch/frontdoor/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Shenandoah",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
10:38:04 MQT: foyer/switch/frontdoor/stat/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":0,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
10:38:04 MQT: foyer/switch/frontdoor/stat/STATUS5 = {"StatusNET":{"Hostname":"Foyer-Switch-FrontDoor","IPAddress":"192.168.0.127","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"B4:E6:2D:16:08:DF","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
10:38:04 MQT: foyer/switch/frontdoor/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1608DF","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:38:04 MQT: foyer/switch/frontdoor/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-12-29T09:38:04","Local":"2020-12-29T10:38:04","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:00"}}
10:38:04 MQT: foyer/switch/frontdoor/stat/STATUS10 = {"StatusSNS":{"Time":"2020-12-29T10:38:04"}}
10:38:04 MQT: foyer/switch/frontdoor/stat/STATUS11 = {"StatusSTS":{"Time":"2020-12-29T10:38:04","Uptime":"0T00:03:18","UptimeSec":198,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Shenandoah","BSSId":"F0:72:EA:00:6E:F2","Channel":1,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

N/A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Tap switch

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

LED, relay and light actuate immediately

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-29:
Hi,

You can configure those behavior using the setoptions command. Please, refer to the docs for that.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-29 by ascillato2

samuelalexmclean, 2020-12-29:
Thanks Adrian! For anyone who find this on Google, SetOption13 1 is the one you want - there are a few of them. Probably only a good idea if the switch (like mine) doesn't support dimming.