arendst/Tasmota #10320: [Feature Request] Telegram bypass certificate check

Tigger2014, 2020-12-30:
**Have you looked for this feature in other issues and in the docs?**
Yes, continuation of  #issue8619

**A clear and concise description of what the problem is.**
Telegram support needs TLS and currently certificate must be validated, we should be able to disable this like is possible for MQTT over TLS

**Describe the solution you'd like**
Being able to set the telegram fingerprint to all FF to disable the check like the MQTT option
MqttFingerprint2 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

Thanks

fmarzocca, 2021-01-24:
Could it be useful that Universal Arduino Telegram Bot library has been updated by including a certificate to connect to https://..telegram...?

closed 2021-02-23 by github-actions[bot]

Tigger2014, 2021-03-28:
Reopen, maybe move to discussion?

bam80, 2023-12-11:
Ping on reopen?