arendst/Tasmota #10329: Latest TasmotaDEV breaks Shelly2.5

kugelkopf123, 2020-12-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If you update the latest dev version onto a Shelly2.5 (previous 9.0.0.2), nothing works afterwards.
There is no longer a web console. The screen remains dark. The complete energy metering in the main menu has disappeared. 
As a test, I did an Erase Flash on a Shelly2.5 I still had in the desk and then installed the latest version. The result is the same.
this template was used: 
{"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}

The ADE7953 is detected. 
Did it via Serial.
Normally always use my own user_config_override.h 
At first I thought it was because of this. I therefore deleted the entire user_config_override, erased Shelly2.5 and then installed Tasmota Sensor. The result is the same. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Shelly2.5
- [ ] Tasmota binary firmware version number used:9.2.0.2(sensors)
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: Esptool/VisualStudio Platformio
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```03:36:40.798 CMD: Backlog Template; Module; GPIO 255
03:36:40.854 RSL: stat/tasmota_94F4AA/RESULT = {"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18}
03:36:41.062 RSL: stat/tasmota_94F4AA/RESULT = {"Module":{"0":"Shelly 2.5"}}
03:36:41.287 RSL: stat/tasmota_94F4AA/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"32":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"193":"Switch_n2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"192":"Switch_n1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ"},"GPIO17":{"4736":"ADC Temp"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
03:37:12.308 CMD: Status 0
03:37:12.314 RSL: stat/tasmota_94F4AA/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_94F4AA","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:37:12.341 RSL: stat/tasmota_94F4AA/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:05","StartupUTC":"2020-12-31T02:33:07","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2020-12-31T03:32:46","SaveCount":11,"SaveAddress":"F9000"}}
03:37:12.378 RSL: stat/tasmota_94F4AA/STATUS2 = {"StatusFWR":{"Version":"9.2.0.2(sensors)","BuildDateTime":"2020-12-31T03:31:11","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"394/699"}}
03:37:12.397 RSL: stat/tasmota_94F4AA/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
03:37:12.430 RSL: stat/tasmota_94F4AA/STATUS4 = {"StatusMEM":{"ProgramSize":641,"Free":360,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
03:37:12.472 RSL: stat/tasmota_94F4AA/STATUS5 = {"StatusNET":{"Hostname":"tasmota_94F4AA-5290","IPAddress":"192.168.1.111","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"E0:98:06:94:F4:AA","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
03:37:12.491 RSL: stat/tasmota_94F4AA/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_94F4AA","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
03:37:12.513 RSL: stat/tasmota_94F4AA/STATUS7 = {"StatusTIM":{"UTC":"2020-12-31T02:37:12","Local":"2020-12-31T03:37:12","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:02"}}
03:37:12.532 RSL: stat/tasmota_94F4AA/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
03:37:12.552 RSL: stat/tasmota_94F4AA/STATUS10 = {"StatusSNS":{"Time":"2020-12-31T03:37:12","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":38.3},"ENERGY":{"TotalStartTime":"2020-12-31T03:32:46","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}}
03:37:12.581 RSL: stat/tasmota_94F4AA/STATUS11 = {"StatusSTS":{"Time":"2020-12-31T03:37:12","Uptime":"0T00:04:05","UptimeSec":245,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXX","BSSId":"E0:28:6D:22:3F:BC","Channel":1,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld

00:00:00.049 CFG: Loaded from flash at F7, Count 13
00:00:00.054 QPC: Count 1
00:00:00.056 CFG: CR 394/699, Busy 0
[... 44 lines cut ...]
03:42:21.221 ADE: U 1773, C 0, I 0 + 0 = 0, P 0 + 0 = 0
03:42:21.222 RSL: tasmota/discovery/E0980694F4AA/sensors = {"sn":{"Time":"2020-12-31T03:42:21","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":38.1},"ENERGY":{"TotalStartTime":"2020-12-31T03:32:46","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"},"ver":1}
03:42:22.208 ADE: 1773, 0, [1773, -1, 0, 0], [1773, 0, 0, 0]
03:42:22.209 ADE: U 1773, C 0, I 0 + 0 = 0, P 0 + 0 = 0
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Load latest Dev Version on a Shelly2.5 and set the right Template. (From Blackadder Templates Website)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Show Energymetering in Main Menu and working weblog Webconsole.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2020-12-31:
@kugelkopf123 Can you please tell us which browser and OS you are using ?
Recent changes on the Javascript code may be the cause.
Thanks

barbudor, 2020-12-31:
Adding @assash to the thread

assachs, 2020-12-31:
These screenshots look like my ipad/safari problem (status missing, console empty). With the newest build this problem is gone.

kugelkopf123, 2020-12-31:
> @kugelkopf123 Can you please tell us which browser and OS you are using ?
> Recent changes on the Javascript code may be the cause.
> Thanks

In this screenshots, I used latest Safari (14.0.2) on an iMac with with Macos 10.14.6 Mojave.
Going to test the Version now once again. Will report back in half an hour.

kugelkopf123, 2020-12-31:
I can confirm! Its working now on my iMac, Macbookpro and iphone11. Webconsole and Energy stuff is back! Thanks, good work!!

closed 2020-12-31 by kugelkopf123