arendst/Tasmota #10333: Problem with DTP Change (#9811, #9888)

ferengi82, 2020-12-31:
### PROBLEM DESCRIPTION
Adrian changed the default DTP from 9 to 14 (#9811, #9888), now i have a big problem.
My Heating Actuator (MTD AKH) expects Temperaturereadings as DTP9 and not DTP14.
As temporary fix i changed back 4byte -> 2byte in xdrv_11_knx.ino, but i think this is not a permanet solution (maybe i need some day DTP14)

Is it possible to Change the KNX Part from Tasmota so we can choose in the Webinterface which DTP should be send?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini with SHT3X sensor
- [x] Tasmota binary firmware version number used: 9.2.0.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: flash with integrated webserver (first minimal then knx)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-12-31:
Hi,

Yes, that was an intended breaking change. In the release notes was explained. It was required for expand the use of the KNX to more devices.

I'm refactoring all the KNX driver to use less flash and RAM and also to be able to select the DPTs.

This will take some time. You can use an old version of Tasmota KNX or build with your change until the new driver is finished.

closed 2020-12-31 by ascillato2

ferengi82, 2020-12-31:
thanks for the fast replay.

missed the info that you refactoring the KNX driver.