arendst/Tasmota #10348: Energy Total value gone after I unplug / plug the module

herangstl, 2021-01-01:
### PROBLEM DESCRIPTION
When I power off & and my module (Gosund SP112) the value for "Energy Total" is 0.

### REQUESTED INFORMATION
```
13:53:16 CMD: status 0
13:53:16 MQT: stat/tasmota2/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota2","FriendlyName":["Tasmota2","Tasmota2"],"Topic":"tasmota2","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:53:16 MQT: stat/tasmota2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin","RestartReason":"Power On","Uptime":"0T11:18:57","StartupUTC":"2021-01-01T01:34:19","Sleep":50,"CfgHolder":4617,"BootCount":28,"BCResetTime":"2020-12-22T21:47:46","SaveCount":99,"SaveAddress":"FB000"}}
13:53:16 MQT: stat/tasmota2/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"417/699"}}
13:53:16 MQT: stat/tasmota2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["***********",""],"TelePeriod":20,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
13:53:16 MQT: stat/tasmota2/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
13:53:16 MQT: stat/tasmota2/STATUS5 = {"StatusNET":{"Hostname":"tasmota2-5688","IPAddress":"192.168.178.33","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"50:02:91:14:56:38","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
13:53:16 MQT: stat/tasmota2/STATUS6 = {"StatusMQT":{"MqttHost":"wickie.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_145638","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:53:16 MQT: stat/tasmota2/STATUS7 = {"StatusTIM":{"UTC":"2021-01-01T12:53:16","Local":"2021-01-01T13:53:16","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:03"}}
13:53:16 MQT: stat/tasmota2/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:53:16 MQT: stat/tasmota2/STATUS10 = {"StatusSNS":{"Time":"2021-01-01T13:53:16","ENERGY":{"TotalStartTime":"2020-12-22T21:14:43","Total":0.000,"Yesterday":0.597,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":219,"Current":0.000}}}
13:53:16 MQT: stat/tasmota2/STATUS11 = {"StatusSTS":{"Time":"2021-01-01T13:53:16","Uptime":"0T11:18:57","UptimeSec":40737,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"***********","BSSId":"2C:91:AB:A8:25:6F","Channel":1,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

### TO REPRODUCE
Turn off modle and power it back on again. After that the value for "Energy Total" is 0

### EXPECTED BEHAVIOUR
value for "Energy Total" is persisted

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
n/a

to-scho, 2021-01-01:
I guess that's because Tasmota do not want to break flash memory by writing any updated value each second or so. Values should be saved to flash each 24h or by reboot request. Power outage for sure is out of a sudden and not feasible to handle by flash storage.

You can work around by using your mqtt broker as a backup using rules. Each time energy value changes publish a persistant mqtt command raising a custom event that sets value back to last published one. Disable that event after boot so that only after boot the restore takes place. I use that for counter value by the following rule. You may adapt this to energy. For sure a total power outage incl. your mqtt brooker will break this again.
 
rule1 on counter#c1>%var4% do backlog var4 %value%; publish2 cmnd/%topic%/event c1restore=%value% endon on event#c1restore do backlog var2 %value%; event c1crestore=%var3% endon on event#c1crestore= do backlog var3 1; counter1 %var2% endon

Jason2866, 2021-01-01:
Duplicate. As designed. If there is no activity that needs saving to flash the energy values are saved once per day (around midnight). This is done to save flash from wear out.

herangstl, 2021-01-01:
The module was running since several days, but it seems the value for "Energy Total" was never persisted.

But some values must have been persisted, for example the value for "Energy Yesterday" (it was greater 0 after I powered the module back on).

I had assumed that when "Energy Yesterday" was persisted "Energy Total" would have also been persisted.

Frogmore42, 2021-01-01:
At around midnight, the value for today is almost always 0.

herangstl, 2021-01-01:
> 
> 
> At around midnight, the value for today is almost always 0.

Sure. But that is not the problem here.

Frogmore42, 2021-01-02:
Since the data is moved to non-volatile at around midnight, it actually probably is your problem.  If you reset your module before pulling the power, it should save to non-volatile, and you will get your desired behavior.

xinhu808, 2021-01-03:
the same issue.

ascillato2, 2021-01-03:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-03 by ascillato2

herangstl, 2021-01-03:
> Closing this issue as it is **duplicated**. Sorry.

Hello ascillato2, can you please add a reference to the duplicated issue?
Thank you

Frogmore42, 2021-01-04:
Not sure why you disliked my response above.  It is the way the system works.  It looks like the code doesn't actually save the Today value at midnight, since it would always be zero.  It only appears to save this on a reset.

Here is the link to a previous issue which is itself a duplicate of many previous ones:
https://github.com/arendst/Tasmota/issues/9132

Saving to flash adds wear to the flash.  The code saves at midnight to minimize the lost information, while still maintaining decent life for the flash.  Saving to flash every second would wear out the flash very quickly.  Saving every hour, would still lose some data and would have a significantly reduced life.