arendst/Tasmota #10357: MPU-6050 does not compile

elik745i, 2021-01-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._Can't compile binary for use of MPU-6050 not through gitpod neither arduino IDE. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____Wemos D1 mini
- [ ] Tasmota binary firmware version number used: _____9.2
  - [ ] Pre-compiled 
  - [ ] Self-compiled Selfcompilled
- [ ] Flashing tools used: _____nodeMCU flasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-01-01:
Tasmota does compile successfully with driver MPU-6050 enabled.
Just tested with Gitpod. Tasmota cannot be compiled with Arduino-IDE.
Support for Arduino-IDE has been dropped since a while

elik745i, 2021-01-02:
> Tasmota does compile successfully with driver MPU-6050 enabled.
> Just tested with Gitpod. Tasmota cannot be compiled with Arduino-IDE.
> Support for Arduino-IDE has been dropped since a while

I have tested it and it exits with error, compilation parameters:

#ifndef USE_I2C
  #define USE_I2C
#endif

#ifndef USE_MPU6050
#define USE_MPU6050              
  #define USE_MPU6050_DMP       
#endif

#ifndef USE_BMP
  #define USE_BMP
#endif

#ifndef USE_DS3231
  #define USE_DS3231
#endif

#ifndef USE_DHT
  #define USE_DHT
#endif

elik745i, 2021-01-02:
Withiut I2C here are no selection to chose SDA and SCL in the module configuration, with I2C it does not compile. Maybe I'm doing something wrong???

Jason2866, 2021-01-02:
It seems you do not use build variant Tasmota for your build. See https://tasmota.github.io/docs/Compile-your-build/#customize-your-build

elik745i, 2021-01-02:
> It seems you do not use build variant Tasmota for your build. See https://tasmota.github.io/docs/Compile-your-build/#customize-your-build

I go to: https://fb2adc08-8807-441c-bd67-22202305e6ec.ws-eu03.gitpod.io/#/workspace/tasmocompiler
it opens: TasmoCompiler v3.0.1
I click on - download source
Next i enter my wifi credentials, next select features I touch nothing
Next - Custom parameters: I enter:

#ifndef USE_I2C
  #define USE_I2C
#endif

#ifndef USE_MPU6050
#define USE_MPU6050              
  #define USE_MPU6050_DMP       
#endif

#ifndef USE_BMP
  #define USE_BMP
#endif

#ifndef USE_DS3231
  #define USE_DS3231
#endif

#ifndef USE_DHT
  #define USE_DHT
#endif

Next - Tasmota version
Core version - development
Language - english
Board version - wemos d1 mini
Board speed - 80mhz

Compillation terminated with error.

Please check yourself mate

Jason2866, 2021-01-02:
So you Tasmocompiler with Gitpod. Sorry we do not support Tasmocompiler.
It is a 3rd Party tool. 
You can use Gitpod directly with this Github. Just hit the Badge `Gitpod Ready to compile`

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

elik745i, 2021-01-02:
mmmm...

What means "we do not support" , I refer to it through official compile guideline...

https://tasmota.github.io/docs/Compile-your-build/#customize-your-build

"Simplest way to compile is with GitPod, requires only a web browser."

the one you referred before....

elik745i, 2021-01-02:
UPDATE^ the issue was with compiller, so I have used Visual Studio PlatformIO. Took 15 minutes to get the logic of it's work and help of community! Successfully compilled the build I neede with the following parameters (user_config_override.h file):

`
#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

// force the compiler to show a warning to confirm that this file is included
#warning **** user_config_override.h: Using Settings from this File ****

#ifndef USE_I2C
  #define USE_I2C
#endif

#ifndef USE_MPU6050
#define USE_MPU6050              
  #define USE_MPU6050_DMP       
#endif

#ifndef USE_BMP
  #define USE_BMP
#endif

#ifndef USE_DS3231
  #define USE_DS3231
#endif

#ifndef USE_DHT
  #define USE_DHT
#endif

#endif  // _USER_CONFIG_OVERRIDE_H_
`

If you face isues during compillation restart Visual Studio!

thanks again for help!

Issue is closed!

closed 2021-01-02 by elik745i