arendst/Tasmota #10361: Tasmota webserver not reachable on Sonoff devices after some time

TungstenE2, 2021-01-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have 2 Sonoff POW2 and 3 Sonoff S20 plugs. All are updated with 9.2.0.

On all devices the webserver becomes unreachable after some time, while MQTT is still working.

I have seen older posts with the same erroro reported by no solution.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Pow R2 and Sonoff Basic 
- [x] Tasmota binary firmware version number used: 9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Webserver becomes not reachable after some time, not special steps to reproduce

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Webserver should be reachable always

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-01-02:
Which wifi equipment do you have (brand type)? Is it a Mesh network?
Have you tried `SetOption41`?
Asus and Ubiquiti is known to have issues for some users. The reason for is still not found.
It is probably a incompatibility between espressif nonos-sdk  (closed source) and the firmware of the wifi equipment

TungstenE2, 2021-01-02:
Yes, it is Mesh by Ubiquiti Unifi wifi network.

I also have serveral ESP8266 and ESP32 devices (espressif). 

ESP8266 running ESPeasy and have no Wifi issues.

So how best to identify the issue with Tasmota?

ascillato2, 2021-01-03:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-03 by ascillato2

TungstenE2, 2021-01-03:
@ascillato2 thx for your reply. Why do you believe this is not a software bug?

The reported issue is only with Tasmota software, none of my other IoT devices is having this issue.
Thats why I opened this ticket reporting an issue with the Tasmota webserver.

TungstenE2, 2021-01-03:
How can this issue be network related if MQTT is working but webserver is not responding? Both is operated by the same network.

Here is older ticket with similar issue:
https://github.com/arendst/Tasmota/issues/266

@ascillato2 you marked the ticket as duplicated, please give ticket number of the other ticket you are referring to. Thx.

Jason2866, 2021-01-03:
Because we have some feedbacks from users with Ubiquiti hardware with exactly this behaviour. Webserver is timing out but mqtt is working. After changing the wifi hardware the issue is gone.
So the question is why are you sure it is NOT the wifi hardware?
The issue you refer is very old. The issue has been found in Arduino ESP8266 core and it is fixed since 2019.

TungstenE2, 2021-01-03:
Dear @Jason2866, thanks for your kind answer.

The reason I have my doubts about the wifi hw being the reason is that I have several other IoT devices connected to the same wifi hw and network without any issues. I have several Wemos D1 mini (ESP8266) and ESP32 where the webserver is not timing out. Only differnce is the software running on the devices. All devices with Tasmota the webserver is timing out and becomes unreachable. That is the reason I doubt the wifi network to be the reason of the issue.

So please reopen the issue in order to find the reason together, but just closing the issue is not a good way of dealing with it.

Happy to support in order to find the root cause, as you already stated this has been reported before. 

thx a lot.

TungstenE2, 2021-01-04:
I have tried `SetOption41 60` via the console on all my devices, and for the last 18hrs the webserver was active and reachable.
May be this solved the issue.

netzspezialist, 2025-09-28:
Same issue here:

**Device:** ESP-01s + [ESP-01 Relay V4.0 Relay Board](https://templates.blakadder.com/ESP-01S-Relay-v4.html)
**Firmware:** 15.0.1

Initially, during the Wi-Fi setup process, the Tasmota web UI worked perfectly fine. However, after the initial configuration was complete, I started experiencing intermittent issues accessing the UI. When I *was* able to connect to the module's IP address, only the main menu would load successfully. I was consistently unable to open any sub-menus (e.g., Configuration, Consoles, etc.) – clicking on them would either do nothing or lead to a page that wouldn't load.

I eventually managed to configure both the Wi-Fi and MQTT settings for the module using Tasmotizer, which allowed me to get the device working for its intended purpose.

**Temporary UI Access & Subsequent Loss:**
Approximately one hour after the initial setup and Tasmotizer configuration, I was surprisingly able to access the Tasmota WebUI completely and it functioned perfectly. During this period, I successfully changed the Tasmota template ({"NAME":"ESP01v4","GPIO":[256,320,0,32,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}) and then mounted the ESP-01s onto a relay board.

Since then, the MQTT functionality and ping responses have been working flawlessly (allowing me to control the relay ON/OFF). However, I am now completely unable to access the WebUI again. Attempts to reach the module's IP address in a browser result in no response or a connection timeout.

sfromis, 2025-09-28:
Typical cause of Wifi trouble is power issues. Both about available ampere, voltage, stability and (especially with borderline hardware like ESP-01) if decoupling capacitor is sufficient. This also fits your initial symptoms about "intermittent issues".