arendst/Tasmota #10375: Active Watts and Watts different output.

andrewchumchal, 2021-01-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Active Watts in /metrics is different output then Watts in WebGui

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): TGWF115APM
- [X] Tasmota binary firmware version number used: 9.2.0.2
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

20:17:19.458 MQT: stat/***/RESULT = {"NAME":"TGWF115APM","GPIO":[0,320,0,32,2720,2656,0,0,2624,321,224,0,0,0],"FLAG":0,"BASE":18}
20:17:19.674 MQT: stat/****/RESULT = {"Module":{"0":"TGWF115APM"}}
20:17:19.883 MQT: stat/****/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"320":"Led_i1"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"321":"Led_i2"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

20:10:59 MQT: stat/****/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
20:10:59 MQT: stat/****/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
20:10:59 MQT: stat/****/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

20:18:06.593 MQT: stat/****/STATUS = {"Status":{"Module":0,"DeviceName":"****","FriendlyName":["****"],"Topic":"****","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:18:06.625 MQT: stat/****/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:35","StartupUTC":"2021-01-02T19:16:31","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2021-01-02T04:54:09","SaveCount":45,"SaveAddress":"FA000"}}
20:18:06.659 MQT: stat/****/STATUS2 = {"StatusFWR":{"Version":"9.2.0.2(tasmota)","BuildDateTime":"2021-01-02T13:08:02","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"445/699"}}
20:18:06.679 MQT: stat/****/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IDNoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
20:18:06.710 MQT: stat/****/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00104004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,75"}}
20:18:06.742 MQT: stat/****/STATUS5 = {"StatusNET":{"Hostname":"****-1327","IPAddress":"10.10.***.***","Gateway":"10.10.108.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.200.20","Mac":"84:0D:8E:4C:45:2F","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
20:18:06.766 MQT: stat/****/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.0.2","MqttPort":1883,"MqttClientMask":"****","MqttClient":"****","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:18:06.790 MQT: stat/****/STATUS7 = {"StatusTIM":{"UTC":"2021-01-02T19:18:06","Local":"2021-01-02T20:18:06","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:04"}}
20:18:06.809 MQT: stat/****/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:18:06.828 MQT: stat/****/STATUS10 = {"StatusSNS":{"Time":"2021-01-02T20:18:06","ENERGY":{"TotalStartTime":"2021-01-02T04:45:55","Total":0.229,"Yesterday":0.000,"Today":0.229,"Power":40,"ApparentPower":40,"ReactivePower":0,"Factor":1.00,"Voltage":119,"Current":0.332}}}
20:18:06.851 MQT: stat/****/STATUS11 = {"StatusSTS":{"Time":"2021-01-02T20:18:06","Uptime":"0T00:01:35","UptimeSec":95,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IDNoT","BSSId":"F2:63:DA:74:88:49","Channel":11,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

20:20:06.947 CMD: weblog 4
20:20:06.954 MQT: stat/****/RESULT = {"WebLog":4}
20:20:07.383 CFG: Saved to flash at F9, Count 46, Bytes 4096
20:20:09.466 HTP: Prometheus
20:20:18.522 WIF: Checking connection...
20:20:24.405 HTP: Prometheus
20:20:25.958 HTP: Console
20:20:28.557 HTP: Main Menu
20:20:30.349 SRC: WebGui from 10.10.***.***
20:20:30.357 MQT: tele/****/STATE = {"Time":"2021-01-02T20:20:30","Uptime":"0T00:03:59","UptimeSec":239,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IDNoT","BSSId":"F2:63:DA:74:88:49","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:20:30.382 MQT: stat/****/RESULT = {"POWER":"OFF"}
20:20:30.387 MQT: stat/****/POWER = OFF
20:20:30.616 CFG: Saved to flash at F8, Count 47, Bytes 4096
20:20:33.701 SRC: WebGui from 10.10.***.***
20:20:33.710 MQT: tele/****/STATE = {"Time":"2021-01-02T20:20:33","Uptime":"0T00:04:02","UptimeSec":242,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IDNoT","BSSId":"F2:63:DA:74:88:49","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:20:33.734 MQT: stat/****/RESULT = {"POWER":"ON"}
20:20:33.739 MQT: stat/****/POWER = ON
20:20:34.642 CFG: Saved to flash at F7, Count 48, Bytes 4096
20:20:38.991 WIF: Checking connection...
20:20:39.935 HTP: Prometheus

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Enable Prometheus
Plug in 40w bulb
See Active watts at 60 in /metrics and 40 in the webgui

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

We should see 40 in both webgui and metrics

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-01-03:
Sorry, cannot reproduce. I see metrics at around the same value as in the GUI.

As these values are measured over several seconds they may be different but I agree that 20W is much.

Both interfaces are using the same active power register though so I see no coding issue other than timing between both readouts. In your interfaces I notice also a difference in ENergy Today so they are definitvely not taken at the same time.

ascillato2, 2021-01-03:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-03 by ascillato2

andrewchumchal, 2021-01-04:
> Sorry, cannot reproduce. I see metrics at around the same value as in the GUI.
>
> As these values are measured over several seconds they may be different but I agree that 20W is much.
>
> Both interfaces [...]

Thank you for taking your time and looking into this.