arendst/Tasmota #10392: RA8876 not working (tried with ESP8266 and ESP32)

helmar74, 2021-01-03:
### PROBLEM DESCRIPTION
Hi,

I tried to get a RA8876 Display running with Tasmota  9.2.0.2 
Unfourtnately, I'm not able to get it up and running either on ESP 8266 nor on ESP32.
Here's, how I connected the Display to ESP8266 or ESP32:

ESP32 			<-> 	RA8876
TDI (GPIO12 MISO) <-> 	7 SDI
TCK (GPIO13 MOSI)	<-> 	6 SDO
TMS (GPIO14 SCLK)<-> 	8 SCLK
TDO (GPIO15 CS)	<-> 	5 CS

ESP8266
Wemos MINI D 	<-> 	RA8876
D6 GPIO12 (MISO) 	<-> 	7 SDI
D7 GPIO13 (MOSI) 	<-> 	6 SDO
D5 GPIO14 (SCLK)	<-> 	8 SCLK
D8 GPIO15 (CS)	<-> 	5 CS

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos Mini D (ESP8266) and ESP32 Devolpment Kit
- [X] Tasmota binary firmware version number used: 9.2.0.2
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
ESP8266 (Wemos Mini D):
21:29:57.133 CMD: Group 0, Index 1, Command "GPIO", Data "255"
21:29:57.139 RSL: stat/tasmota_1D88E5/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"672":"SPI MISO"},"GPIO13":{"864":"SSPI MOSI"},"GPIO14":{"736":"SPI CLK"},"GPIO15":{"6080":"RA8876 CS"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

ESP32:
21:36:54.161 CMD: Group 0, Index 1, Command "GPIO", Data "255"
21:36:54.171 MQT: stat/tasmota_F765E8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"672":"SPI MISO"},"GPIO13":{"704":"SPI MOSI"},"GPIO14":{"736":"SPI CLK"},"GPIO15":{"6560":"RA8876 CS"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"}}
21:36:54.221 MQT: stat/tasmota_F765E8/RESULT = {"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
ESP8266 (Wemos Mini D):
21:30:45.371 RSL: stat/tasmota_1D88E5/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_1D88E5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:30:45.397 RSL: stat/tasmota_1D88E5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:01:58","StartupUTC":"2021-01-03T20:28:47","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2021-01-03T21:16:08","SaveCount":20,"SaveAddress":"F8000"}}
21:30:45.433 RSL: stat/tasmota_1D88E5/STATUS2 = {"StatusFWR":{"Version":"9.2.0.2(display)","BuildDateTime":"2021-01-03T17:47:27","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"349/699"}}
21:30:45.453 RSL: stat/tasmota_1D88E5/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFISSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
21:30:45.484 RSL: stat/tasmota_1D88E5/STATUS4 = {"StatusMEM":{"ProgramSize":563,"Free":440,"Heap":29,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAA858E","04049FA1","000000C3","00000000","00004080","00000020","000F8000"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6"}}
21:30:45.512 RSL: stat/tasmota_1D88E5/STATUS5 = {"StatusNET":{"Hostname":"tasmota_1D88E5-2277","IPAddress":"192.168.1.178","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"DC:4F:22:1D:88:E5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:30:45.536 RSL: stat/tasmota_1D88E5/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1D88E5","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:30:45.557 RSL: stat/tasmota_1D88E5/STATUS7 = {"StatusTIM":{"UTC":"2021-01-03T20:30:45","Local":"2021-01-03T21:30:45","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:05"}}
21:30:45.574 RSL: stat/tasmota_1D88E5/STATUS10 = {"StatusSNS":{"Time":"2021-01-03T21:30:45"}}
21:30:45.584 RSL: stat/tasmota_1D88E5/STATUS11 = {"StatusSTS":{"Time":"2021-01-03T21:30:45","Uptime":"0T00:01:58","UptimeSec":118,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"WIFISSID","BSSId":"CC:CE:1E:AB:0A:3F","Channel":6,"RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:03"}}}

ESP32:
21:37:13.007 CMD: Group 0, Index 1, Command "STATUS", Data "0"
21:37:13.016 MQT: stat/tasmota_F765E8/STATUS = {"Status":{"Module":1,"DeviceName":"ESP32Test","FriendlyName":["ESP32Test"],"Topic":"tasmota_F765E8","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:37:13.048 MQT: stat/tasmota_F765E8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:00:32","StartupUTC":"2021-01-03T20:36:41","Sleep":50,"CfgHolder":4617,"BootCount":44,"BCResetTime":"2020-12-02T09:01:42","SaveCount":123}}
21:37:13.082 MQT: stat/tasmota_F765E8/STATUS2 = {"StatusFWR":{"Version":"9.2.0.2(display)","BuildDateTime":"2021-01-02T14:03:11","Core":"1_0_5_RC4","SDK":"v3.3.4-286-gb4c075169","CpuFrequency":80,"Hardware":"ESP32","CR":"410/699"}}
[... 126 lines cut ...]
21:37:13.158 MQT: stat/tasmota_F765E8/STATUS5 = {"StatusNET":{"Hostname":"tasmota_F765E8-1512","IPAddress":"192.168.1.226","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"4C:11:AE:F7:65:E8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:37:13.182 MQT: stat/tasmota_F765E8/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F765E8","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:37:13.201 MQT: stat/tasmota_F765E8/STATUS7 = {"StatusTIM":{"UTC":"2021-01-03T20:37:13","Local":"2021-01-03T21:37:13","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:01","Sunset":"16:25"}}
21:37:13.224 MQT: stat/tasmota_F765E8/STATUS10 = {"StatusSNS":{"Time":"2021-01-03T21:37:13"}}
21:37:13.235 MQT: stat/tasmota_F765E8/STATUS11 = {"StatusSTS":{"Time":"2021-01-03T21:37:13","Uptime":"0T00:00:32","UptimeSec":32,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:04"}}}
21:37:27.644 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect the RA 8876 Display as described above.
Try to send Text with "Displaytext Command"
You can either try ESP32 or ESP8266. The Display will never work. I tried this for weeks now!

### EXPECTED BEHAVIOUR
Text should be displayed. But this doesn't happen!

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

helmar74, 2021-01-03:
This is the ESP32 I'm using:

[image]

arendst, 2021-01-03:
As can be seen on the esp32 line 4 this should also be shown on esp8266.

On esp8266 remove mosi/miso/clk gpio config and use only the ra8876_cs on gpio15.

On esp32 use the vspi gpios.

arendst, 2021-01-03:
You probably need to configure a backlight gpio too.

helmar74, 2021-01-03:
Thanks for the info.
Currently trying ESP32.
Should I also remove the mosi/miso/clk gpio config on ESP32?

helmar74, 2021-01-03:
Backlight PIN must not be used, can be switch on with 5V DC

helmar74, 2021-01-03:
DO I have to "cross" MISO and MOSI?

helmar74, 2021-01-03:
This is my current config on ESP32:
`22:00:30.535 CMD: Group 0, Index 1, Command "GPIO", Data "255"
22:00:30.545 MQT: stat/tasmota_F765E8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"6560":"RA8876 CS"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"736":"SPI CLK"},"GPIO19":{"672":"SPI MISO"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"704":"SPI MOSI"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"}}
22:00:30.599 MQT: stat/tasmota_F765E8/RESULT = {"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}`

[image]

Is thiis correct?

arendst, 2021-01-03:
Yes. But the driver needs a gpio configured as backlight.

helmar74, 2021-01-03:
Configured it that way:
[image]

Not working ;-)

Here's the Datasheet of the display:
[image]

helmar74, 2021-01-03:
Strange things are also:
1. I can't measure a clock signal on GPIO18 of the ESP32
2. If I hard powercycle the ESP32 it doesn't boot correctly. I first have to disconnect VCC from the display.
3. It can't be an power issue. Using a LAbor Power supply with 5V/10A

gemu2015, 2021-01-04:
i have this display in use currently on esp32 and initially on esp8266. i also wrote the driver.

here is my current esp32 configuration:

18 = SPI_SCLK
19 = SPI_MISO
21 = SPI_CS with the most recent driver use "RA8876 CS"
23 = SPI_MOSI

// touch and other functions
4 = SCL
5 = SDA
22 = MP3 Player
27 = CS_SDC

backlight is not used by the driver, since the driver has its own backlight PWM 
it happens that i updated the driver yesterday to the newest dev version and it works without problems.

what brand do you have ?

mine is from buydisplay.com and was preconfigured to 4 line spi , 3,3 V

the driver uses 4 line spi without dc

i remember it was a hassle to identify the right pins on the various double row pin connectors on my board

according to power supply the board and the esp32 together may draw more then 1 A when backlight is at 100 %

i assume at least your pinout is wrong
ESP32: <-> RA8876
GPIO19 (MISO) <-> 7 SDI
GPIO23 (MOSI) <-> 6 SDO
GPIO18 (SCLK) <-> 8 SCLK
GPIO05 (CS) <-> 5 CS

sdo = serial data out has to be connected to MISO (master in slave out)
sdi = serial data in has to be connected to MOSI (master out slave in)

the pin numbers you use on esp32 are not critical since the esp32 multiplexes the pins for hardware spi.

however you should not use pins that can modify the boot process, yours seem to be ok

it is a good tip to compile this driver  with
#define SHOW_SPLASH
because this initially draws an init message on the screen. without this you always have to draw something yourself to show if it works

you did not provide a dump of cmd display yet.

helmar74, 2021-01-04:
Hi,

thank you for the detailed information.
I have also a display from  buydisplay.com. IT is configured as 4-SPI with 5V
Here's the datasheet:
[ER-TFTM101-1_Datasheet.pdf](https://github.com/arendst/Tasmota/files/5763548/ER-TFTM101-1_Datasheet.pdf)
Maybe, we have the same?

Could you please tell me, how you cabeled the ESP32 with the display?

I'm currently using this Pin of the "big" JP1 Connector:
[image]

My Backlight is currently configured for external PIN:
[image]

How is yours configured? (I'm currently thinking of changing it to PWM (solder J27 Open, J28 Short)

Which Pins have you connected on the display in order to use the touch screen?
(... and how did you configure the ESP Pins)?

Thank you very much

helmar74, 2021-01-04:
Change Cabeling:
ESP32: 				<-> 	RA8876
GPIO19 (MISO) 		<-> 	6 SDO
GPIO23 (MOSI) 		<-> 	7 SDI
GPIO18 (SCLK)		<-> 	8 SCLK
TDO GPIO15 (CS)		<-> 	5 CS

Now I can send Text with displaytext command.
Thank you very much!

Next steps will be:
- Changing Backlight to PWM. Do I have to use the displaydimmer command to set display lights?
- Trying to use touchscreen. Do you have some tips, how I can simply create a touchbutton?

gemu2015, 2021-01-04:
the DisplayDimmer handles the pwm inside the chip.
for touch connect and configure the i2c interface and enable the touch chip with
#define USE_FT5206
touch buttons are only usable together with scripts.
so make yourself familiar with scripts
you then may define up to 16 (default) touch buttons which you can directly be used with relays or virtual buttons which you may use without relays.
meanwhile i have also support for picture buttons, with one or 2 pictures (per button state)
the sd card is also supported in scripts. you may then display jpeg pictures from sd card

these are my defines:

#define USE_SCRIPT
#define USE_DISPLAY
#define USE_SPI
#define USE_DISPLAY_RA8876
#define USE_FT5206

#define USE_SCRIPT_FATFS 27  // GPIO 27 to card cs
#undef FAT_SCRIPT_SIZE
#define FAT_SCRIPT_SIZE 8192
#define USE_TOUCH_BUTTONS
#define JPEG_PICTS

helmar74, 2021-01-04:
I also "configured" for PWM Backlight.
When I set "displaydimmer 100" it is set to 15.
Is this correct?
Here's the cmd log:
`12:58:54.509 CMD: displaydimmer 100
12:58:54.510 SRC: WebConsole from 192.168.1.36
12:58:54.513 CMD: Group 0, Index 1, Command "DISPLAYDIMMER", Data "100"
12:58:54.522 MQT: stat/tasmota_F765E8/RESULT = {"DisplayDimmer":15}
12:58:55.545 CFG: Saved, Count 164, Bytes 4096
12:58:56.733 WIF: Checking connection...
12:59:04.936 CMD: displaydimmer
12:59:04.938 SRC: WebConsole from 192.168.1.36
12:59:04.940 CMD: Group 0, Index 1, Command "DISPLAYDIMMER", Data ""
12:59:04.948 MQT: stat/tasmota_F765E8/RESULT = {"DisplayDimmer":15}`

gemu2015, 2021-01-04:
this is correct , the displays have only 16 steps

helmar74, 2021-01-04:
Two further things ;-)

1. How can I use special characters (e.g. °&$§%)  ?
2. Has someone a "working example" how I can display a jpeg?
   - With which Filesystem must the SD Card be formatted?

gemu2015, 2021-01-04:
1. special characters are displays using the escape character tilde (see in tasmota display docs)
2. first you must compile with scripts and make yourself familiar with it.
the disk must be formatted with FAT32.
you then get a file system with open, close etc file calls.
to display a picture use the display text cmd e.g. Displaytext [x220y40P/wifi.jpg:]
displays a wifi Symbol on position 220,40

use
#define SDCARD_DIR
this shows a directory of the card with options to upload and download files.

helmar74, 2021-01-04:
Thank you.
Found the documentation about special characters, also working now.
One question regarding this:
How can I show ° when using EPD Font?

gemu2015, 2021-01-04:
the EPD font does not have this char. (see character fonts in the docs)
you have to modify the c source code of the fonts.
i wrote es special c source code font editor which directly edits the source files with a gui
this allows to add any chars you want to all font files, also edit gfx fonts.

you may edit the fonts and then make a pull request with your modified font. I am sure other people would be interested too in such special chars.

the font editor is included in my script editor under Menu Various->Start

the font files are in lib/lib_display/esp-epaper.../src

https://www.dropbox.com/sh/0us18ohui4c3k82/AACcVmpZ4AfpdrWE_MPFGmbma?dl=0

[image]

ascillato2, 2021-01-04:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-04 by ascillato2

helmar74, 2021-01-07:
> 1. special characters are displays using the escape character tilde (see in tasmota display docs)
> 2. first you must compile with scripts and make yourself familiar with it.
>    the disk must be [...]

Hi,

thanks for all of your help.
Finally I get the display running.
I'm able to display text, draw lines and so on.
I still have trouble to show pictures.
I have compiled Tasmota 9.2.0.2 dev version. with following overrides set:
`#undef USE_DISPLAY_RA8876
#define USE_DISPLAY_RA8876
#define SDCARD_DIR
#undef USE_RULES // undefined due to Compile errors
#define USE_SCRIPT`

I have inserted a FAT32 formated SD Card in the SD Card Slot of the display.
With Command:
`Displaytext [x220y40P/wifi.jpg:]`
I'm not able to show a picture.

So my questions are:
1. Do I also have to connect the PIN's of the SDCARD Reader of the Display to the ESP32?
2. IF so, how mus I connect the SDCARD Reader to the ESP GPIO's and how are the settings of the GPIOS?

Thanks in advance.

gemu2015, 2021-01-07:
yes you must connect the sd card header pins, there is no internal connection on the board.

there is no change in settings except the SPI_CS of the sd card.

use the most recent dev version. there you can choose the sd card chip select pin directly in the GUI

there were some recent changes to #defines
you have to define

#define USE_UFILESYS (i think this now default anyhow)
#define USE_SDCARD (allow SD card detection)

in the setup menu you will find an item "Manage file system" where you should see the contents of your sd card. you have to provide your picture files yourself on the card. wifi.jpg was only an example 

once this is done you may test the picture drawing again.

helmar74, 2021-01-07:
> yes you must connect the sd card header pins, there is no internal connection on the board.
>
> there is no change in settings except the SPI_CS of the sd card.
>
> use the most recent dev version. there [...]

Sorry to bother you again.

This are my Pins of the SD CARD REader:
[image]

Which Pins from the SD CARD Reader Do I Have connect?
Only PIN8 to GPIO27 on ESP32?

gemu2015, 2021-01-07:
// these 3 pins connect to the same as the display (SPI is a bus)
DIN = SPI_MOSI
DOUT = SPI_MISO
SCLK = SPI_SCLK
// connect this to an unused pin of ESP32 pin number < 34
/CS = and select SPI_CS_SDCARD in Tasmota GUI

helmar74, 2021-01-07:
Hi again,

I tried to compile newest TAsmota version (9.2.0.3) with the following options in override:
`/ Für RA 8876 SPI DISPLAY

#undef USE_DISPLAY_RA8876
#define USE_DISPLAY_RA8876

#define SHOW_SPLASH  

#define USE_UFILESYS
#define USE_SDCARD`

The build process failed with this error:
`Linking .pio\build\tasmota32\firmware.elf
.pio\build\tasmota32\libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'
.pio\build\tasmota32\libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'
.pio\build\tasmota32\libFrameworkArduino.a(main.cpp.o): In function `loopTask(void*)':
C:\users\helmar\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:18: undefined reference to `setup()'
C:\users\helmar\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:21: undefined reference to `loop()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\tasmota32\firmware.elf] Error 1
=================================================================================== [FAILED] Took 218.12 seconds ===================================================================================
Environment      Status    Duration
---------------  --------  ------------
tasmota-minimal  FAILED    00:00:15.036
tasmota32        FAILED    00:03:38.122
=====================================================================`

gemu2015, 2021-01-07:
i just compiled the newest version, no error.
clean your build and try again.

helmar74, 2021-01-07:
Hi,

now I see the the filesystem:
[image]

Last (hopefully) question:
How can I display one of the images on the 10.1 display?

gemu2015, 2021-01-07:
you may only display JPEG images! 
they should also have the ending .jpg and best use short names only

helmar74, 2021-01-07:
OK,  tried this command:
`backlog displaytext [z]; Displaytext [x0y0P/eingang.jpg:]`

the File eingang.jpg ist present on SD CARD:
[image]

Picture isn't displayed, what I'm doing wrong?

gemu2015, 2021-01-07:
your picture has ZERO bytes ????

gemu2015, 2021-01-07:
i see you also switched to Flash file system!
you should see UFS not FFS
simply remove power and restart until you see UFS again.

gemu2015, 2021-01-07:
you also should know that you cannot resize pictures. the picture is displayed in original size

gemu2015, 2021-01-07:
the UFS FFS switch is brand new (from today) by clicking you may switch between the SD card and the internal Flash file system which is very small on your system (only 64kb)
watch the total and free size info

helmar74, 2021-01-07:
Switch back tu UFS:
[image]

Picture sensor1.jpg still not shown:
backlog displaytext [z]; Displaytext [x100y100P/sensor1.jpg:]

The Syntax is correct for displaying pictures?
Does the picture itself need special settings (Color Depth, resolution....)?

gemu2015, 2021-01-07:
oh i see you did not enable jpeg pictures!
#define JPEG_PICTS

helmar74, 2021-01-07:
I also compiled in this option.
JPEG still not shown with command:
`backlog displaytext [z]; Displaytext [x100y100P/sensor1.jpg:]; Displaytext [x800y200f2] TEST`

TEST is shown...

gemu2015, 2021-01-07:
ok just tried from console and it works. i am not quite sure but i sometimes had jpegs that did fail in the decoder
i attach a picture that worked here

[wifi.jpg.zip](https://github.com/arendst/Tasmota/files/5782895/wifi.jpg.zip)

helmar74, 2021-01-07:
Just tried the wifi.jpg file. This isn't also displayed

Could you please send me your complete user_override.h ?

gemu2015, 2021-01-07:
i do not use rules, so i have a bunch of script options which however should not affect jpeg pictures

#define USE_SCRIPT
#define USE_UFILESYS
#define USE_SDCARD
#define UFSYS_SIZE 8192
#define SDCARD_CS_PIN 27

#define USE_DISPLAY
#define USE_SPI
#define USE_DISPLAY_RA8876
#define USE_FT5206

#define USE_GOOGLE_CHARTS
#define SCRIPT_FULL_WEBPAGE
#define USE_SCRIPT_GLOBVARS
#define USE_I2S_AUDIO
#define SAY_TIME
#define USE_WEBRADIO

#define USE_MP3_PLAYER
#define USE_TOUCH_BUTTONS
#undef NUM_GRAPHS
#define NUM_GRAPHS 8

#define JPEG_PICTS

#define USE_SCRIPT_TASK
#define SCRIPT_GET_HTTPS_JP

#define USE_SENDMAIL
#define USE_ESP32MAIL

helmar74, 2021-01-07:
Here's my override:
// Für RA 8876 SPI DISPLAY

#undef USE_DISPLAY_RA8876
#define USE_DISPLAY_RA8876
//***************************************
// Aus https://github.com/arendst/Tasmota/issues/10392
#define SHOW_SPLASH  // Schreibt Inimeldung auf das Display
//#define SDCARD_DIR
#define USE_UFILESYS
#define USE_SDCARD
#define JPEG_PICTS //Für Anzeige von JPEG Bildern

gemu2015, 2021-01-07:
as far as i know you may NOT comment after a #define line !!!
this define may be ignored !

helmar74, 2021-01-07:
OK, thanks for the hint, will try.
But SHOW_SPLASH is working...

helmar74, 2021-01-07:
With this settings it is working now:
`// Für RA 8876 SPI DISPLAY

#undef USE_DISPLAY_RA8876
#define USE_DISPLAY_RA8876
//***************************************
// Aus https://github.com/arendst/Tasmota/issues/10392
// Schreibt Init-Meldung auf das Display
#define SHOW_SPLASH  

#define USE_UFILESYS
#define USE_SDCARD
//Für Anzeige von JPEG Bildern 
#define JPEG_PICTS 
// Wegen Compilefehler
#undef USE_RULES 
#define USE_SCRIPT`

(I added USE_SCRIPT...)

Not every picture is shown.
The maximum picture size for me is 166x166 pixel. Is there any possibility to increase picture size?

gemu2015, 2021-01-07:
this is a memory problem. You should use an ESP32 board with PSRAM.
then you may display full size pictures

helmar74, 2021-01-07:
OK, can you suggest a ESP32 Board with PSRAM?
Currently I have a ESP32 Mini Board... 
I also have a ESP32 DEVELOPMENT Board (Node MCU)

helmar74, 2021-01-07:
Are 8 MB PSRAM enough?

gemu2015, 2021-01-07:
i also have this board on my ra8876

the ESP32 can only use 4 MB PSRAM but that is enough for large pictures

helmar74, 2021-01-07:
OK, ordered one, let's wait til saturday ;-)
I'll provide feedback...
Do I have to change compile options in order to use PSRAM on the Board?

gemu2015, 2021-01-07:
yes you must enable PSRAM in compiler.

-DBOARD_HAS_PSRAM

or select the board directly in PlatformIO

helmar74, 2021-01-07:
I'm compiling with Visual Studio Code, How can I enable PSRAM there?
Sorry for my stupid questions, but I just don't know...

gemu2015, 2021-01-07:
i don't know either, i use platformio

helmar74, 2021-01-07:
OK, and where do I enable it there?

gemu2015, 2021-01-07:
append this option to build_flags
-DBOARD_HAS_PSRAM

in platform_override.ini

helmar74, 2021-01-07:
[Tasmota-development_v9203_platformiooverrides.zip](https://github.com/arendst/Tasmota/files/5783668/Tasmota-development_v9203_platformiooverrides.zip)
I have a .sample file, which I can rename...
... Or should I add it in platformio_tasmota_env32.ini
`.
.
[esp32_defaults]
build_unflags               = ${esp_defaults.build_unflags}
                              -Wpointer-arith
build_flags                 = ${esp_defaults.build_flags}
                              -DCORE_DEBUG_LEVEL=0
                              -DBUFFER_LENGTH=128
                              -DHTTP_UPLOAD_BUFLEN=2048
                              -DMQTT_MAX_PACKET_SIZE=1200
                              -Duint32=uint32_t
                              -Duint16=uint16_t
                              -Duint8=uint8_t
                              -Dsint8_t=int8_t
                              -Dsint32_t=int32_t
                              -Dsint16_t=int16_t
                              -Dmemcpy_P=memcpy
                              -Dmemcmp_P=memcmp`
Where Do I have to add the build_flag?

gemu2015, 2021-01-07:
at the end of the build_flags (not build_unflags)

 or you may also directly select this board

and i suggest to also set the CPU frequency to 240Mhz

please google for platformio board setup to find these options

helmar74, 2021-01-07:
OK, will give it a try...
But I have to set it in platformio_tasmota_env32.ini ?

gemu2015, 2021-01-07:
no there is a sample file vor platform overrides, rename by removing the "sample" and then in this file adapt your settings

helmar74, 2021-01-07:
OK, understood.
So I'll rename the file and change:
[common32]
platform                    = ${core32.platform}
platform_packages           = ${core32.platform_packages}
build_unflags               = ${core32.build_unflags}
build_flags                 = ${core32.build_flags}
board                       = esp32dev
board_build.ldscript        = esp32_out.ld
board_build.partitions      = esp32_partition_app1984k_spiffs64k.csv
board_build.flash_mode      = ${common.board_build.flash_mode}
board_build.f_flash         = ${common.board_build.f_flash}
board_build.f_cpu           = ${common.board_build.f_cpu}
monitor_speed               = ${common.monitor_speed}
upload_port                 = ${common.upload_port}
upload_resetmethod          = ${common.upload_resetmethod}
upload_speed                = 921600
extra_scripts               = ${common.extra_scripts}
lib_extra_dirs              = ${library.lib_extra_dirs}
; *** ESP32 lib. ALWAYS needed for ESP32 !!!
                              lib/libesp32
[core32]
; Activate Stage Core32 by removing ";" in next 4 lines, if you want to override the standard core32
;platform                    = ${core32_stage.platform}
;platform_packages           = ${core32_stage.platform_packages}
;build_unflags               = ${core32_stage.build_unflags}
;build_flags                 = ${core32_stage.build_flags}

to:

[common32]
platform                    = ${core32.platform}
platform_packages           = ${core32.platform_packages}
build_unflags               = ${core32.build_unflags}
build_flags                 = ${core32.build_flags}
board                       = esp32dev
board_build.ldscript        = esp32_out.ld
board_build.partitions      = esp32_partition_app1984k_spiffs64k.csv
board_build.flash_mode      = ${common.board_build.flash_mode}
board_build.f_flash         = ${common.board_build.f_flash}
board_build.f_cpu           = ${common.board_build.f_cpu}
monitor_speed               = ${common.monitor_speed}
upload_port                 = ${common.upload_port}
upload_resetmethod          = ${common.upload_resetmethod}
upload_speed                = 921600
extra_scripts               = ${common.extra_scripts}
lib_extra_dirs              = ${library.lib_extra_dirs}
; *** ESP32 lib. ALWAYS needed for ESP32 !!!
                              lib/libesp32
[core32]
; Activate Stage Core32 by removing ";" in next 4 lines, if you want to override the standard core32
;platform                    = ${core32_stage.platform}
;platform_packages           = ${core32_stage.platform_packages}
;build_unflags               = ${core32_stage.build_unflags}
build_flags                 = ${core32_stage.build_flags}
                                      **-DBOARD_HAS_PSRAM**

?

gemu2015, 2021-01-08:
in section [common32]
build_flags                 = ${core32.build_flags} -DBOARD_HAS_PSRAM
;board_build.f_cpu           = ${common.board_build.f_cpu}
board_build.f_cpu           = 240000000L

[41 more replies]