arendst/Tasmota #10400: Gosund SP1 forgets its meter readings after power failure

meierchen006, 2021-01-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hello,
Error with the device

Gosund SP1 Power Meter socket
Gosund SP1 v23 Module

After a power outage, the Gosund SP1 v23 Module software forgets your meter readings.
Firmware is the Tasmota 9.2.0.2 download at 14:30 on 04.01.2021.

Example:
I enter in the console:
Energy reset1 123456 == Today
Power reset3 235611 == Total

Then I simulate a power failure, for this I unplug the Gosund SP1 from the socket, wait 10 seconds, plug the Gosund SP1 back in.

The counter now shows:
Energy Today is at "0"
Energy Total is "0"

The expected result would be that the meter readings are the same, both meter readings should not be "0".

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Gosund SP1 v23 Module
- [x ] Tasmota binary firmware version number used: _____9.2.0.2
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [ ] Flashing tools used: _____Tasmota Firmware Update
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

15:26:27.197 CMD: Backlog Template
15:26:27.259 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}

15:26:36.892 CMD: Modul
15:26:36.898 RSL: RESULT = {"Command":"Unknown"}

15:26:47.498 CMD: GPIO 255
15:26:47.506 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"576":"LedLink_i"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

15:27:58.012 CMD: Status 0
15:27:58.018 RSL: STATUS = {"Status":{"Module":55,"DeviceName":"157-Test","FriendlyName":["157-Test"],"Topic":"157-Test","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:27:58.023 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:29:00","StartupUTC":"2021-01-04T13:58:58","Sleep":50,"CfgHolder":4627,"BootCount":45,"BCResetTime":"2020-12-13T16:05:48","SaveCount":752,"SaveAddress":"FB000"}}
15:27:58.027 RSL: STATUS2 = {"StatusFWR":{"Version":"9.2.0.2(tasmota)","BuildDateTime":"2021.01.04 14:38:19","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"421/699"}}
15:27:58.031 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxxxr",""],"TelePeriod":60,"Resolution":"559580C0","SetOption":["00209001","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
15:27:58.037 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":525,"Free":476,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000407","07000582","04180001","00000007","00000040","00004000","00004000","00000000"],"Drivers":"1,2,3,4,9,10,21","Sensors":"1,2,3"}}
15:27:58.042 RSL: STATUS5 = {"StatusNET":{"Hostname":"157-Test-5453","IPAddress":"192.168.178.157","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"xxxxxxxxxxxxxx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:27:58.047 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-01-04T14:27:58","Local":"2021-01-04T15:27:58","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"08:26","Sunset":"16:45"}}
15:27:58.053 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-01-04T15:27:58","ENERGY":{"TotalStartTime":"2021-01-04T00:00:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0.00,"ApparentPower":0.00,"ReactivePower":0.00,"Factor":0.00,"Voltage":232.80,"Current":0.000}}}
[... 57 lines cut ...]
00:00:00.019 Script: nv=0, tv=0, vns=0, ram=16
00:00:00.557 WIF: verbinden mit AP1 xxxxxxxx Channel 13 BSSId xxxxxxxxxxxxxxin Modus 11n als 157-Test-5453...
00:00:01.753 WIF: verbunden
00:00:02.004 HTP: Web-Server aktiv bei 157-Test-5453 mit IP-Adresse 192.168.178.157
00:00:02.008 RSL: INFO1 = {"Module":"Gosund SP1 v23","Version":"9.2.0.2(tasmota)","FallbackTopic":"cmnd/157-Test_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:02.015 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"157-Test-5453","IPAddress":"192.168.178.157"}
00:00:02.025 RSL: INFO3 = {"RestartReason":"Power On"}
00:00:02.029 RSL: RESULT = {"POWER":"OFF"}
00:00:02.033 RSL: POWER = OFF
14:59:06.465 QPC: Reset

14:59:06.469 RSL: STATE = {"Time":"2021-01-04T14:59:06","Uptime":"0T00:00:08","UptimeSec":8,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxxxxxx","BSSId":"xxxxxxxxxxxxxx","Channel":13,"RSSI":100,"Signal":-22,"LinkCount":1,"Downtime":"0T00:00:03"}}

14:59:06.491 RSL: SENSOR = {"Time":"2021-01-04T14:59:06","ENERGY":{"TotalStartTime":"2021-01-04T00:00:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0.00,"Power":0.00,"ApparentPower":0.00,"ReactivePower":0.00,"Factor":0.00,"Voltage":231.80,"Current":0.000}}

[image]

The expected result would be that the meter readings are the same, both meter readings should not be "0".

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

The expected result would be that the meter readings are the same, both meter readings should not be "0".

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-01-04:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-04 by ascillato2

barbudor, 2021-01-08:
For reference : energy restoration from flash on boot-up has been fixed in 9.2.0.3