arendst/Tasmota #10407: Rotary Encoder working with Shelly Dimmer 1/2

BPSoft, 2021-01-05:
### PROBLEM DESCRIPTION
I've a rotary encoder (PEC11L without On/Off switch) connected to a Shelly DImmmer 2.

Conection is similar to Instructable:
https://www.instructables.com/Shelly-Dimmer-Wall-Switch-With-Rotary-Knob-and-Hom/

Shelly                  
SW1 < -------- > PinA
GND < -------- > PinC
SW2 < -------- > PinB

In Tasmota library "support_rotary.ino", lines 139 and 140 we have:

- pinMode(Encoder[index].pina, INPUT_PULLUP);
- pinMode(Encoder[index].pinb, INPUT_PULLUP);

Problem is that for hardware reasons an INPUT_PULLUP will not work with Shelly Dimmer 1/2 and probably other devices.

I've changed to:

- pinMode(Encoder[index].pina, INPUT);
- pinMode(Encoder[index].pinb, INPUT);

After compile, everything is working extremely well!

These two lines are shared with both:

- Rotary.model = 1;    // Normal Rotary
- Rotary.model = 0;    // MI_DESK_LAMP (66)

I recommend the separation of pinMode by Rotary.model, either changing the Normal Rotary Model or creating a New Model.

For a Normal Rotary Encoder the number of steps doesn't matter, but this change:

- #define ROTARY_MAX_STEPS 100 // Rotary step boundary, default was 10

Will ease the rule set because we can do a direct rule like this:

- ON Rotary1#Pos1 DO Dimmer %value% ENDON

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly Dimmer 2 (also fix 1 and possible other devices)
- [X] Tasmota binary firmware version number used: 9.2.0
  - [X] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,3264,0,3296,0,0,4736],"FLAG":0,"BASE":18}
{"Module":{"0":"Shelly Dimmer 2"}}
{"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"5568":"SHD Boot 0"},"GPIO5":{"5600":"SHD Reset"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"3264":"Rotary_a1"},"GPIO13":{"0":"None"},"GPIO14":{"3296":"Rotary_b1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":39,"Free":472,"Rules":"ON Rotary1#Pos1 DO DIMMER %value% ENDON"}
{"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
{"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
{"Status":{"Module":0,"DeviceName":"SHDM-2","FriendlyName":["SHDM-2"],"Topic":"tasmota_14C4EB","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[11,11,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
{"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:12:24","StartupUTC":"2021-01-05T03:48:06","Sleep":50,"CfgHolder":4617,"BootCount":33,"BCResetTime":"2020-12-31T17:02:54","SaveCount":323,"SaveAddress":"FA000"}}
{"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2021-01-03T16:29:52","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"437/699"}}
{"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Dummy",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0002A00B","010501000100060001005A00000000000000","00000200","00036800","00000000"]}}
{"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
{"StatusNET":{"Hostname":"tasmota_14C4EB-6137","IPAddress":"192.168.10.24","Gateway":"192.168.10.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.254","Mac":"8C:AA:B5:14:C4:EB","WeBServer":2,"WifiConfig":4,"WifiPower":17.0}}
{"StatusMQT":{"MqttHost":"192.168.10.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_14C4EB","MqttUser":"mqtt_user","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
{"StatusTIM":{"UTC":"2021-01-05T04:00:30","Local":"2021-01-05T05:00:30","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:07"}}
{"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
{"StatusSNS":{"Time":"2021-01-05T05:00:30","ANALOG":{"Temperature":55.0},"ENERGY":{"TotalStartTime":"2020-12-31T17:02:54","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":5},"TempUnit":"C"}}
{"StatusSTS":{"Time":"2021-01-05T05:00:30","Uptime":"0T00:12:24","UptimeSec":744,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":12,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Dummy","BSSId":"8C:89:C4:96:3B:D1","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Not Applicable
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-01-05:
Thx. Will find a way to integrate.

arendst, 2021-01-06:
The latest commit adds the option to select Rotary GPIO with no pullup.

I haven't tried it yet but you might want to change the following in the Shelly Dimmer 1/2 template:
- change Rotary GPIO's to no pullup
- select base module Mi Desk Lamp (66)
- disable your rule

and see how it works.

BPSoft, 2021-01-07:
Thank you for your work and Time!

SetOption98 0 //turn off rules, turn on direct light

Forced Mi Desk Lamp (66) as base model and introduced you suggested template.

Result is that it triggers at each step but always the same result:

{"POWER":"ON","Dimmer":4,"Color":"10,0","White":4,"CT":153,"Channel":[4,0]}

Also notice a little buzzing on the lamp.
Manual Toggle or Slide Controls (Color or Brightness) nothing works.

I notice that your template has Rotary A_n and Rotary B_n but your commit created Rotary A and Rotary B

So i swap it to the new ones, Rotary A and Rotary B, result is that also triggers at each step but with weird results.

/RESULT = {"POWER":"ON","Dimmer":53,"Color":"51,84","White":53,"CT":369,"Channel":[20,33]}
/RESULT = {"POWER":"ON","Dimmer":56,"Color":"54,89","White":56,"CT":369,"Channel":[21,35]}
/RESULT = {"POWER":"ON","Dimmer":53,"Color":"51,84","White":53,"CT":369,"Channel":[20,33]}
/RESULT = {"POWER":"ON","Dimmer":53,"Color":"51,84","White":53,"CT":369,"Channel":[20,33]}
/RESULT = {"POWER":"ON","Dimmer":53,"Color":"51,84","White":53,"CT":369,"Channel":[20,33]}
/RESULT = {"POWER":"ON","Dimmer":53,"Color":"51,84","White":53,"CT":369,"Channel":[20,33]}
/RESULT = {"POWER":"ON","Dimmer":53,"Color":"51,84","White":53,"CT":369,"Channel":[20,33]}
/RESULT = {"POWER":"ON","Dimmer":53,"Color":"51,84","White":53,"CT":369,"Channel":[20,33]}

Then i rebased to Shelly Dimmer 2 (Rotary) Module / Generic (18) => one toggle plus one brightness slidder.
Inserted your sugested template, result is it doesn't trigger anything.

Again i've changed to Rotary A and Rotary B as i sugested before and it triggers with almost good quality:

/RESULT = {"POWER":"OFF","Dimmer":0}
/STATE = {"LoadAvg":37,"POWER":"ON","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":38,"POWER":"ON","Dimmer":30,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":39,"POWER":"ON","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":40,"POWER":"ON","Dimmer":60,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":41,"POWER":"ON","Dimmer":70,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":42,"POWER":"ON","Dimmer":90,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":43,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":44,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/RESULT = {"POWER":"ON","Dimmer":80}
/STATE = {"LoadAvg":58,"POWER":"ON","Dimmer":70,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":59,"POWER":"ON","Dimmer":80,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":60,"POWER":"ON","Dimmer":90,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":61,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/RESULT = {"POWER":"ON","Dimmer":100}
/STATE = {"LoadAvg":62,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":63,"POWER":"ON","Dimmer":80,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":65,"POWER":"ON","Dimmer":70,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":66,"POWER":"ON","Dimmer":60,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/STATE = {"LoadAvg":67,"POWER":"ON","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"OFF"}
/RESULT = {"POWER":"ON","Dimmer":50}

Another test i've done is with rules, and it seems to work just like before, just limited between 0 and 10 because i've not changed ROTARY_MAX_STEPS 10, but the step is 1.

In resume:

- If Rotary A_n and Rotary B_n no values are changed because they work with INPUT_PULLUP
- If Rotary A and Rotary B values are changed as normal INPUT

Detected problems:
DimmerStep 1 doesn't change the step, it's stuck at 10.
If i turn the knob fast some jumps of 20 or no trigger could occour.
I have to see if debounce commands exist for rotary encoder like for buttons and switches

I am at yout disposal for testing more, and for the Detected Problems also.
Thank you very much!

arendst, 2021-01-07:
Thx

> PS: Today i learned how important is to reset config between compiles.
Config was adressing Rotary A/B as Rotary A/B_n and so with opposite result

So this

> n resume:

If Rotary A and Rotary B no values are changed because they work with INPUT_PULLUP
If Rotary A_n and Rotary B_n values are changed as normal INPUT

has been solved?

BPSoft, 2021-01-07:
Before your commit, Rotary Encoder imply INPUT_PULLUP, and Shelly Dimmer 1/2 and virtually many other devices don't work with REncoder.

This behaviour is what i observe compiling your late dev commit with reset config on by behalve:

- If Rotary A and Rotary B no values are changed because they work with INPUT_PULLUP
- If Rotary A_n and Rotary B_n values are changed as normal INPUT

So after your commit we have the option to select two types of rotary encoder pinmodes.
I'm pretty much like the way you integrate, because even Mi Desk Lamp (66) or any module could test with INPUT without PULLUP

But because initially i've not reseted the config, what i was observing is the opposite:
- If Rotary A_n and Rotary B_n no values are changed because they work with INPUT_PULLUP
- If Rotary A and Rotary B values are changed as normal INPUT

This is probably due to the way config is managed, between compiles the table of sensors changed, and an incorrect assignment of gpio could occur!? Right now i don't have sufficient Tasmota knowledge to assume that.

I've just mentioned this because i've litterally tested for hours with the opposite behaviour.
But the switch bit was working anyway, it was just a question of what Rotary A is versus Rotary_A_n

What i do if i was you is the creation of a new SetOptionXXX Rotary Endoder Pin Mode 

- 0 INPUT_PULLUP (default)
- 1 INPUT

Because no other input pinmodes are possible, and Rotary A versus Rotary A_n is confuse without doc.
So only a Rotary A_n and Rotary B_n with an aditional SetOptionXXX defaulted to 0 is better.
Default 0 still does not cause disrupt for the Mi Desk Lamp (66) users.

The only limitations i found is:

- DimmerStep 1 doesn't change the step, it's stuck at 10. (but DimmerRange work)
- If i turn the knob fast, some jumps of 20+ or no trigger could occur.

arendst, 2021-01-07:
Thx. I build the hardware connection to a Shelly Dimmer 2 as documented here https://www.instructables.com/Shelly-Dimmer-Wall-Switch-With-Rotary-Knob-and-Hom/

and it works using this template:
```
{"NAME":"Shelly Dimmer 2 (Rotary)","GPIO":[32,3200,0,3232,5568,5600,0,0,6272,0,6304,0,0,4736],"FLAG":0,"BASE":66}
```
and new command ``SetOption43 30``.

arendst, 2021-01-07:
Yeah it seems very Rotary Encoder hardware dependant.

Did you do the ``SetOption43 30`` command while using the latest development version?

BPSoft, 2021-01-07:
Literally:
reset 5
`{"NAME":"Shelly Dimmer 2 (Rotary)","GPIO":[32,3200,0,3232,5568,5600,0,0,6272,0,6304,0,0,4736],"FLAG":0,"BASE":66}`
SetOption43 30

and unsusable debounce problems.

With:
`{"NAME":"Shelly Dimmer 2 (Rotary)","GPIO":[0,3200,0,3232,5568,5600,0,0,6272,0,6304,0,320,4736],"FLAG":0,"BASE":18}`
LedTable 0

18:28:54.845 /RESULT = {"POWER":"OFF","Dimmer":0}
18:28:54.944 /RESULT = {"POWER":"OFF","Dimmer":0}
18:28:55.044 /RESULT = {"POWER":"OFF","Dimmer":0}
18:28:58.238 /RESULT = {"POWER":"ON","Dimmer":20}
18:28:58.492 /RESULT = {"POWER":"ON","Dimmer":40}
18:28:58.643 /RESULT = {"POWER":"ON","Dimmer":60}
18:28:58.845 /RESULT = {"POWER":"ON","Dimmer":80}
18:28:59.045 /RESULT = {"POWER":"ON","Dimmer":100}
18:28:59.196 /RESULT = {"POWER":"ON","Dimmer":100}
18:28:59.297 /RESULT = {"POWER":"ON","Dimmer":100}
18:28:59.490 /RESULT = {"POWER":"ON","Dimmer":100}
18:28:59.692 /RESULT = {"POWER":"ON","Dimmer":90}
18:28:59.840 /RESULT = {"POWER":"ON","Dimmer":100}
18:28:59.993 /RESULT = {"POWER":"ON","Dimmer":90}
18:29:00.146 /RESULT = {"POWER":"ON","Dimmer":80}
18:29:01.839 /RESULT = {"POWER":"ON","Dimmer":70}
18:29:02.041 /RESULT = {"POWER":"ON","Dimmer":60}
18:29:02.192 /RESULT = {"POWER":"ON","Dimmer":50}
18:29:02.346 /RESULT = {"POWER":"ON","Dimmer":30}
18:29:02.447 /RESULT = {"POWER":"ON","Dimmer":10}
18:29:02.597 /RESULT = {"POWER":"OFF","Dimmer":0}

But lets not forget that if you use Base 66, the code used expect Button values that i don't have. 
Maybe thats the reason it works for you

BPSoft, 2021-01-08:
Shelly Dimmer 2 pinout is different than 1.
The [Instructable](https://www.instructables.com/Shelly-Dimmer-Wall-Switch-With-Rotary-Knob-and-Hom/l) is for Shelly Dimmer 1.

Maybe thats the reason why your BASE 18 results are bad!?

The correct pinout is this:
[image]
[source: Shelly Dimmer 2 page](https://shelly.cloud/knowledge-base/devices/shelly-dimmer-2/)

arendst, 2021-01-08:
Thx. I have the correct connections to both Gnd and GPIO0 in addition to SW1 and SW2.

[image]

The difference is probably the way the Rotary Encode is made hardware wise. I used this one: ``Rotary Encoder - EC11 - 20mm - Model: ROTENC``

Maikel-K, 2021-01-08:
I have used the ky-040 rotary encoder connected to the header connector and flashed the dimmer with ESPEASY and use the dimmer as a remote sending http commands via Rules. This is working great but I would love the Tasmota firmware to work with the Dimmer2 and the rotary encoder.

[image]

Jason2866, 2021-01-08:
@Maikel-K  Tasmota supports rotary decoder with Shelly dimmer 2.

Maikel-K, 2021-01-08:
Still in testing mode right?
The push on/off function would be nice to have too. Does that work.

Jason2866, 2021-01-08:
Sure. On off is easy, works always with Tasmota. Just a button define.

BPSoft, 2021-01-12:
Tested with a different encoder [EC11E12-15P30C-SW](https://www.tme.eu/Document/44e8c47524c4eb6c460cbc7fca5d0c53/EC11E12-15P30C-SW.pdf)

This one already have a switch.

Now the behaviour is similar to @arendst , as it only works with BASE 66 and not with BASE 18.

So his assumption that different Rotary Encoder could imply different Base Models is correct.

DimmerStep doesn't affect results
SetOption43 30, results:

01:12:06.173 RSL: RESULT = {"POWER":"ON","Dimmer":6}
01:12:06.423 RSL: RESULT = {"POWER":"ON","Dimmer":12}
01:12:06.617 RSL: RESULT = {"POWER":"ON","Dimmer":18}
01:12:06.820 RSL: RESULT = {"POWER":"ON","Dimmer":24}
01:12:06.973 RSL: RESULT = {"POWER":"ON","Dimmer":30}
01:12:07.116 RSL: RESULT = {"POWER":"ON","Dimmer":36}
01:12:07.267 RSL: RESULT = {"POWER":"ON","Dimmer":42}
01:12:07.369 RSL: RESULT = {"POWER":"ON","Dimmer":48}
01:12:07.769 RSL: RESULT = {"POWER":"ON","Dimmer":66}
01:12:07.924 RSL: RESULT = {"POWER":"ON","Dimmer":72}
01:12:08.074 RSL: RESULT = {"POWER":"ON","Dimmer":78}
01:12:08.267 RSL: RESULT = {"POWER":"ON","Dimmer":84}
01:12:08.417 RSL: RESULT = {"POWER":"ON","Dimmer":90}
01:12:08.518 RSL: RESULT = {"POWER":"ON","Dimmer":96}
01:12:08.774 RSL: RESULT = {"POWER":"ON","Dimmer":100}
01:12:08.866 RSL: RESULT = {"POWER":"ON","Dimmer":100}
01:12:08.968 RSL: RESULT = {"POWER":"ON","Dimmer":100}
01:12:09.123 RSL: RESULT = {"POWER":"ON","Dimmer":100}
01:12:12.317 RSL: RESULT = {"POWER":"ON","Dimmer":100}
01:12:12.922 RSL: RESULT = {"POWER":"ON","Dimmer":100}
01:12:17.422 RSL: RESULT = {"POWER":"ON","Dimmer":94}
01:12:17.525 RSL: RESULT = {"POWER":"ON","Dimmer":88}
01:12:17.665 RSL: RESULT = {"POWER":"ON","Dimmer":82}
01:12:17.819 RSL: RESULT = {"POWER":"ON","Dimmer":76}
01:12:17.971 RSL: RESULT = {"POWER":"ON","Dimmer":70}
01:12:18.122 RSL: RESULT = {"POWER":"ON","Dimmer":64}
01:12:18.265 RSL: RESULT = {"POWER":"ON","Dimmer":58}
01:12:18.365 RSL: RESULT = {"POWER":"ON","Dimmer":52}
01:12:18.868 RSL: RESULT = {"POWER":"ON","Dimmer":22}
01:12:19.020 RSL: RESULT = {"POWER":"ON","Dimmer":16}
01:12:19.423 RSL: RESULT = {"POWER":"OFF","Dimmer":0}
 
A little too fast on the knob and we could get super fast dim from 0 to 100, or no result change at all.

closed 2021-01-20 by arendst

chrisburrows, 2021-11-15:
Hi,

I'm considering using a Shelly Dimmer 2 & rotary encoder to replace my traditional dimmer switches which use a rotary control by pulling off the knob and replacing the dimmer module with the rotary encoder / shelly dimmer 2 so I can keep the brushed stainless steel look.

I've never use the Dimmer 2 (but I haved used 1, 1PM, 2 etc), but I'm concerned that since it's intended that the SW1 / SW2 are normally connected to either L or N via a switch, that connecting the rotary encoder between GND and SW1 / SW2 could exceed the voltage rating on the rotary encoder. I know the Shelly design generally means the voltage is floating and that the GPIOs could have a high voltage on them if reference to L or N (but 3.3V between pins on the GPIO connector). Doesn't this imply there could be a high voltage between GND and SW1/SW2? Aren't the SW1 / SW2 normally floating at around 120V (for a 240V L)?

Has anyone actually measured this voltage?

 Am I worrying unncessarily?

Thanks

Jason2866, 2021-11-15:
Shelly has it on its site https://shelly.cloud/knowledge-base/devices/shelly-dimmer-2/#projects
Why not ask there. The know best

barbudor, 2021-11-15:
Indeed I would not send mains voltage in a rotary encoder.
What this insrtuctable seems to say is that you can use the S1 and S2 as low voltage GPIO too. Quite surprising.

If you can validate this, it would be worth adding in Tasmota documentation or template page.

chrisburrows, 2021-11-16:
I have asked Shelly about this and reproduce their reply below.

```
The project that you have shared is not from our website.
We cannot suggest nor confirm if it's safe to try and achieve it.
We remain at your disposal in case of need.
```

So despite linking the article from their website which could be regarded as an endorsement of the project, they're not prepared to comment on it's safety or the electrical characteristics of the Dimmer 2 - 
```
In regards to the Shelly DImmer 2 pins, we cannot share information about them,
and yes they are different between connecting them on L N SW1.
```

I'm a big fan of the Shelly devices, but disappointed by their support response. They encourage creative uses and flashing alternative firmware, but aren't clear on the electrical details.

It looks like an empirical approach is required and actually measuring the voltage will be necessary.

barbudor, 2021-11-16:
I would more open it and try to reconstruct the schematic

BPSoft, 2021-11-16:
I don't know why we want Shelly to behave differently. 
Not when people are asking if we could connect rotary encoders to the high side of the board (L N SW).
You could connect a rotary encoder to the GPIO Header and even then we should know what we are doing.
The rotary encoder will be at ESP potencial and that should be safe, BUT if something huge happen inside the shelly, your body will be exposed to the same danger unless we know about it and earth grounded the rotary encoder and/or provide full ac isolation for your body.
Of course this is the kind of communication a company want to avoid and doesn't even have to start with.
Use Shelly modules as intended to be, the high side is the high side, the GPIO Header is for factory or rescue flash, and more important, we should not use both at the same time (unless we know what we are doing).
Yes they want makers around it, but hardware and code are so protected as it could be.

Jason2866, 2021-11-16:
@BPSoft  I disagree. The have posted on there website a modification, without a disclamer. If it is dangerous it is everything else as professional.
Even a disclaimer would not help. There has to be a strict warning. Do not connect anything not builded and designed for AC mains and remove the link to the dangerous modification / use.

BPSoft, 2021-11-16:
> The have posted on there website a modification, without a disclamer.

Yes i am well aware of the modification on their web site and i think it is criminal, not just because of the risks i mentioned before, but also because they link the modification to an instructable that belongs to a different hardware (Shelly Dimmer 1).
Shelly  Dimmer 1 and 2 doesn't share the same pinout on GPIO header.
Even then i would say is preferable to remove the link to the dangerous modification and keep it "builded and designed for AC mains" than open the "pandora" box that could potential kill someone. Then all the legal differentiation between a disclamer or a warning is unnecessary.
There is no AC DC Isolation in this Shelly Module! Any oficial interaction with GPIO shouldn't be endorsed.

kabongsteve, 2022-02-04:
@BPSoft 

Just to clarify, you say the Shelly Dimmer 2 is AC connected thru GPIO, yet you started this thread/issue around connecting a Rotary encoder to the SW inputs on the shelly dimmer 2 and linked to GND on the GPIO header. 

And you're ok with this and any potential issues?

BusinessClaes, 2023-11-18:
Sorry for waking up an old thread, but I don’t find any other place talking about this…
I have bought a rotary switch from Plejd because it fits on my wall switches, and I have managed to get the rotary working with GND/SW1/SW2 and base 66. But I don’t get the push button working… any idea why this could be? The board with the rotary encoder has resistors and capacitors as in the drawing. When pushing the button, it goes from capacitor to 1k resistor.
[image]

BusinessClaes, 2023-11-19:
Well it didn’t took long, I figured it out myself…
I had to disable the shellys internal pull-up resistor with So62 1
And then only some fine tuning and I changed GPIO0 from Button to Button_n
So everything is working now, I just have to make a nice connector for the GPIOs and then it will be ready for use. I hope to come back soon with pictures!

barbudor, 2023-11-19:
I don't know where you read that SetOption 62 would disable the internal pull-up as it's related to MQTT messages
But the '_n' on Button does disable the pull up.

I assume pin 1 goes to GND? 
Removing the pull has high chance of makig the button unreliable. You would better replace the resistor by a short.

BusinessClaes, 2023-11-19:
Yes, pin 1 in my drawing goes to GND, 2 to GPIO0 and 3-4 to SW1/SW2.

I read about SetOption 62 at the [docs for Shelly 2](https://tasmota.github.io/docs/devices/Shelly-2/) but I see now at the commands docs that it is for mqtt as you say, so the shelly2 page is wrong/outdated.

Something strange but the whole thing stopped working in front of my eyes… no magic smoke but I was at the console and saw how the dimmer messages started to slow down and then got totally unresponsive. Then no response at all to rotation or push anymore. I have measured the board and everything seems to be OK there, something might have got fried inside the shelly… I have put the shelly back to normal dimmer configuration so on/off gets controlled with sw1/sw2 but that doesn’t seem to be working either, so something might have got broken…

I have another spare shelly dimmer so I will see if I can get that one working, hopefully not frying it. I will also flash tasmota from scratch in the one I have used now and see if that gets it back on track… Do you now if disabling the internal pullups creates a risk for internal components getting fried?