arendst/Tasmota #10410: Scripting: `#define USE_BH1750` triggers `>T` every second

marcvs, 2021-01-05:
### PROBLEM DESCRIPTION

I'm using tasmota v9.2.0 with scripting (`Core/SDK Version | 2_7_4_9/2.2.2-dev(38a443e)`)

When I include support for the BH1750, my `>T` section of the script is called every second.
Problem disappears when removing `#define USE_BH1750` from my_config_override.h.

My user_config_override.h (essentially) looks like:
```
#ifndef USE_SCRIPT
#define USE_SCRIPT
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif 

#define USE_BUTTON_EVENT
#define USE_SCRIPT_SUB_COMMAND
#define USE_I2C
#define USE_BH1750                             // [I2cDriver11] Enable BH1750 sensor (I2C address 0x23 or 0x5C) (+0k5 code)
```

 
### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used: Sonoff Mini
- [ ] Tasmota binary firmware version number used: `Core/SDK Version | 2_7_4_9/2.2.2-dev(38a443e)`
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"0":"Generic"}}
{"GPIO0":{"608":"I2C SCL"},"GPIO1":{"32":"Button1"},"GPIO2":{"640":"I2C SDA"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
```
- [x] Script
```lua
  Script output here:
>D
b1=0

>b
b1=bt[0]

=>publish %topic% ButtPress:%0b1%

if upsecs%tper==0
then
    =>publish %topic% manual tper b1:%0b1%
endif

>T
=>publish %topic% actual tper b1:%0b1%
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"/licht/tuer/1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.0.34:8000/tasmota-i2cbh1750.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:05:54","StartupUTC":"2021-01-05T09:43:13","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2021-01-04T20:07:17","SaveCount":128,"SaveAddress":"F8000"}}
/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2021-01-05T09:17:18","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"353/699"}}
/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["knarz","WLAN-CEB019"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["0000800B","2805C8000100060000005A00000000000000","0000C0C8","00006000","00000000"]}}
/STATUS4 = {"StatusMEM":{"ProgramSize":602,"Free":400,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAC787","043E8001","000010CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,10"}}
/STATUS5 = {"StatusNET":{"Hostname":"/licht/tuer/1-6669","IPAddress":"192.168.0.99","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.12","Mac":"2C:F4:32:A9:BA:0D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
/STATUS6 = {"StatusMQT":{"MqttHost":"q","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A9BA0D","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
/STATUS7 = {"StatusTIM":{"UTC":"2021-01-05T09:49:07","Local":"2021-01-05T10:49:07","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:07"}}
/STATUS10 = {"StatusSNS":{"Time":"2021-01-05T10:49:07","BH1750":{"Illuminance":227}}}
/STATUS11 = {"StatusSTS":{"Time":"2021-01-05T10:49:07","Uptime":"0T00:05:54","UptimeSec":354,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"knarz","BSSId":"A4:2B:B0:DF:2B:40","Channel":11,"RSSI":32,"Signal":-84,"LinkCount":1,"Downtime":"0T00:00:34"}}}
1
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
10:51:33 CMD: Group 0, Index 1, Command "PUBLISH", Data "/licht/tuer/1 actual tper b1:0"
10:51:33 MQT: /licht/tuer/1 = actual tper b1:0
10:51:34 Script: performs "publish /licht/tuer/1 actual tper b1:0"
10:51:34 SRC: Rule
10:51:34 CMD: Group 0, Index 1, Command "PUBLISH", Data "/licht/tuer/1 actual tper b1:0"
10:51:34 MQT: /licht/tuer/1 = actual tper b1:0
10:51:35 Script: performs "publish /licht/tuer/1 actual tper b1:0"
10:51:35 SRC: Rule
10:51:35 CMD: Group 0, Index 1, Command "PUBLISH", Data "/licht/tuer/1 actual tper b1:0"
10:51:35 MQT: /licht/tuer/1 = actual tper b1:0

```

### TO REPRODUCE
- Use user_config as above
- Use script from above
- Set `teleperiod` to e.g. 20
- Observe the frequency of the messages published in `>T` section
- Compare with toggling `#define USE_BH1750`

### EXPECTED BEHAVIOUR
`>T` section called every `teleperiod` time

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
n/a

gemu2015, 2021-01-05:
the documentation is misleading.   All Items that are send as MQTT (tele) objects  arrive here but are sampled much faster to give a faster response to rules or scripts. (actually up to 100ms update rate)

this is no error, works as designed (however documentation should be clearer)

marcvs, 2021-01-06:
But then: why is the frequency that the `>T` is called different whether `#define USE_BH1750` is set vs not.

gemu2015, 2021-01-06:
to get faster access to sensor data the >T section is cycled faster when sensors are defined. Otherwise you would have to wait tele period time until you could read a sensor.

marcvs, 2021-01-06:
Alright; Got it.
Sort of makes sense. I was trying to find a section for updated sensor data in the config, but this is probably one workaround.
My workaround was doing that in the `>S` section. But I guess your insight is much deeper than mine :)

closed 2021-01-06 by marcvs