arendst/Tasmota #10448: HTTP API credential transfer insecure

gorootde, 2021-01-07:
### PROBLEM DESCRIPTION
When making HTTP calls to tasmota, the credentials need to be provided within the URL:

`http://<ip>/cm?user=<username>&password=<password>&cmnd=Power%20On`

This leads to credentials being visible / recorded in places where you don't want to have credentials: Browser history, Proxy logs, Visibility on the wire, since even with HTTPS GET parameters are not encrypted.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Any
- [x] Tasmota binary firmware version number used: 9.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Execute this command on any PC:

`curl -u user:password http://<ip>/cm\?cmnd=Status`

will result in `{"WARNING":"Need user=<username>&password=<password>"}`

### EXPECTED BEHAVIOUR
Request is executed, since credentials are provided within the Basic Auth Header of the HTTP request.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2021-01-07:
The best solution is to move the credentials into the header. Then https will encrypt it and you do not have it in logs.

arendst, 2021-01-07:
This is the result of legacy support; earlier Arduino cores did not support the generic use of userid:password in the header hence this solution.

One day, if I find the time this might be changed now that both ESP8266 and ESP32 support the "official" way of doing things.

Feel free to provide a PR.

gorootde, 2021-01-07:
Well at least this looks pretty easy to solve. I intentionally didn't touch the existing implementation, since it would break tools using the legacy implementation to authenticate.

I feel the legacy auth should be marked as deprecated somehow, and then removed at a later point in time.

PR #10451 (2021-01-08): Added support for basic auth
## Description:

fixes #10448

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ ] The code change is tested and works on Tasmota core ESP32 V.1.0.5-rc4
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass C [... cut]
closed 2021-01-08 by arendst