arendst/Tasmota #10459: Wifi disconnects and doesn't know it nor reconnect

luke-jr, 2021-01-08:
### PROBLEM DESCRIPTION
Left a new Tasmota power monitoring device in use overnight. Woke up to it not responding on network, apparently unaware it had [been?] disconnected.

From my router log:

```
Fri Jan  8 15:53:43 2021 daemon.info hostapd: wlan1-1: STA *MAC* IEEE 802.11: authenticated
Fri Jan  8 15:53:43 2021 daemon.info hostapd: wlan1-1: STA *MAC* IEEE 802.11: associated (aid 2)
Fri Jan  8 15:53:43 2021 daemon.notice hostapd: wlan1-1: AP-STA-CONNECTED *MAC*
Fri Jan  8 15:53:43 2021 daemon.info hostapd: wlan1-1: STA *MAC* WPA: pairwise key handshake completed (RSN)
(NOTE: Logs for 16:03:16 also show wlan1 - my router's uplink - had to be reconnected, which tends to cause my other wifi clients to all reconnect too for some reason)
Fri Jan  8 16:03:16 2021 daemon.notice hostapd: wlan1-1: AP-STA-DISCONNECTED *MAC*
(here I pressed the button on the device 6 times)
Fri Jan  8 16:32:52 2021 daemon.notice hostapd: Station *MAC* trying to disassociate, but it is not associated
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): TOPGREENER TGWF115PQM
- [x] Tasmota binary firmware version number used: 9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tuya-Upgrade followed by upgrade instructions (multiple step upgrade) via web interface
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:40:45 RSL: RESULT = {"NAME":"TGWF115APM","GPIO":[0,320,0,32,2720,2656,0,0,2624,321,224,0,0,0],"FLAG":0,"BASE":18}
17:40:45 RSL: RESULT = {"Module":{"0":"TGWF115APM"}}
17:40:45 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"320":"Led_i1"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"321":"Led_i2"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

n/a
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:41:33 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_*ID*","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:41:33 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:05:40","StartupUTC":"2021-01-08T16:35:53","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-01-08T03:02:41","SaveCount":39,"SaveAddress":"F5000"}}
17:41:33 RSL: STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"397/699"}}
17:41:33 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["*SSID*",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
17:41:33 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:41:33 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_*ID*-*NNNN*","IPAddress":"192.168.1.193","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"*MAC*","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
17:41:33 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-01-08T16:41:33","Local":"2021-01-08T17:41:33","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:42","Sunset":"17:11"}}
17:41:33 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:41:33 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-01-08T17:41:33","ENERGY":{"TotalStartTime":"2021-01-08T03:02:41","Total":0.027,"Yesterday":0.000,"Today":0.027,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
17:41:33 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-01-08T17:41:33","Uptime":"0T00:05:40","UptimeSec":340,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF"}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

cannot access when it occurs
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
On a router with a wifi client + AP, I think `wpa_cli reattach` will do the job

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tasmota reconnects like everything else

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
n/a

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-01-08:
Try `Setoption41 60`

luke-jr, 2021-01-08:
I'll try it, but I doubt it will help since it's not associated...

luke-jr, 2021-01-08:
Seems to have survived manual testing - will wait and see if it survives the "normal" environment

ascillato2, 2021-01-08:
https://github.com/arendst/Tasmota/issues/7770

closed 2021-01-08 by ascillato2

luke-jr, 2021-01-08:
I don't see how that issue is related at all, but I'll let you know if the ppt41 doesn't fix it.

ascillato, 2021-01-08:
If your device don't respond to your web interface but if after doing a ping to the device, you can reach the web interface (and that mqtt never disconnects) this is the same problem as the linked issues. Your router forgets its ARP table. You need to configure your router properly to solve your issue. In the linked issues and also in older issues, this has been long discussed and tested.

luke-jr, 2021-01-08:
I was not able to ping it either.

luke-jr, 2021-01-20:
Looks like the only non-reboot action is the 6 presses for `WifiConfig 2`

Is there a way to get a log out of it in that mode??

luke-jr, 2021-01-27:
Hello?

luis33338, 2021-01-27:
https://github.com/arendst/Tasmota/issues/10459#issue-782256639

ascillato, 2021-01-27:
> Looks like the only non-reboot action is the 6 presses for WifiConfig 2

`wificonfig 2` broadcasts an AccessPoint. To do that, your device reboots. 

> Is there a way to get a log out of it in that mode??

Yes, go to http://192.168.4.1/cs?

In the linked issues above and also in older issues, this has been long discussed and tested. Please, try what was discussed there and if that don't solve your issue, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)