arendst/Tasmota #10465: Light - Hassio Auto-Discovery json message will not be closed

rainu, 2021-01-08:
### PROBLEM DESCRIPTION
_The command  "SetOption19 1" publish a json message for the Home Assistant automatic discovery. This json is broken cause of missing ending curly bracked (}). This has the effect that the light will not be auto descovered by hassio. I know that this command worked somedays before. As i flashed the device (light-bulb) and onboard that into my hassio instance, the light was discovered without any trouble. After that i have updated the firmware mutliple times. So i dont realy know the version, which worked before. Becuase the corresponding mqtt message is retained, i can see the worked message of an another (equal) device._

Broken json message (1040 bytes - unformated):
```json
{
  "name": "iot-led-110",
  "cmd_t": "home/cmnd/iot-led-110/POWER",
  "stat_t": "home/tele/iot-led-110/STATE",
  "val_tpl": "{{value_json.POWER}}",
  "pl_off": "OFF",
  "pl_on": "ON",
  "pl_avail": "Online",
  "pl_not_avail": "Offline",
  "avty_t": "home/tele/iot-led-110/LWT",
  "uniq_id": "001B6E_LI_1",
  "device": {
    "identifiers": [
      "001B6E"
    ]
  },
  "bri_cmd_t": "home/cmnd/iot-led-110/Dimmer",
  "bri_stat_t": "home/tele/iot-led-110/STATE",
  "bri_scl": 100,
  "on_cmd_type": "brightness",
  "bri_val_tpl": "{{value_json.Dimmer}}",
  "rgb_cmd_t": "home/cmnd/iot-led-110/Color2",
  "rgb_stat_t": "home/tele/iot-led-110/STATE",
  "rgb_val_tpl": "{{value_json.Color.split(',')[0:3]|join(',')}}",
  "fx_cmd_t": "home/cmnd/iot-led-110/Scheme",
  "fx_stat_t": "home/tele/iot-led-110/STATE",
  "fx_val_tpl": "{{value_json.Scheme}}",
  "fx_list": [
    "0",
    "1",
    "2",
    "3",
    "4"
  ],
  "clr_temp_cmd_t": "home/cmnd/iot-led-110/CT",
  "clr_temp_stat_t": "home/tele/iot-led-110/STATE",
  "clr_temp_val_tpl": "{{value_json.CT}}",
  "whit_val_cmd_t": "home/cmnd/iot-led-110/White",
  "whit_val_stat_t": "home/tele/iot-led-110/STATE",
  "whit_val_scl": 100,
  "whit_val_tpl": "{{value_json.White}}"
```

Worked message (of an equal device):
```json
{
  "name": "iot-led-133",
  "cmd_t": "~cmnd/POWER",
  "stat_t": "~tele/STATE",
  "val_tpl": "{{value_json.POWER}}",
  "pl_off": "OFF",
  "pl_on": "ON",
  "avty_t": "~tele/LWT",
  "pl_avail": "Online",
  "pl_not_avail": "Offline",
  "uniq_id": "6506D5_LI_1",
  "device": {
    "identifiers": [
      "6506D5"
    ],
    "connections": [
      [
        "mac",
        "50:....:D5" //anonymized
      ]
[... 115 lines cut ...]
22:35:36 MQT: homeassistant/switch/001B6E_RL_8/config =  (retained)
22:35:36 MQT: homeassistant/light/001B6E_LI_8/config =  (retained)
22:35:36 MQT: homeassistant/sensor/001B6E_status/config = {"name":"iot-led-110 status","stat_t":"home/tele/iot-led-110/HASS_STATE","avty_t":"home/tele/iot-led-110/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"home/tele/iot-led-110/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"001B6E_status","dev":{"ids":["001B6E"],"name":"iot-led-110","mdl":"Bomcosy 600lm ","sw":"9.2.0(tasmota)","mf":"Tasmota"}} (retained)
22:35:36 MQT: tasmota/discovery/D8BFC0001B6E/config =  (retained)
```

```lua
    Hassio Log:

2021-01-08 22:39:34 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON 001B6E_LI_1: '{"name":"iot-led-110","stat_t":"home/tele/iot-led-110/STATE","avty_t":"home/tele/iot-led-110/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"home/cmnd/iot-led-110/POWER","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","uniq_id":"001B6E_LI_1","dev":{"ids":["001B6E"]},"bri_cmd_t":"home/cmnd/iot-led-110/Dimmer","bri_stat_t":"home/tele/iot-led-110/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","rgb_cmd_t":"home/cmnd/iot-led-110/Color2","rgb_stat_t":"home/tele/iot-led-110/STATE","rgb_val_tpl":"{{value_json.Color.split(',')[0:3]|join(',')}}","fx_cmd_t":"home/cmnd/iot-led-110/Scheme","fx_stat_t":"home/tele/iot-led-110/STATE","fx_val_tpl":"{{value_json.Scheme}}","fx_list":["0","1","2","3","4"],"clr_temp_cmd_t":"home/cmnd/iot-led-110/CT","clr_temp_stat_t":"home/tele/iot-led-110/STATE","clr_temp_val_tpl":"{{value_json.CT}}","whit_val_cmd_t":"home/cmnd/iot-led-110/White","whit_val_stat_t":"home/tele/iot-led-110/STATE","whit_val_scl":100,"whit_val_tpl":"{{value_json.White}}"'
```

### TO REPRODUCE
_Use the command "SetOption19 1" in web interface._

### EXPECTED BEHAVIOUR
_A valid json should be generated. So that the hassio instance can auto discover the light._

### SCREENSHOTS
_-_

### ADDITIONAL CONTEXT
_Let me know if you need more information._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-01-09:
Please, use instead the native integration for home assistance using `setoption19 0`

You will see your device under integrations on home assistant.

ascillato, 2021-01-09:
If you want to use the MQTT discovery instead, please follow the error message you have in your console:

```lua
22:35:36 HASS: MQTT discovery failed due to too long topic or device/friendly name. Please shorten topic and/or device/friendly name. Failed to format (1/1):
```

closed 2021-01-09 by ascillato2