arendst/Tasmota #10486: Telegram response to bot is truncated

ajithvasudevan, 2021-01-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Tasmota's Telegram response is truncated as viewed in the Telegram mobile app.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 9.2.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmotizer 1.2.1
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:55:02 CMD: Backlog Template; Module; GPIO 255
12:55:02 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:55:02 RSL: RESULT = {"Module":{"18":"Generic"}}
12:55:02 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"224":"Relay1"},"GPIO14":{"160":"Switch1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
12:55:51 CMD: rule1
12:55:51 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":115,"Free":396,"Rules":"on system#boot do backlog IPAddress1 endon on IPAddress1#data do backlog tmstate 1; tmstate 3; tmsend %value% endon"}
12:55:52 CMD: rule2
12:55:52 RSL: RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":115,"Free":396,"Rules":"on power1#state=1 do  tmsend Porch Light Turned ON endon on power1#state=0 do  tmsend Porch Light Turned OFF endon "}
12:55:54 CMD: rule3
12:55:54 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":36,"Free":475,"Rules":"on power1#state=1 do power2 on endon"}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:39:13 SRC: WebConsole from 172.20.1.2
12:39:13 CMD: Group 0, Index 1, Command "STATUS", Data "0"
12:39:13 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"timer","FriendlyName":["timer"],"Topic":"timer","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:39:13 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:03:39","StartupUTC":"2021-01-10T07:05:34","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2021-01-09T22:35:39","SaveCount":39,"SaveAddress":"3F4000"}}
12:39:13 RSL: STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2021-01-09T16:00:25","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"430/699"}}
12:39:13 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Andromeda","tsvk24"],"TelePeriod":0,"Resolution":"558180C0","SetOption":["0000A003","2805C8000100060000005A00000000000000","00000021","00006000","00000000"]}}
12:39:13 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":750,"Free":2320,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAEE596","04260081","000000C3","00000400","C000F981","00008024","00001000"],"Drivers":"1,2,4,5,6,8,9,10,13,18,19,22,24,26,27,29,30,35,37,40,45","Sensors":"1,2,4,5,6,33"}}
12:39:13 RSL: STATUS5 = {"StatusNET":{"Hostname":"porchtimer","IPAddress":"172.20.1.52","Gateway":"172.20.1.1","Subnetmask":"255.255.255.0","DNSServer":"172.20.1.1","Mac":"48:3F:DA:xx:xx:xx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:39:13 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-01-10T07:09:13","Local":"2021-01-10T12:39:13","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+05:30","Sunrise":"06:45","Sunset":"18:12"}}
12:39:13 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-01-10T12:39:13","Switch1":"ON"}}
12:39:13 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-01-10T12:39:13","Uptime":"0T00:03:39","UptimeSec":219,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Andromeda","BSSId":"00:11:32:xx:xx:xx","Channel":5,"RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
12:35:53 TGM: getUpdates
12:35:55 TGM: Response {"ok":true,"result":[]}
12:36:02 WIF: Checking connection...
12:36:02 DNS: MDNS.update
12:36:07 TGM: getUpdates
12:36:09 TGM: Response {"ok":true,"result":[]}
12:36:20 TGM: getUpdates
12:36:22 TGM: Response {"ok":true,"result":[{"update_id":75814xxxx,
"message":{"message_id":204,"from":{"id":xxxxxxxxxx,"is_bot":false,"first_name":"Ajith","username":"xxxxxxxx","language_code":"en"},"chat":{"id":xxxxxxxxxx,"first_name":"Ajith","username":"xxxxxxxx","type":"private"},"date":1610262370,"text":"Rule3"}}]}
12:36:22 TGM: Parsed update_id xxxxxxxxx, chat_id xxxxxxxxxx, text "Rule3"
12:36:22 SRC: Chat
12:36:22 CMD: Group 0, Index 3, Command "RULE", Data ""
12:36:22 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":36,"Free":475,"Rules":"on power1#state=1 do power2 on endon"}
12:36:22 TGM: sendMessage
12:36:27 WIF: Checking connection...
12:36:27 DNS: MDNS.update
12:36:34 TGM: getUpdates
12:36:35 TGM: Response {"ok":true,"result":[]}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
After building **Tasmota 9.2.0** with _Telegram_ support, set it up with _token_ and _chatid_, and confirm that messages can be sent both ways.

Setup a simple rule such as "Rule3 on power1#state=1 do power2 on endon". 

Now send the message "Rule3" from Telegram (bot) on mobile phone.

See that the response message received in the mobile after a couple seconds is truncated, like - 
```lua
{"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":36,"Free":475,"Rules":"on power1
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The response received should be complete, for example, 
```lua
{"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":36,"Free":475,"Rules":"on power1#state=1 do power2 on endon"}
```
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-01-10:
What you posted in https://discord.com/channels/479389167382691863/490244847568158751/797881187636609035 is concistent with some tool "on the way" cutting off the string at the # character, not too unusual as a delimiter. Did you check the actual MQTT message emanating from Tasmota?

ajithvasudevan, 2021-01-10:
The "on the way" tool I used, and the screenshot I provided happens to be the official Telegram app (Android). And yes, the MQTT message is received fine. I checked that and confirmed in Node-RED. I too suspected the "#" being used as a delimiter to be the cause and tried sending, from the Tasmota console, a message with hashes ( tmsend test1#test2#test3 ), and that came through in the mobile app properly. Do you want me to test anything else?

ajithvasudevan, 2021-01-10:
Findings from discussion here: https://discord.com/channels/479389167382691863/490244847568158751/797897053921345607

It seems the command -
"tmsend test1test1test1test1test1 # test22222222 #test33333333"  causes "test1test1test1test1test1" to be received in Telegram

and the command -
"tmsend test1test1test1test1test1 %23 test22222222%23test33333333" causes "tmsend test1test1test1test1test1 # test22222222 #test33333333" to be received.

In the latter case, the "#" has been URLEncoded, and the expected message is received.

This leads me to believe that the Telegram API call from the Tasmota Telegram Driver is currently not being URLEncoded, and some special characters in the query parameters of the Telegram API could cause it to lose information or fail.

arendst, 2021-01-11:
Thx for providing the needed information.

Latest development commit adds UrlEncode to any text send from Tasmota to Telegram.

Pls try.

ajithvasudevan, 2021-01-11:
> Thx for providing the needed information.
> 
> Latest development commit adds UrlEncode to any text send from Tasmota to Telegram.
> 
> Pls try.

Tried it, and it works as expected, as seen in the screen-shot below from the Telegram app on my mobile phone. No truncation!
[image]

> BTW Nice rule1 usage!!!

Thanks!

ajithvasudevan, 2021-01-11:
Closing the issue as it is fixed.

closed 2021-01-11 by ajithvasudevan

SHA-NDY, 2021-02-17:
May I ask which defines have to be added before compiling to get telegram functionality and is there a doc?

ajithvasudevan, 2021-02-18:
#define USE_TELEGRAM

SHA-NDY, 2021-02-18:
> `#define USE_TELEGRAM`
> 
> There's no official doc yet, AFAIK.  I found this  https://github.com/arendst/Tasmota/issues/8619 and then I looked through the source-code to figure it out :)
>  
> EDIT:  here's some documentation: https://github.com/arendst/Tasmota/issues/8619#issuecomment-642764152
> 

Thank you very much! Tasmota rocks!

SHA-NDY, 2021-02-18:
> `#define USE_TELEGRAM`
> 
> There's no official doc yet, AFAIK. I found this #8619 and then I looked through the source-code to figure it out :)
> 
> EDIT: here's some documentation: [#8619 (comment)](https://github.com/arendst/Tasmota/issues/8619#issuecomment-642764152)

Do you know if this works in Tasmota32 or how I can find out if it works? I'm new in the Tasmota-Universe and didn't dive deep into it by now

ajithvasudevan, 2021-02-18:
> 
> Do you know if this works in Tasmota32 or how I can find out if it works? I'm new in the Tasmota-Universe and didn't dive deep into it by now

Yes, it does. A sure way to find out is to flash it on your device and try it. I just did :)

SHA-NDY, 2021-02-18:
Thanks a lot

SHA-NDY, 2021-02-19:
Would be nice to have Telegram functionality on tasmota32-webcam builds to send pictures. I found several examples for arduino so it should be possible...where would be the right place to communicate a feature request?

ascillato, 2021-02-19:
Feature requests goes to Ideas in Discussions. Thanks.

supperka, 2021-02-23:
Hello , I can send info from tasmota to telegram, but not from telegram to tasmota.
what can be wrong? Telegram just reply my msg.  Tmstate are {"TmState":{"Send":"ON","Receive":"ON","Echo":"ON"}}

SHA-NDY, 2021-02-23:
> 
> 
> Hello , I can send info from tasmota to telegram, but not from telegram to tasmota.
> what can be wrong? Telegram just reply my msg. Tmstate are {"TmState":{"Send":"ON","Receive":"ON","Echo":"ON"}}

Which command did you send? You can send Tasmota commands like "Power1 1"

supperka, 2021-02-23:
> > Hello , I can send info from tasmota to telegram, but not from telegram to tasmota.
> > what can be wrong? Telegram just reply my msg. Tmstate are {"TmState":{"Send":"ON","Receive":"ON","Echo":"ON"}}
> 
> Which command did you send? You can send Tasmota commands like "Power1 1"

Yes I tried rule1 , power1 1 but nothing .From Tasmota command tmsend works fine without problems

SHA-NDY, 2021-02-23:
Is your ChatId correct? Perhaps Tasmota filters commands from unauthorized (unknown) ChatIDs

supperka, 2021-02-23:
> Is your ChatId correct? Perhaps Tasmota filters commands from unauthorized (unknown) ChatIDs

I tried change last numbers of my chatid and still getting messages from Tasmota so maybe is that problem. How can easily get chatid?

SHA-NDY, 2021-02-23:
Search for "IDBot" in Telegram and start a conversation with the bot. Type "/getid" to get your ChatID

supperka, 2021-02-23:
Thank you , so my chatid is correct! 🙄

SHA-NDY, 2021-02-23:
> 
> 
> Thank you , so my chatid is correct! 🙄

I just saw it. You have to disable ECHO --> {"TmState":{"Send":"ON","Receive":"ON","**Echo":"ON**"}}

supperka, 2021-06-22:
Hello, my telegram integration stopped working on all tasmota devices two days ago. Before was all fine , i didn't get any update or setting. I'm only one with this problem?
If I try tmsend xxx got send failed , I tried revoke token or reset tmstate but without success.

s-hadinger, 2021-06-22:
Any logs?

ajithvasudevan, 2021-06-23:
> Any logs?

I confirm it stopped working.
Setting weblog 4 and trying to send did not result in any logs, though:

```
09:04:13.940 MQT: stat/BELL/RESULT = {"WebLog":4}
09:04:13.025 CFG: Saved to flash at 3F9, Count 186, Bytes 4096
09:04:19.189 WIF: Checking connection...
09:04:25.695 CMD: tmsend test
09:04:25.696 SRC: WebConsole from 172.20.1.2
09:04:25.698 CMD: Group 0, Index 1, Command "TMSEND", Data "test"
09:04:25.700 TGM: sendMessage
09:04:27.098 MQT: stat/BELL/RESULT = {"TmSend":"Failed"}
09:04:40.636 WIF: Checking connection...
```

`tmToken` and `tmClhatId` are set to correct values, and `tmState`  is `{"TmState":{"Send":"ON","Receive":"OFF","Echo":"OFF"}}`

Is there something else I can try?

ajithvasudevan, 2021-06-23:
I printed a part of the response from the call to Telegram API (sendMessage) in the telegram driver, and this is what I got:

```
06:42:18.831 TGM: response = <!DOCTYPE html>
<html class="">
  <head>
    <meta charset="utf-8">
    <title>sendMessage</title>
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <meta property="description" content="">
    <meta property="og:title" content="sendMessage">
    <meta property="og:image" content="">
    <meta property="og:description" content="">
    <link rel="shortcut icon" href="/favicon.ico?4" type="image/x-icon" />

    <link href="/css/bootstrap.min.css?3" rel="stylesheet">
    
    <link href="/css/telegram.css?212" rel="stylesheet" media="screen">
    <style>
    </style>
  </head>
  <body class="preload">
    <div class="dev_page_wrap">
      <div class
06:42:18.841 RSL: stat/tasmota_74E359/RESULT = {"TmSend":"Failed"}
```

Shouldn't this be a JSON instead of HTML ?

s-hadinger, 2021-06-23:
Yes indeed, it should be JSON. Maybe the API requires an additional parameter to force JSON

ajithvasudevan, 2021-06-23:
> Yes indeed, it should be JSON. Maybe the API requires an additional parameter to force JSON

The API works if you try the same URL that is being called from Tasmota (https://api.telegram.org/bot<token>/sendMessage?chat_id=<chatId>&text=test), from a browser. So additional parameter doesn't seem to be the issue.

sfromis, 2021-06-23:
Maybe the HTML contains some error message indicating a failure before even coming to the API. I've seen such from other web services unable to execute a call I posted.

ajithvasudevan, 2021-06-23:
> Maybe the HTML contains some error message indicating a failure before even coming to the API. I've seen such from other web services unable to execute a call I posted.

I have added a screen-shot of the complete HTML in the issue I added, here: #12451

sfromis, 2021-06-23:
That indeed looks like a generic message page instead of API output. May be an error on the Telegram end.

ajithvasudevan, 2021-06-23:
> That indeed looks like a generic message page instead of API output. May be an error on the Telegram end.

The error should then appear even when calling the Telegram API GET request from the browser. However, it works fine if you hit the URL https://api.telegram.org/bot[token]/sendMessage?chat_id=[chatId]&text=test)  from a browser, or even a `curl` of the same URL. You get JSON response in both cases. As far as I can tell, Tasmota's telegram driver too calls the same URL using BearSSL lib. I verified the URL it calls. It is the same as the one which works from the browser or `curl`.

ajithvasudevan, 2021-06-23:
Let's discuss further in issue #12451