arendst/Tasmota #10493: Energy Monitoring for Blitzwolf SHP13 not working (zigbee)

drtorchwood, 2021-01-10:
### PROBLEM DESCRIPTION
My Blitzwolf SHP13 plug does not report energy measurements properly. (The on/off function works.)
The SHP13 is a Zigbee device. I use Tasmota version 9.2.0 and CC2530 on a Wemos D1 mini. The only data regarding energy in the sensor messages is: `{"ZbReceived": { ..."EnergyTotal":"0x00000000005B" ...`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini with CC2530
- [x] Tasmota binary firmware version number used: 9.2.0
  - [ ] Pre-compiled
  - [x] Self-compiled (tasmocompiler online)
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```11:57:00 CMD: Backlog Template; Module; GPIO 255
11:57:00 SRC: WebConsole from 192.168.178.22
11:57:00 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
11:57:00 SRC: Backlog
11:57:00 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
11:57:00 MQT: haus/arbeitszimmer/stat/D1-1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:57:01 SRC: Backlog
11:57:01 CMD: Group 0, Index 1, Command "MODULE", Data ""
11:57:01 MQT: haus/arbeitszimmer/stat/D1-1/RESULT = {"Module":{"18":"Generic"}}
11:57:01 SRC: Backlog
11:57:01 CMD: Group 0, Index 1, Command "GPIO", Data "255"
11:57:01 MQT: haus/arbeitszimmer/stat/D1-1/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"224":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"3584":"Zigbee Rx"},"GPIO14":{"1216":"AM2301"},"GPIO15":{"3552":"Zigbee Tx"},"GPIO16":{"0":"None"},"GPIO17":{"4768":"ADC Light"}}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```11:58:46 CMD: status 0
11:58:46 SRC: WebConsole from 192.168.178.22
11:58:46 CMD: Group 0, Index 1, Command "STATUS", Data "0"
11:58:46 MQT: haus/arbeitszimmer/stat/D1-1/STATUS = {"Status":{"Module":18,"DeviceName":"D1-1","FriendlyName":["D1-1"],"Topic":"D1-1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
11:58:46 MQT: haus/arbeitszimmer/stat/D1-1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:19:39","StartupUTC":"2021-01-10T10:39:07","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2020-05-23T23:52:12","SaveCount":101,"SaveAddress":"3F6000"}}
11:58:47 MQT: haus/arbeitszimmer/stat/D1-1/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-29T10:25:57","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"399/699"}}
11:58:47 MQT: haus/arbeitszimmer/stat/D1-1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Heremias",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["80008229","2805C8000100060000005A00000000000000","00000001","00006002","00000000"]}}
11:58:47 MQT: haus/arbeitszimmer/stat/D1-1/STATUS4 = {"StatusMEM":{"ProgramSize":638,"Free":2432,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F804587","04660001","14009CCF","410013C4","C800FDC1","00004004","00001000"],"Drivers":"1,2,3,4,6,8,9,10,18,19,22,23,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,9,10,12,31"}}
11:58:47 MQT: haus/arbeitszimmer/stat/D1-1/STATUS5 = {"StatusNET":{"Hostname":"D1-1","IPAddress":"192.168.178.59","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"EC:FA:BC:5E:C0:55","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:58:47 MQT: haus/arbeitszimmer/stat/D1-1/STATUS6 = {"StatusMQT":{"MqttHost":"boris","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5EC055","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:58:47 MQT: haus/arbeitszimmer/stat/D1-1/STATUS7 = {"StatusTIM":{"UTC":"2021-01-10T10:58:47","Local":"2021-01-10T11:58:47","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"08:41","Sunset":"17:14"}}
11:58:47 MQT: haus/arbeitszimmer/stat/D1-1/STATUS10 = {"StatusSNS":{"Time":"2021-01-10T11:58:47","ANALOG":{"Illuminance":66},"AM2301":{"Temperature":19.1,"Humidity":47.1,"DewPoint":7.5},"BME680":{"Temperature":20.0,"Humidity":41.9,"DewPoint":6.6,"Pressure":1014.6,"SeaPressure":1028.4,"Gas":879.23},"PressureUnit":"hPa","TempUnit":"C"}}
11:58:47 MQT: haus/arbeitszimmer/stat/D1-1/STATUS11 = {"StatusSTS":{"Time":"2021-01-10T11:58:47","Uptime":"0T00:19:40","UptimeSec":1180,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Heremias","BSSId":"34:81:C4:A8:9A:0A","Channel":11,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```00:00:00 CFG: Loaded from flash at 3F6, Count 101
00:00:00 QPC: Count 1
00:00:00 CFG: CR 399/699, Busy 0
00:00:00 SRC: Restart
00:00:00 Project tasmota D1-1 Version 9.2.0(tasmota)-2_7_4_9(2020-12-29T10:25:57)
00:00:00 DHT: (v5) Sensors found 1
00:00:00 I2C: BME680 found at 0x77
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Heremias Channel 11 BSSId 34:81:C4:A8:9A:0A in mode 11n as D1-1...
00:00:01 WIF: Connected
00:00:02 HTP: Web server active on D1-1 with IP address 192.168.178.59
12:01:25 RTC: UTC 2021-01-10T11:01:25, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
12:01:26 MQT: Attempting connection...
12:01:26 MQT: Connected
12:01:26 MQT: haus/arbeitszimmer/tele/D1-1/LWT = Online (retained)
12:01:26 MQT: haus/arbeitszimmer/cmnd/D1-1/POWER = 
12:01:26 MQT: Subscribe to haus/arbeitszimmer/cmnd/D1-1/#
12:01:26 MQT: Subscribe to haus/arbeitszimmer/cmnd/tasmotas/#
[... 77 lines cut ...]

### EXPECTED BEHAVIOUR
In the best case I would expect more information regarding the energy monitoring (current, energy consumption). But it would already be an improvement if the `EnergyTotal` would be a numeric value. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
As you might have seen from the output, there are some other sensors connected to the device, and two "IKEA Tradfri ON/OFF Switch" are part of the zigbee net. 

The result of `ZbStatus`:
```12:10:06 CMD: ZbStatus
12:10:06 SRC: WebConsole from 192.168.178.22
12:10:06 CMD: Group 0, Index 1, Command "ZBSTATUS", Data ""
12:10:06 MQT: haus/arbeitszimmer/stat/D1-1/RESULT = {"ZbStatus1":[{"Device":"0x5C6C","Name":"Kuehlschrank"},{"Device":"0x1AC7","Name":"Gesamt"},{"Device":"0x0AF2","Name":"Schlafzimmer"}]}
```
And the result of `ZbStatus2 0x5C6C`:
```
12:13:25 CMD: ZbStatus2 0x5C6C
12:13:25 SRC: WebConsole from 192.168.178.22
12:13:25 CMD: Group 0, Index 2, Command "ZBSTATUS", Data "0x5C6C"
12:13:25 MQT: haus/arbeitszimmer/stat/D1-1/RESULT = {"ZbStatus2":[{"Device":"0x5C6C","Name":"Kuehlschrank","IEEEAddr":"0x0000000000000000","Endpoints":[1],"Config":["O01"]}]}
```

**(Please, remember to close the issue when the problem has been addressed)**

bogorad, 2021-02-01:
Same issue here.

jeroenst, 2021-02-23:
I confirm this issue still exists in 9.3.0.1

bogorad, 2021-02-23:
As was explained to me on Discord, it's not really an issue - the hardware doesn't report power consumption, so if you need it, just write a rule that sends this as often as you need:

```ZbSend {"Device":"XXXXXXXXXXXXXXXX","Read":{"ActivePower":true}}```

E.g. when I need the monitoring, I activate the rule and get power measurements once a second. When power=0 I don't really need the zeroes.

jeroenst, 2021-02-25:
Hmmz, that means no realtime monitoring of this switch unfortunately.

closed 2021-02-28 by ascillato2

jeroenst, 2021-02-28:
Is polling by tasmota an option for this device? (once per minute i.e.)

s-hadinger, 2021-02-28:
Yes. It's the only way. You can use a group address to poll all at once

drtorchwood, 2021-02-28:
Ok, did I understand it correctly that ActivePower in @bogorad proposal returns the current consumption in W? And what does the message `"EnergyTotal":"0x00000000005B"` or `"EnergyTotal":"0x000000000AA4"` in the tele message means? Why is this not a normal integer number?

s-hadinger, 2021-02-28:
It's a 48 bits number as per zigbee specification - which would normally not fit in standard 32 bits. I agree that we can do better to make it more easily usable. I'm not sure however which unit this is.

drtorchwood, 2021-03-02:
> I'm not sure however which unit this is.

I performed a short experiment: I attached a bulb and measured the power/consumption. The `ActivePower` command displayed 34 W. After 3 hours the EnergyTotal value increased by 10. Theoretically, it should be 100 Wh (34 W * 3 h) or 0.1 kWh. Therefore, the recorded value must be divided by 100 to get the kWh value.

s-hadinger, 2021-03-02:
Fantastic. So the unit is increments of 10 Wh.

There is supposed to be attribute 0702/0006 giving the PowerFactor but it's not implemented in SHP13.