arendst/Tasmota #10499: setting ip bug

koos147, 2021-01-10:
i am trying to use the Tuya door sensor.

this device has the anoying function to shutdown after every transmit.
i wanted to set a static ip to speedup the boot proces.
however due to limited time i must do that in a single line (otherwise the device is already back to sleep.
in the manual i see the bellow part.

`Set networking IP (XXX.XXX.XXX.XXX) addresses
IPAddress1 = set device IP address
0.0.0.0 = use dynamic IP address (DHCP)
XXX.XXX.XXX.XXX = set static IP address
IPAddress2 = set gateway IP address
IPAddress3 = set subnet mask
IPAddress4 = set DNS server IP address

follow IPAddress commands with restart 1 to apply changes`

so i typed the following in notepad. go to the webconsole. wait for another wakeup and past the information bellow.
`IPAddress1 172.16.20.172
IPAddress3 255.255.255.0
IPAddress2 172.16.20.254`

my device ended with the ip 255.255.255.255 (no not 0 255) 
i removed the device since it is impossible to fix this over the network.
connected the wifi chip to a serial console.
however now it has a bootloop and don't let me enter anything.

I also blacklisted the device on the AP in case this would stop the bootloop.

So the questions for today.

1. Why not a single line of code? **IPAddress 192.168.1.5/24,192.168.1.1,192.168.1.1** should be a great option. 
2. how is it possible that the device got an invallid ip 
3. how to revert the device to dhcp (or the correct ip)

bellow the log from the serial console

`00:00:00 Project tasmota Tasmota Version 8.2.0(tasmota)-STAGE
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP-, SSId ***
****
00:00:04 WIF: Network 21, AP-, SSId ******
00:00:04 WIF: Connecting to AP1 *** Channel 3 BSSId ***** in mode 11N as Zb-Lsc08...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on Zb-Lsc08.local with IP address 255.255.255.255
00:00:05 UPP: Multicast (re)joined
00:00:08 DNS: Query done. MQTT services found 0
00:00:08 UPP: Multicast disabled
00:00:08 MQT: Attempting connection...
00:00:13 MQT: Connect failed to 172.16.20.60:65535, rc -2. Retry in 65535 sec
00:00:13 UPP: Multicast (re)joined
00:00:14 APP: Boot Count 270
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Connect failed with AP timeout
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 APP: Boot Count 270
00:00:08 CFG: Saved to flash at FB, Count 329, Bytes 4096
00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Checking connection...
00:00:09 WIF: Attempting connection...
00:00:10 WIF: Checking connection...
00:00:10 WIF: Attempting connection...
00:00:12 WIF: Checking connection...
00:00:12 WIF: Attempting connection...
00:00:13 WIF: Checking connection...
00:00:13 WIF: Connect failed with AP timeout
`

closed 2021-01-10 by close-issue-app[bot]

koos147, 2021-01-10:
what is this? the template is useless in this case. stop auto closing bugs

ascillato2, 2021-01-10:
Not a bug, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

meingraham, 2021-01-10:
@koos147 

> so i typed the following in notepad. go to the webconsole. wait for another wakeup and past the information bellow.
IPAddress1 172.16.20.172 IPAddress3 255.255.255.0 IPAddress2 172.16.20.254

🤦 

See `Backlog`

koos147, 2021-01-10:
So.we have an 3 lines of commands.
The website transforms it to one line. 
The system doesn't understand the one line (so far so good) 
Why does it accept it and brik the device? 
What about input validation? 
So still a bug.

meingraham, 2021-01-10:
I recommend you start holding your breath while waiting... or read the documentation regarding how to issue chained commands. Your choice.