arendst/Tasmota #10504: Nested IF statements not working

poggenpower, 2021-01-10:
### PROBLEM DESCRIPTION
Nested IF Statements doesn't seems to work.
Following rule 
```
ON Event#SetShutterBlock DO IF (%value% == 0) IF (%var3% != %var2%) Event ShutterMove=%var3% ENDIF;var1 %value% ELSEIF (%value% == 1) var1 %value% ELSE var1 0 ENDIF ENDON 
```
is not working as expected. The inner IF clause doesn't trigger the Event `ShutterMove`, here the log from the console:
```
17:11:12 RUL: EVENT#SETSHUTTERBLOCK performs "IF (0 == 0) IF (0.000 != 50.000) Event ShutterMove=0.000 ENDIF;var1 0 ELSEIF (0 == 1) var1 0 ELSE var1 0 ENDIF"
17:11:12 MQT: esp/esp03/stat/RESULT = {"If":"Done"}
17:11:12 MQT: esp/esp03/stat/RESULT = {"If":"Done"}
17:11:12 MQT: esp/esp03/stat/RESULT = {"Var1":"0"}
```

Simple IF clauses are working as expected. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly2.5
- [X] Tasmota binary firmware version number used: 9.1.0.2
  - [X] Self-compiled
- [X] Flashing tools used: OTA file upload
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:24:23 MQT: esp/esp03/stat/RESULT = {"NAME":"Shelly 2.5 (buttons)","GPIO":[320,0,34,0,224,129,0,0,640,128,608,225,3456,4736],"FLAG":0,"BASE":18}
17:24:23 MQT: esp/esp03/stat/RESULT = {"Module":{"0":"Shelly 2.5 (buttons)"}}
17:24:24 MQT: esp/esp03/stat/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"34":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"129":"Button_in2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"128":"Button_in1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ"},"GPIO17":{"4736":"ADC Temp"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
17:25:01 MQT: esp/esp03/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":138,"Free":373,"Rules":"ON Clock#Timer=1 DO ShutterPosition1 100 ENDON ON Clock#Timer=2 DO ShutterPosition1 100 ENDON ON Clock#Timer=3 DO ShutterPosition1 0 ENDON"}
17:25:02 MQT: esp/esp03/stat/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":464,"Free":47,"Rules":"ON Event#SetShutterBlock DO IF (%value% == 0) IF (%var3% != %var2%) Event ShutterMove=%var3% ENDIF;var1 %value% ELSEIF (%value% == 1) var1 %value% ELSE var1 0 ENDIF ENDON  ON Event#ShutterMove DO IF ((var1 == 0 ) AND (var2 != %value% )) ShutterPosition1 %value%;var2 %value%; publish esp/esp04/cmnd/event esp03ShutterMove=%value% ENDIF ENDON ON Event#SetVar1 DO var1 = %value% ENDON ON Event#SetVar2 DO var2 = %value% ENDON ON Event#SetVar3 DO var3 = %value% ENDON"}
17:25:02 MQT: esp/esp03/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:25:27 MQT: esp/esp03/stat/STATUS = {"Status":{"Module":0,"DeviceName":"WZ Rolladen","FriendlyName":["up","down"],"Topic":"esp03","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
17:25:27 MQT: esp/esp03/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"1T00:00:10","StartupUTC":"2021-01-09T16:25:17","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2020-12-03T23:57:37","SaveCount":334,"SaveAddress":"FB000"}}
17:25:27 MQT: esp/esp03/stat/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(thl-rule)","BuildDateTime":"2021-01-09T17:22:35","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"465/699"}}
17:25:27 MQT: esp/esp03/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000AC029","2805C8000100060000005A00000000000000","40000209","00206000","00000000"]}}
17:25:27 MQT: esp/esp03/stat/STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":372,"Heap":19,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FCAE5D7","04268001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,8,9,10,12,16,18,19,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:25:27 MQT: esp/esp03/stat/STATUS5 = {"StatusNET":{"Hostname":"esp03","IPAddress":"192.168.7.104","Gateway":"192.168.7.248","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.248","Mac":"48:3F:DA:82:36:99","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:25:27 MQT: esp/esp03/stat/STATUS6 = {"StatusMQT":{"MqttHost":"haus.wupp","MqttPort":8885,"MqttClientMask":"esp03","MqttClient":"esp03","MqttUser":"esp03","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:25:27 MQT: esp/esp03/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-01-10T16:25:27","Local":"2021-01-10T17:25:27","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:41","Sunset":"17:14"}}
17:25:27 MQT: esp/esp03/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:25:27 MQT: esp/esp03/stat/STATUS10 = {"StatusSNS":{"Time":"2021-01-10T17:25:27","ANALOG":{"Temperature":46.9},"ENERGY":{"TotalStartTime":"2020-12-03T23:57:37","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":100,"Direction":0,"Target":100},"TempUnit":"C"}}
17:25:27 MQT: esp/esp03/stat/STATUS11 = {"StatusSTS":{"Time":"2021-01-10T17:25:27","Uptime":"1T00:00:10","UptimeSec":86410,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"iot","BSSId":"78:8A:20:49:35:67","Channel":11,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}
17:25:27 MQT: esp/esp03/stat/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":230,"Close":200,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}
17:25:28 MQT: esp/esp03/tele/STATE = {"Time":"2021-01-10T17:25:28","Uptime":"1T00:00:11","UptimeSec":86411,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":37,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"iot","BSSId":"78:8A:20:49:35:67","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:25:28 MQT: esp/esp03/tele/SENSOR = {"Time":"2021-01-10T17:25:28","ANALOG":{"Temperature":47.0},"ENERGY":{"TotalStartTime":"2020-12-03T23:57:37","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":100,"Direction":0,"Target":100},"TempUnit":"C"}
17:25:29 MQT: esp/esp03/tele/HASS_STATE = {"Version":"9.1.0.2(thl-rule)","BuildDateTime":"2021-01-09T17:22:35","Module or Template":"Shelly 2.5 (buttons)","RestartReason":"Software/System restart","Uptime":"1T00:00:12","Hostname":"esp03","IPAddress":"192.168.7.104","RSSI":"72","Signal (dBm)":"-64","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:03","MqttCount":1,"LoadAvg":24}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
17:26:33 CMD: Event SetShutterBlock=0
17:26:33 SRC: WebConsole from 192.168.11.200
17:26:33 CMD: Group 0, Index 1, Command "EVENT", Data "SetShutterBlock=0"
17:26:33 MQT: esp/esp03/stat/RESULT = {"Event":"Done"}
[... cut ...]
 8> But not like this: Power1 off; LedPower on; IF (Mem1==0) Var1 Var1+1; Mem1 1 ENDIF; Delay 10; Power1 on
 9> 
10> You should split it in two lines like: ON Power2#state=1 DO Power1 off; LedPower on; ENDON ON 11> Power2#state=1 DO IF (Mem1==0) Var1 Var1+1; Mem1 1 ENDIF; Delay 10; Power1 on ENDON```

So in a `<statement-list>` another IF statement is allowed (line 3).
But some things are not clear to me: Line 5 and 8 gives the same rule, but one is wrong!? 

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-01-10:
Closing this issue as it is **duplicated**. Sorry.

This question has been answered before. Nested IF are not supported and will not be supported due to Flash and RAM restrictions. Please, rewrite your IF logic in the supported format.

If you need help on that, please address this to the Tasmota support chat. Thanks.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-10 by ascillato2

poggenpower, 2021-01-10:
@ascillato2 
https://tasmota.github.io/docs/Rules/#conditional-rules says it is supported:
```
Syntax

IF statement supports 3 formats:

    IF (<logical-expression>) <statement-list> ENDIF
    IF (<logical-expression>) <statement-list> ELSE <statement-list> ENDIF
    IF (<logical-expression>) <statement-list> [ELSEIF (<logical-expression>) <statement-list> ] ELSE <statement-list> ENDIF
[...]
<statement-list>
- A Tasmota command (e.g.,LedPower on)
- Another IF statement (IF ... ENDIF)
- Multiple Tasmota commands or IF statements separated by ;
[...]
```

poggenpower, 2021-01-10:
Filed https://github.com/tasmota/docs/issues/589 for the doc repo

barbudor, 2021-03-13:
https://github.com/tasmota/docs/pull/668