arendst/Tasmota #10509: [Feature  Request] Multi Button press Delay

marcohald, 2021-01-10:
### PROBLEM DESCRIPTION
Sometimes it would be better to have more time to press a button multiple times.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)

### TO REPRODUCE
Press button two slow and the action for one press is executed two times instead the action of a double press

### EXPECTED BEHAVIOUR
Maybe a SETOPTION for the timeout for the Button Multipress

### SCREENSHOTS

### ADDITIONAL CONTEXT

stefanbode, 2021-01-20:
You can try to change the code directly to check if this helps your use case. You will find the responsible line in support_button.ino line: 243

            Button.window_timer[button_index] = loops_per_second / 2;  // 0.5 second multi press window

Remove the /2 and you have 1 second to press again instead of 0.5 seconds.

ascillato2, 2021-01-20:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-20 by ascillato2

jkwim, 2023-08-27:
Apologize for digging up this old thread.

Has there been any change in this setting by way of a SetOptions? I also want to change this interval (possibly without having to edit code)

R1DEN, 2023-12-15:
@jkwim completely agree with you. For me however it's the opposite case, the default delay really bugged me and I disabled it with SetOption 13, however that disables multipress detection too...