arendst/Tasmota #10515: Can't assign components

Istvan70, 2021-01-11:
### PROBLEM DESCRIPTION
I can't set some of the components to GPIO pins.
On the Configuration - Configure Module page some of the components is not even listed as an option, but on the console page I can check it's availability:
for the command "GPIOs all" I get a huge list including "1856":"SR04 Tri/TX","1888":"SR04 Ech/RX",
But when I try to set: "GPIO12 1856" t isn't changing.
For the command "GPIO12 322" or "GPIO12 0" I can see the changing.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [y ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ y] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [y ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [y ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ y] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic):Wemos D1 mini
- [ ] Tasmota binary firmware version number used: Version 9.2.0.2(tasmota)-2_7_4(2021-01-10T21:18:50)
  - [ ] Pre-compiled
  - [ y] Self-compiled
- [ ] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:00:00.049 CFG: Loaded from flash at FB, Count 25
00:00:00.054 QPC: Count 1
00:00:00.060 Project tasmota Tasmota Version 9.2.0.2(tasmota)-2_7_4(2021-01-10T21:18:50)
00:00:00.505 WIF: Connecting to AP1 OpenWrt5_2.4G Channel 11 BSSId 3C:84:6A:44:A6:C8 in mode 11n as tasmota_5B6533-1331...
00:00:01.837 WIF: Connected
00:00:02.096 HTP: Web server active on tasmota_5B6533-1331 with IP address 192.168.1.238
11:55:30.087 MQT: Attempting connection...
11:55:31.044 MQT: Connected
11:55:31.047 MQT: tele/tasmota_5B6533/LWT = Online (retained)
11:55:31.050 MQT: cmnd/tasmota_5B6533/POWER = 
11:55:31.057 MQT: tele/tasmota_5B6533/INFO1 = {"Module":"Generic","Version":"9.2.0.2(tasmota)","FallbackTopic":"cmnd/DVES_5B6533_fb/","GroupTopic":"cmnd/tasmotas/"}
11:55:31.064 MQT: tele/tasmota_5B6533/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_5B6533-1331","IPAddress":"192.168.1.238"}
11:55:31.076 MQT: tele/tasmota_5B6533/INFO3 = {"RestartReason":"Software/System restart"}
11:55:33.444 QPC: Reset

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
GPIO12 1856

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
12:00:10.590 MQT: stat/tasmota_5B6533/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"1856":"SR04"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2021-01-11:
You have to use a binary that supports the devices related to the GPIO pins you're trying to select.

ascillato2, 2021-01-11:
Closing this duplicated issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-11 by ascillato2

Istvan70, 2021-01-11:
> You have to use a binary that supports the devices related to the GPIO pins you're trying to select.

Do you mean the pre-compiled binary? 
With that I get not supported answer...
13:27:35 CMD: GPIO12 1856
13:27:35 MQT: stat/tasmota_E0DEF8/RESULT = {"GPIO":"Not supported"}

but not where I compiled from the source.
13:45:56.536 CMD: GPIO12 1856
13:45:56.545 MQT: stat/tasmota_5B6533/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
Here I get no error message, but no success either.

Can you please help me to find the pair of my duplicated issue?