arendst/Tasmota #10516: SH1106 Text Issue - every other line of dots

scargill, 2021-01-11:
I have an SH1106 OLED I2c  display - 1.3" 128*64. I'm using Latest Tasmota-display.bin dev version 9.2.0.3

The display will not show text correctly - every alternative line seems blank. 

If I use an SSD1306 - again 128*64 - text, circles, rectangles work perfectly but the SH1106 not.

I've tried various DISPLAYMODEL including 2 and 7.

Can anyone help, Antonio (Mr Shark, Squazaloo) had the same issue with the SH1106 - but is now stuck miles away from the board thanks to Covid. Anyone with experience of this board and Tasmota please help?
[image]
[image]

The SSD1306 is shown working fine here, see SH1106 - every other line is blank

closed 2021-01-11 by close-issue-app[bot]

arendst, 2021-01-11:
Pete,

pls provide output of command ``Display``

arendst, 2021-01-11:
I've played around with some parameters but I fail to see the result like yours.

It could well be your display is broken failing to display the odd lines.

scargill, 2021-01-11:
Thanks for the quick response Theo - I normally figure out stuff like this on my own - but this has me. Ok, so to make sure neither of us uis making assumptions - tasmota-display.bin 9.2.0.3  - plugged the display in. Set displaymode 0. Displaymodel 7. Rebooted (no other flags set - this was a clean ESP8266.

Your request... "Display" output.
18:07:52.871 MQT: stat/sh1106/RESULT = {"Display":{"Model":7,"Width":128,"Height":64,"Mode":0,"Dimmer":1,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}

I've no idea why I see "Row!:2" at the end. My REFRESH is different to yours, as is "Cols" -  no idea why.

And after that if I simply type "DisplayText Hi There" I get "Hi there" with the first line there, second missing, third there etc.  So the horizontal line in "H" is missing etc.

Anything I need to change/add?

scargill, 2021-01-11:
Elsewhere you commented about a potential duff display. Depending o your response to the above, I will tackle anggood for a replacement - or not... But as Antonio (Mr Shark) in Italy suffered the same issue, possibly we're missing something?

scargill, 2021-01-11:
I figured out that rows and cols so mine is now the same as yours - exactly the same result. The only differnce is "mode" and thats because you are showing time/date wher I'm doing text. I changed mode to 1 - same result - every other line missing - but it is clearly trying to show the time and date inc seconds.

scargill, 2021-01-11:
My display IS duff - Antonio just replicated what exactly what I did and his works. I am SO sorry to take up your time.