arendst/Tasmota #10526: Device Reboots When Button Pressed Before Wi-Fi is Configured

KennyStier, 2021-01-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

A smart plug pre-configured using either a custom compiled binary or a web request cannot be toggled before the Wi-Fi is configured.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X ] Device used (e.g., Sonoff Basic): Sonoff S31, NodeMCU
- [ X ] Tasmota binary firmware version number used: 9.1
  - [ X ] Pre-compiled
  - [ X ] Self-compiled
- [ X ] Flashing tools used: esptool
- [ X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```lua
  Configuration output here:

00:02:08 CMD: Backlog Template; Module; GPIO 255
00:02:08 RSL: stat/tasmota_07E111/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:02:08 RSL: stat/tasmota_07E111/RESULT = {"Module":{"41":"Sonoff S31"}}
00:02:08 RSL: stat/tasmota_07E111/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ X ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

00:02:48 CMD: Status 0
00:02:48 RSL: stat/tasmota_07E111/STATUS = {"Status":{"Module":41,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_07E111","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:02:48 RSL: stat/tasmota_07E111/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:02:49","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":1,"BCResetTime":"1970-01-01T00:00:00","SaveCount":4,"SaveAddress":"F8000"}}
00:02:48 RSL: stat/tasmota_07E111/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2021-01-09T01:03:38","Boot":7,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"330/699"}}
00:02:48 RSL: stat/tasmota_07E111/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100068000005A00000000000000","00000000","00006000","00000000"]}}
00:02:48 RSL: stat/tasmota_07E111/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FD04783","04360001","140000CF","410013C0","C800FDC1","00004004","00000000"],"Drivers":"1,2,3,4,6,7,8,9,10,12,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
00:02:48 RSL: stat/tasmota_07E111/STATUS5 = {"StatusNET":{"Hostname":"tasmota_07E111-0273","IPAddress":"(IP unset)","Gateway":"(IP unset)","Subnetmask":"(IP unset)","DNSServer":"(IP unset)","Mac":"C8:2B:96:07:E1:11","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:02:48 RSL: stat/tasmota_07E111/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_07E111","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:02:48 RSL: stat/tasmota_07E111/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:02:48","Local":"1970-01-01T00:02:48","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:02:48 RSL: stat/tasmota_07E111/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:02:48 RSL: stat/tasmota_07E111/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:02:48","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
00:02:48 RSL: stat/tasmota_07E111/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:02:48","Uptime":"0T00:02:49","UptimeSec":169,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF"}}

```
- [ X ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

00:00:00 CFG: Loaded from flash at F5, Count 7
00:00:00 QPC: Count 1
00:00:00 CFG: CR 330/699, Busy 0
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 9.1.0(tasmota)-2_7_4_7
00:00:00 SER: Set to 8E1 4800 bit/s
00:00:00 SNS: Hardware Serial
00:00:00 DMP: F2 5A 02 D7 58 00 05 E9 00 3D 22 03 A4 0B 4C 66 98 46 65 54 61 00 00 DA
00:00:00 WIF: Checking connection...
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 WIF: Wifimanager set AccessPoint
00:00:00 HTP: Web server active on tasmota_07E111-0273 with IP address 192.168.4.1
00:00:00 DMP: F2 5A 02 D7 58 00 05 E9 00 3D 22 03 A4 0B 4C 66 98 47 FE B9 61 00 00 D9
00:00:00 DMP: F2 5A 02 D7 58 00 05 E9 00 3D 22 03 A4 0B 4C 66 98 49 98 1E 61 00 00 DA
[... 52 lines cut ...]
### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Flash Tasmota on device and configure it before connecting to Wi-Fi
2. Press button on device

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Relay inside device would toggle, plug would not restart

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Serial output of issue (includes info before the reboot)
[image]

Video of issue

https://user-images.githubusercontent.com/4749568/104277123-42e25200-5474-11eb-88d8-150524735229.mp4

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Please do not dismiss this issue as unimportant. Properly functioning plugs would be immensely helpful in the factory testing of Tasmota devices for CloudFree.

KennyStier, 2021-01-14:
Can confirm the issue also occurs in 9.2 and dev.

arendst, 2021-01-14:
It's as designed.

I know from other resellers they use a specially written driver to test wifi too during manufacturing so they do not hit your issue.

closed 2021-01-14 by ascillato2

tarontop, 2021-05-20:
Hello, I am a staff member of Athom, and we have encountered the same problem. Is there any solution to solve the factory test problem, so that users can get good products