arendst/Tasmota #10530: Sonoff devices and release 9.2.0.3

elkorleone, 2021-01-12:
### PROBLEM DESCRIPTION
I have update all devices in my lan but a couple of them cant update to 9.2.0.3. It shows 9.2.0 and program size is lower than 9.2.0.3 (see snapshop below)

### REQUESTED INFORMATION
How can I update to the last version?

### TO REPRODUCE
I use the same .bin file for all of them. First the minimal one and then the last version

### EXPECTED BEHAVIOUR

### SCREENSHOTS
https://fotos.subefotos.com/5be194b16f848a4a2e3d2133f5d61238o.png

### ADDITIONAL CONTEXT

sfromis, 2021-01-12:
Always use the .bin.gz versions for OTA upgrades

elkorleone, 2021-01-12:
But I've used the local upload

Jason2866, 2021-01-12:
Use the .bin.gz. files for local upload too.

sfromis, 2021-01-12:
Anytime you use the web UI for upgrades, it is OTA, be it via URL or file.

elkorleone, 2021-01-12:
Ah ok! I mean via file, not url

Problem fixed using the .bin.bz file!!!

closed 2021-01-12 by elkorleone