arendst/Tasmota #10541: Energy calculation resets from time to time without device restart

hacksics, 2021-01-13:
### PROBLEM DESCRIPTION
I have flashed nodemcu with tasmota and using PZEM-004T V3 for energy monitor. I've switched off energy monitor counter with SetOption72 0. However it appears that energy calculation reset time or show incorrect values time to time. 

Please see the console output below. 
- At 13:34:47 total energy was at 0.466KWh 
- At 13:39:33 (after 300 seconds) total energy is again at 0.038KWh. This means it got reset to 0 between 13:34 to 13:39

```
13:34:47 MQT: stat/enregy_mt_floor2/STATUS10 = {"StatusSNS":{"Time":"2021-01-13T13:34:47","ENERGY":{"TotalStartTime":"2021-01-13T12:14:48","Total":0.002,"Yesterday":0.000,"Today":0.002,"Power":450,"ApparentPower":528,"ReactivePower":275,"Factor":0.85,"Frequency":50,"Voltage":238,"Current":2.214}}}
13:39:33 MQT: tele/enregy_mt_floor2/STATE = {"Time":"2021-01-13T13:39:33","Uptime":"0T02:11:23","UptimeSec":7883,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Redmi_CE97","BSSId":"9C:9D:7E:E2:07:64","Channel":2,"RSSI":100,"Signal":-30,"LinkCount":1,"Downtime":"0T00:01:00"}}
13:39:33 MQT: tele/enregy_mt_floor2/SENSOR = {"Time":"2021-01-13T13:39:33","ENERGY":{"TotalStartTime":"2021-01-13T12:14:48","Total":0.038,"Yesterday":0.000,"Today":0.038,"Period":42949244,"Power":449,"ApparentPower":524,"ReactivePower":271,"Factor":0.86,"Frequency":50,"Voltage":239,"Current":2.198}}
13:44:33 MQT: tele/enregy_mt_floor2/STATE = {"Time":"2021-01-13T13:44:33","Uptime":"0T02:16:23","UptimeSec":8183,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Redmi_CE97","BSSId":"9C:9D:7E:E2:07:64","Channel":2,"RSSI":100,"Signal":-30,"LinkCount":1,"Downtime":"0T00:01:00"}}
13:44:33 MQT: tele/enregy_mt_floor2/SENSOR = {"Time":"2021-01-13T13:44:33","ENERGY":{"TotalStartTime":"2021-01-13T12:14:48","Total":0.075,"Yesterday":0.000,"Today":0.075,"Period":37,"Power":431,"ApparentPower":500,"ReactivePower":254,"Factor":0.86,"Frequency":50,"Voltage":238,"Current":2.101}}
13:46:15 MQT: stat/enregy_mt_floor2/STATUS10 = {"StatusSNS":{"Time":"2021-01-13T13:46:15","ENERGY":{"TotalStartTime":"2021-01-13T12:14:48","Total":0.087,"Yesterday":0.000,"Today":0.087,"Power":434,"ApparentPower":505,"ReactivePower":258,"Factor":0.86,"Frequency":50,"Voltage":238,"Current":2.125}}}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Node MCU
- [x ] Tasmota binary firmware version number used: 9.2.0 (Sensors)
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
13:58:14 CMD: Status 0
13:58:14 MQT: stat/enregy_mt_floor2/STATUS = {"Status":{"Module":18,"DeviceName":"Energy Meter Floor 2","FriendlyName":["Energy Meter Floor 2"],"Topic":"enregy_mt_floor2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:58:14 MQT: stat/enregy_mt_floor2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T02:30:04","StartupUTC":"2021-01-13T05:58:10","Sleep":50,"CfgHolder":4617,"BootCount":28,"BCResetTime":"2021-01-12T09:20:42","SaveCount":632,"SaveAddress":"F4000"}}
13:58:14 MQT: stat/enregy_mt_floor2/STATUS2 = {"StatusFWR":{"Version":"9.2.0(sensors)","BuildDateTime":"2020-12-21T15:04:56","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"423/699"}}
13:58:14 MQT: stat/enregy_mt_floor2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Redmi_CE97",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
13:58:14 MQT: stat/enregy_mt_floor2/STATUS4 = {"StatusMEM":{"ProgramSize":638,"Free":364,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
13:58:14 MQT: stat/enregy_mt_floor2/STATUS5 = {"StatusNET":{"Hostname":"enregy_mt_floor2-8108","IPAddress":"192.168.2.31","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"CC:50:E3:08:DF:AC","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:58:14 MQT: stat/enregy_mt_floor2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.138","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_08DFAC","MqttUser":"user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:58:14 MQT: stat/enregy_mt_floor2/STATUS7 = {"StatusTIM":{"UTC":"2021-01-13T08:28:14","Local":"2021-01-13T13:58:14","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+05:30","Sunrise":"13:09","Sunset":"21:48"}}
13:58:14 MQT: stat/enregy_mt_floor2/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:58:14 MQT: stat/enregy_mt_floor2/STATUS10 = {"StatusSNS":{"Time":"2021-01-13T13:58:14","ENERGY":{"TotalStartTime":"2021-01-13T12:14:48","Total":0.176,"Yesterday":0.000,"Today":0.176,"Power":451,"ApparentPower":524,"ReactivePower":266,"Factor":0.86,"Frequency":50,"Voltage":237,"Current":2.214}}}
13:58:14 MQT: stat/enregy_mt_floor2/STATUS11 = {"StatusSTS":{"Time":"2021-01-13T13:58:14","Uptime":"0T02:30:04","UptimeSec":9004,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Redmi_CE97","BSSId":"9C:9D:7E:E2:07:64","Channel":2,"RSSI":100,"Signal":-30,"LinkCount":1,"Downtime":"0T00:01:00"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-01-13:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

hacksics, 2021-01-13:
Sorry @ascillato2  for the trouble. I've gathered all the information I could at the time when I was able to reproduce the issue. I'm not using TX/RX and currently PZEM is connected with software serial emulation as below:
GPIO4 -> PZEM16 RX
GPIO5 -> PZEM0XX TX

The reason I choose GPIO pins because I wanted to make 3P power monitor with 3 PZEM devices and utilise all GPIO pins available.

ascillato, 2021-01-13:
How are you powering the 5V to your PZEM? The totals (as you have configured) are the ones from the PZEM and not the ones from Tasmota. So your PZEM is informing those. Seems that the 5V power is insufficient.

For testing purposes, please move to hardware serial and add a better power source for the 5V of the PZEM.

Anyway, If you use the Totals from Tasmota, you shouldn't see that, but the problem in your PZEM of losing totals is not normal.

hacksics, 2021-01-13:
I'm powering the PZEM from NodeMCU 5v out. NodeMCU is powered by USB cable and USB Adapter is 5V/1000mA adaptor. How do you say that I'm using the total from PZEM? I've already set the option "SetOption72 0" so tasmota will use the software calculation.

ascillato, 2021-01-13:
> "SetOption72 0" so tasmota will use the software calculation.

Oh, you are right. Sorry. I saw it wrongly in your description. 

So, today's totals are resetted only if your device is restarting. From your lasts logs, seems that is not resetting (the uptime is increasing).

hacksics, 2021-01-13:
> So, Tasmota today's totals are resetted only if your device is restarting. From your lasts logs, seems that is not resetting (the uptime is increasing).

Correct. It looks like a bug isn't it? 

> Please, try using Hardware Serial just for testing.

I will try this out. Can you tell me the connection mapping? Will it be PZEM TX to NodeMCU RX and PZEM RX to NodeMCU TX. And in Tasmota, GPIO3 -> PZEM0XX TX and GPIO 1 -> PZEM16 RX

hacksics, 2021-01-13:
Also is this the source code for tasmota energy calculation?
[xdrv_03_energy.ino](https://github.com/arendst/Tasmota/blob/fb2ff5b8691b48a5f7c7d16ed8633cba117ccfb3/tasmota/xdrv_03_energy.ino)

ascillato, 2021-01-13:
> It looks like a bug isn't it?

I really don't know. I can't reproduce it. That is why I asked you if you can test Hardware Serial.

> this the source code for tasmota energy calculation xdrv_03_energy.ino

Yes, exactly.

hacksics, 2021-01-14:
I've done some digging on this. Looks like this is a bug. Here is what I did.

To make sure that tasmota gets the correct total energy from PZEM, I added debug entry to function EnergyUpdateTotal in xdrv_03_energy.ino 
```
void EnergyUpdateTotal(float value, bool kwh)
{
  char energy_total_chr[FLOATSZ];
  dtostrfd(value, 4, energy_total_chr);
  uint32_t multiplier = (kwh) ? 100000 : 100;  // kWh or Wh to deca milli Wh

  if (0 == Energy.start_energy || (value < Energy.start_energy)) {
    Energy.start_energy = value;  // Init after restart and handle roll-over if any
    AddLog_P(LOG_LEVEL_INFO, PSTR("NRG: Setting start_energy: %d"), Energy.start_energy);
  }
  else if (value != Energy.start_energy) {
    Energy.kWhtoday = (unsigned long)((value - Energy.start_energy) * multiplier);
  }

  AddLog_P(LOG_LEVEL_INFO, PSTR("NRG: PZEM: %s %sWh, start_energy: %d, kWhtoday: %d, kWhtoday_offset: %d"), energy_total_chr, (kwh) ? "k" : "", Energy.start_energy, Energy.kWhtoday, Energy.kWhtoday_offset);

```
EnergyUpdateTotal function is called by the PzemEvery250ms function in xnrg_03_pzem004t.ino with Pzem.energy

Here is my log output -- see my inline comments for more details 

```
// at this point tasmota reporting correct today energy which is 0.902 + 0.691
// you can see it in the SENSOR message as "Total":1.593
Jan 14 16:42:28 tasmota_08DFAC-8108 ESP-NRG: PZEM: 26652.0000 Wh, start_energy: 0, kWhtoday: 1087972736, kWhtoday_offset: 90200
Jan 14 16:42:28 tasmota_08DFAC-8108 ESP-NRG: kWhtoday:  90200,  kWhtoday_delta: 0, kWhtoday_offset: 69100
Jan 14 16:42:36 tasmota_08DFAC-8108 ESP-MQT: tele/tasmota_08DFAC/STATE = {"Time":"2021-01-14T12:12:35","Uptime":"0T02:02:01","UptimeSec":7321,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Red
mi_CE97","BSSId":"9C:9D:7E:E2:07:64","Channel":2,"RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:03"}}
Jan 14 16:42:36 tasmota_08DFAC-8108 ESP-MQT: tele/tasmota_08DFAC/SENSOR = {"Time":"2021-01-14T12:12:35","ENERGY":{"TotalStartTime":"2021-01-14T07:19:55","Total":1.593,"Yesterday":0.000,"Today":1.593,"Period":1,"Power":0,"ApparentPower":0,
"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}

// but interestingly here start_energy sets to 0 by tasmota. why? 
// you can see there PZEM energy continue to increase and also tasmota didn't experience restart
// and bingo! today energy counter reset to 0 and same reflected in the SENSOR message as "Today":0.694
Jan 14 16:42:36 tasmota_08DFAC-8108 ESP-NRG: Setting start_energy: 0
Jan 14 16:42:36 tasmota_08DFAC-8108 ESP-NRG: PZEM: 26655.0000 Wh, start_energy: 0, kWhtoday: 1088030656, kWhtoday_offset: 90200
Jan 14 16:42:36 tasmota_08DFAC-8108 ESP-NRG: kWhtoday:  90200,  kWhtoday_delta: 0, kWhtoday_offset: 69100

Jan 14 16:42:38 tasmota_08DFAC-8108 ESP-NRG: PZEM: 26656.0000 Wh, start_energy: 0, kWhtoday: 1088030656, kWhtoday_offset: 100
Jan 14 16:42:38 tasmota_08DFAC-8108 ESP-NRG: kWhtoday:  100,  kWhtoday_delta: 0, kWhtoday_offset: 69100
Jan 14 16:42:38 tasmota_08DFAC-8108 ESP-NRG: energy_kWhtoday:  69200,  daily: 0, total: 536870912
Jan 14 16:42:38 tasmota_08DFAC-8108 ESP-NRG: last_return_kWhtotal:  0
Jan 14 16:42:41 tasmota_08DFAC-8108 ESP-NRG: PZEM: 26657.0000 Wh, start_energy: 0, kWhtoday: 1088030656, kWhtoday_offset: 200
Jan 14 16:42:41 tasmota_08DFAC-8108 ESP-NRG: kWhtoday:  200,  kWhtoday_delta: 0, kWhtoday_offset: 69100
Jan 14 16:42:41 tasmota_08DFAC-8108 ESP-NRG: energy_kWhtoday:  69300,  daily: 0, total: 1610612736
Jan 14 16:42:41 tasmota_08DFAC-8108 ESP-NRG: last_return_kWhtotal:  0
Jan 14 16:42:43 tasmota_08DFAC-8108 ESP-NRG: PZEM: 26658.0000 Wh, start_energy: 0, kWhtoday: 1088030656, kWhtoday_offset: 300
Jan 14 16:42:43 tasmota_08DFAC-8108 ESP-NRG: kWhtoday:  300,  kWhtoday_delta: 0, kWhtoday_offset: 69100
Jan 14 16:42:43 tasmota_08DFAC-8108 ESP-NRG: energy_kWhtoday:  69400,  daily: 0, total: -2147483648
Jan 14 16:42:43 tasmota_08DFAC-8108 ESP-NRG: last_return_kWhtotal:  0
Jan 14 16:42:46 tasmota_08DFAC-8108 ESP-MQT: tele/tasmota_08DFAC/STATE = {"Time":"2021-01-14T12:12:45","Uptime":"0T02:02:11","UptimeSec":7331,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Redmi_CE97","BSSId":"9C:9D:7E:E2:07:64","Channel":2,"RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:03"}}
Jan 14 16:42:46 tasmota_08DFAC-8108 ESP-MQT: tele/tasmota_08DFAC/SENSOR = {"Time":"2021-01-14T12:12:45","ENERGY":{"TotalStartTime":"2021-01-14T07:19:55","Total":0.694,"Yesterday":0.000,"Today":0.694,"Period":42948776,"Power":1391,"ApparentPower":1411,"ReactivePower":234,"Factor":0.99,"Frequency":50,"Voltage":238,"Current":5.928}}
Jan 14 16:42:46 tasmota_08DFAC-8108 ESP-NRG: PZEM: 26659.0000 Wh, start_energy: 0, kWhtoday: 1088030656, kWhtoday_offset: 400
Jan 14 16:42:46 tasmota_08DFAC-8108 ESP-NRG: kWhtoday:  400,  kWhtoday_delta: 0, kWhtoday_offset: 69100
Jan 14 16:42:46 tasmota_08DFAC-8108 ESP-NRG: energy_kWhtoday:  69500,  daily: 0, total: -1610612736
Jan 14 16:42:46 tasmota_08DFAC-8108 ESP-NRG: last_return_kWhtotal:  0
```

So somehow Energy.start_energy is set to 0 or value < Energy.start_energy but I don't understand how value could be lower than start_energy as PZEM reported energy keeps increasing. I'm going to add more debugging and see if I can find where the issue is. But for now I believe this doesn't have to do anything with TX and RX pins of NodeMCU not being used as PZEM appears to be reporting the correct energy values

hacksics, 2021-01-15:
An update on this:

I've commented out the Energy.start_energy = 0; as below and running the new build without energy being reset randomly.  But I don't know any other consequence of not resetting start_energy when ENERGY_WATCHDOG times out. I looked through the code but I didn't see any reason to reset the start_energy to 0. 

```
  if (!data_valid) {
    //Energy.start_energy = 0;
    AddLog_P(LOG_LEVEL_INFO, PSTR("NRG: start_energy reset to 0"));
    XnrgCall(FUNC_ENERGY_RESET);
  }
```

arendst, 2021-01-15:
If communication is lost between the PZEM and Tasmota for a prolonged time users requested to see a value of zero here. In the past I left it as it was but then they complained that they couldn't detect a broken PZEM or diconnected.

So it seems your comms between the PZEM and Tasmota fails sometimes for a number of retries and therefor resets total energy.

hacksics, 2021-01-15:
Thanks now it's clear.. 

I've tested this with ENERGY_WATCHDOG = 9 (10 seconds) but I can still see data_valid = 0 entries in my log. If this is for the purpose of identifying broken/disconnected PZEM only, is there any downside if we increase the ENERGY_WATCHDOG to higher value, say 300 seconds?

arendst, 2021-01-15:
Give it a try and let us know what happens

ascillato, 2021-01-15:
> So it seems your comms between the PZEM and Tasmota fails sometimes for a number of retries and therefor resets total energy.

We should add an entry in the console log when this reset is applied in order to help the troubleshooting. What do you think?

hacksics, 2021-01-15:
I will check this on Sunday and let you guys know. I’m not at home during the weekend 

I agree that it would be nice to print this as console log error.

ascillato2, 2021-01-15:
Theo have added the console log error message.

Please try it in latest version from development branch.

Thanks.

closed 2021-01-15 by ascillato2

hacksics, 2021-01-15:
Can the message be INFO or ERROR so it’s visible in lower log level ?

arendst, 2021-01-15:
Enable higher log for now and see if it occurs.

hacksics, 2021-01-18:
```
#grep "Energy reset by" /var/log/syslog*
/var/log/syslog:Jan 18 02:09:34 tasmota_08DFAC-8108 ESP-NRG: Energy reset by 4 seconds invalid data
/var/log/syslog:Jan 18 04:21:06 tasmota_08DFAC-8108 ESP-NRG: Energy reset by 4 seconds invalid data
/var/log/syslog:Jan 18 06:32:37 tasmota_08DFAC-8108 ESP-NRG: Energy reset by 4 seconds invalid data
/var/log/syslog:Jan 18 06:46:56 tasmota_08DFAC-8108 ESP-NRG: Energy reset by 4 seconds invalid data
/var/log/syslog.1:Jan 17 18:16:24 tasmota_08DFAC-8108 ESP-NRG: Energy reset by 4 seconds invalid data
/var/log/syslog.1:Jan 17 19:20:39 tasmota_08DFAC-8108 ESP-NRG: Energy reset by 4 seconds invalid data
/var/log/syslog.1:Jan 17 21:32:11 tasmota_08DFAC-8108 ESP-NRG: Energy reset by 4 seconds invalid data
/var/log/syslog.1:Jan 17 23:43:43 tasmota_08DFAC-8108 ESP-NRG: Energy reset by 4 seconds invalid data
/var/log/syslog.1:Jan 17 23:58:02 tasmota_08DFAC-8108 ESP-NRG: Energy reset by 4 seconds invalid data
```