arendst/Tasmota #10548: ZigBee not working when WiFi has no internet access

werner-j, 2021-01-13:
### PROBLEM DESCRIPTION
Hello,

I've noticed on Tasmota 9.2.0.3 (for prior versions I can't tell) that Tasmota does not start ZigBee if the Wifi network is not allowed to access the internet. The console shows:

`00:00:16.254 HTP: Web server active on tasmota_E2AE87-3719 with IP address 192.168.8.3 00:00:16.470 RSL: tele/tasmota_E2AE87/RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2}} 00:00:16.525 RSL: tele/tasmota_E2AE87/RESULT = {"ZbState":{"Status":55,"Version":"6.7.6.0","Protocol":8,"Stack":2}} 00:00:16.526 RSL: tele/tasmota_E2AE87/RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}} 00:00:18.623 ZIG: timeout, goto label 99 00:00:18.625 RSL: tele/tasmota_E2AE87/RESULT = {"ZbState":{"Status":99,"Message":"Abort"}} 00:00:18.626 ZIG: Abort 00:00:18.626 ZIG: Stopping (99) 00:00:19.636 RSL: tele/tasmota_E2AE87/INFO1 = {"Module":"Sonoff ZbBridge","Version":"9.2.0.3(zbbridge)","FallbackTopic":"cmnd/DVES_E2AE87_fb/","GroupTopic":"cmnd/tasmotas/"} 00:00:19.637 RSL: tele/tasmota_E2AE87/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_E2AE87-3719","IPAddress":"192.168.8.3"} 00:00:19.638 RSL: tele/tasmota_E2AE87/INFO3 = {"RestartReason":"Software/System restart"} 00:00:23.665 RSL: tele/tasmota_E2AE87/STATE = {"Time":"1970-01-01T00:00:23","Uptime":"0T00:00:25","UptimeSec":25,"Vcc":3.466,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":625,"MqttCount":0,"Wifi":{"AP":1,"SSId":"SMART_HOME","BSSId":"2A:E8:22:A7:32:18","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:17"}} 00:02:49.280 CMD: ZbPermitJoin 00:02:49.284 RSL: stat/tasmota_E2AE87/RESULT = {"ZbPermitJoin":"Zigbee not started"}`

This is really inconvenient, as my whole smart home is non-internet based and on a fully separate network. It would be nice if that issue could be fixed.

Thanks and best!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff ZigBee
- [X] Tasmota binary firmware version number used: 9.2.0.3
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Came on device
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
00:08:50.822 CMD: Backlog Template; Module; GPIO 255
00:08:50.835 RSL: stat/tasmota_E2AE87/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:08:51.071 RSL: stat/tasmota_E2AE87/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
00:08:51.324 RSL: stat/tasmota_E2AE87/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:09:55.032 CMD: Status 0
00:09:55.036 RSL: stat/tasmota_E2AE87/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_E2AE87","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:09:55.038 RSL: stat/tasmota_E2AE87/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:56","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"1970-01-01T00:00:00","SaveCount":7,"SaveAddress":"F7000"}}
00:09:55.041 RSL: stat/tasmota_E2AE87/STATUS2 = {"StatusFWR":{"Version":"9.2.0.3(zbbridge)","BuildDateTime":"2021-01-13T13:54:39","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"365/699"}}
00:09:55.044 RSL: stat/tasmota_E2AE87/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Intranet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
00:09:55.049 RSL: stat/tasmota_E2AE87/STATUS4 = {"StatusMEM":{"ProgramSize":570,"Free":1220,"Heap":28,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
00:09:55.052 RSL: stat/tasmota_E2AE87/STATUS5 = {"StatusNET":{"Hostname":"tasmota_E2AE87-3719","IPAddress":"192.168.8.3","Gateway":"192.168.0.1","Subnetmask":"255.255.240.0","DNSServer":"192.168.0.3","Mac":"98:F4:AB:E2:AE:87","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:09:55.055 RSL: stat/tasmota_E2AE87/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E2AE87","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:09:55.059 RSL: stat/tasmota_E2AE87/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:09:55","Local":"1970-01-01T00:09:55","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:09:55.061 RSL: stat/tasmota_E2AE87/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:09:55"}}
00:09:55.065 RSL: stat/tasmota_E2AE87/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:09:55","Uptime":"0T00:09:56","UptimeSec":596,"Vcc":3.466,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"SMART_HOME","BSSId":"2A:E8:22:A7:32:18","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:17"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Start Device

### EXPECTED BEHAVIOUR
Device should be able to work with ZigBee devices.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-01-13:
Closing this issue as it is **duplicated**. Sorry.

Tasmota needs a valid NTP server. Can be a local server but needs a valid time to work. 

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-13 by ascillato2

s-hadinger, 2021-01-13:
Although it is a VERY bad idea not to have time set up in Tasmota, I don't see why the Zigbee MCU wouldn't start. I believe you have another uncorrelated issue.

That said, nobody will spend time trying to make Tasmota work without time configured.