arendst/Tasmota #10553: 9.2.0 - zigbee stack in Sonoff ZBBRIDGE

moodyblue, 2021-01-14:
### PROBLEM DESCRIPTION
zigbee stack aborts on boot with error message `zig: timeout, goto label 99`. All was ok in 9.1.0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff ZBBRIDGS
- [ x] Tasmota binary firmware version number used: 9.2.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: Tasmotizer 1.2
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-01-14:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4)
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

moodyblue, 2021-01-14:
Updated first post, I hope it's ok.

Jason2866, 2021-01-14:
Please configure NTP so the device gets correct time. Most functions in Tasmota do not work without correct time.

ascillato2, 2021-01-14:
Thanks for filling the template. 

Issue duplicated of https://github.com/arendst/Tasmota/issues/10548

Please, allow your device to sync time with internet or provide a time source for it and try again. Thanks

moodyblue, 2021-01-14:
I have setup my own NTP server (using OpenWrt) and now it's Ok. I don't want my IoT devices to have access to the internet.

Thanks a lot.

closed 2021-01-14 by moodyblue

spcqike, 2021-02-17:
just a thing i noticed with the newer versions of tasmota
although I have entered a local NTP server in the DHCP settings, Tasmota does not seem to use it (anymore). Since the internet traffic is blocked in the IoT network, there are problems with new devices. You have to add the NTP server manually if the devices stay long enough in the WLAN.

ascillato, 2021-02-18:
Hi, sorry that is not a new thing in Tasmota. If you use DHCP,  Tasmota uses the NTP assigned by your router in the DHCP package if your router supports it. If you set a fixed IP (using command IPADDRESS1 X.X.X.X) Tasmota will use your NTP settings.