arendst/Tasmota #10556: software crash in shutter mode with stepper motor

J-Mau-5, 2021-01-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

when operating tasmota in shutter mode with stepper motor, it randomly crashes on hitting the up/down buttons for the shutter in web interface. The stepper motor then makes a low frequency humming noise and spins very slowly, at the same time webinterface does not react and does not print console output. In most cases tasmota restarts automatically after some seconds and comes up operational but i also observed situations where the full tasmota user setup was deleted after the crash.
As mentioned the software crash randomly appers on normal shutter operation like hitting the up/down buttons but it can quickly be provoked by hitting just one button (e.g. up) repeatedly in a very short interval.

to resolve the problem i tried the following:
-reflashed tasmota
-did the whole setup again
-used an different esp8266 board

for the tasmota setup i followed this guide:
https://tasmota.github.io/docs/Blinds-and-Shutters/
in particular:
- set up template according to Diagram v411 in the guide
- further setup via console according to section "using Stepper Motors" in the guide:
a) Backlog PulseTime1 0; PulseTime2 0
    Interlock OFF
b) SetOption80 1
c) ShutterRelay1 1
d) ShutterFrequency 1500
e) ShutterMotorDelay1 1.0

Hardware used: A4988 Stepper Driver, NEMA17 Motor, lab power supply, evering wired up according to Diagram v412

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Geekreit ESP8266 dev board (ESP12F DEVKIT V3)
- [ ] Tasmota binary firmware version number used: v9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
09:09:48 RSL: stat/tasmota_A90636/RESULT = {"NAME":"Generic","GPIO":[416,1,352,1,225,256,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:09:48 RSL: stat/tasmota_A90636/RESULT = {"Module":{"0":"Generic"}}
09:09:48 RSL: stat/tasmota_A90636/RESULT = {"GPIO0":{"416":"PWM1"},"GPIO1":{"0":"None"},"GPIO2":{"352":"Counter1"},"GPIO3":{"0":"None"},"GPIO4":{"225":"Relay2"},"GPIO5":{"256":"Relay_i1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:09:04 RSL: stat/tasmota_A90636/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3"],"Topic":"tasmota_A90636","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:09:04 RSL: stat/tasmota_A90636/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:12:01","StartupUTC":"2021-01-14T07:57:03","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2021-01-14T08:34:57","SaveCount":43,"SaveAddress":"F9000"}}
09:09:04 RSL: stat/tasmota_A90636/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-17T10:05:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"349/699"}}
09:09:04 RSL: stat/tasmota_A90636/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Skynet2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","40000000","00006000","00000000"]}}
09:09:04 RSL: stat/tasmota_A90636/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
09:09:04 RSL: stat/tasmota_A90636/STATUS5 = {"StatusNET":{"Hostname":"tasmota_A90636-1590","IPAddress":"192.168.1.150","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:CC:A8:A9:06:36","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:09:04 RSL: stat/tasmota_A90636/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A90636","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:09:04 RSL: stat/tasmota_A90636/STATUS7 = {"StatusTIM":{"UTC":"2021-01-14T08:09:04","Local":"2021-01-14T09:09:04","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:38","Sunset":"17:19"}}
09:09:04 RSL: stat/tasmota_A90636/STATUS10 = {"StatusSNS":{"Time":"2021-01-14T09:09:04","COUNTER":{"C1":14886},"Shutter1":{"Position":0,"Direction":0,"Target":0}}}
09:09:04 RSL: stat/tasmota_A90636/STATUS11 = {"StatusSTS":{"Time":"2021-01-14T09:09:04","Uptime":"0T00:12:01","UptimeSec":721,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Skynet2G","BSSId":"E0:63:DA:81:D5:24","Channel":11,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:19"}}}
09:09:04 RSL: stat/tasmota_A90636/STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"Exception","EPC":["4000dce5","00000000","40000f02"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4020e318","4025df7a","40247f06","4026d4d0","40248b06","4026d4dd","4026d522","40245084","40000f49","40000f49","40000f49","40000e19","40105a61","40105a67","4010000d","40265f78","40265f29","40104761","401058a9","40104761","4010548e","40104ddd","4010045e","401013c9","4010045e","401013c9","401013b1","401012f0","4000050c","4024bb20","401012f0"]}}
09:09:04 RSL: stat/tasmota_A90636/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":20,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
software crashes and restarts. so console output in issue state not possible

```

### TO REPRODUCE
- Setup tasmota to shutter mode using stepper motor
As mentioned the software crash randomly appers on normal shutter operation like hitting the up/down buttons but it can quickly be provoked by hitting just one button (e.g. up) repeatedly in a very short interval.

### EXPECTED BEHAVIOUR
- i expect tasmota to run smoothly in shutter mode, servicing the shutter without crashing randomly when pushing shutter-up and -down buttons.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

TrDA-hab, 2021-01-15:
I've tested this question.

Indeed, if you quickly press the up / down button many times, then Tasmota reboots.

I don't know why I might need to "quickly press the up / down button many times."

This is BUT normal work for the Shutter. I don't see any problem or error here.

J-Mau-5, 2021-01-15:
> 
> 
> I've tested this question.
> 
> Indeed, if you quickly press the up / down button many times, then Tasmota reboots.
> 
> I don't know why I might need to "quickly press the up / down button many times."
> 
> **This is not normal work for the Shutter**. I don't see any problem or error here.

you are right, it would be ok for me if "normal" shutter operation without repeatedly pressing the button would work.
But unfortunately the software crash even occures from time to time when i handle the button carefully and wait till the shutter has reached its position.

stefanbode, 2021-01-20:
Maybe I can find out what causes the crash. I agree that a crash is always a bad choice to react on something un-normal. I assume there is a problem with the RAMP function that is required to operate with stepper motors. Let me investigate later today on my test bench.

stefanbode, 2021-01-20:
For a first fix/workaround to have your shutter operating again: Please define a DEBOUNCE of 1 second on your buttons. 
ButtonDebounce 1000

stefanbode, 2021-01-21:
I now have my testbench up and running. Pressing the "up" button several times will send at the end following command in a short time frame to the shutter: up,stop,up,stop,up,stop,up,stop.
Until now I cannot produce a crash. Anyhow this is very bad behavior hammering the buttons :-)

stefanbode, 2021-01-21:
After hitting the web interface I was not able to make a crash. Now try the following backlog. Also no success. There is no crash. Do you have PHYSICAL buttons connected? I see also in my configuration that there are high frequency interferences coming from the stepper causing "ghost" button actions.

`backlog shutteropen;shutterstop;shutteropen;shutterstop`

stefanbode, 2021-01-21:
@TrDA-hab : Can you reproduce the error only one time or was it no problem. Until now I had no chance to create a crash. My MQTT is not connected. Maybe I should fix this and the crash comes from MQTT. Maybe you can switch it off and check, if the crash still exists. Will help me to find the root cause.

TrDA-hab, 2021-01-21:
> @TrDA-hab : Can you reproduce the error only one time or was it no problem. Until now I had no chance to create a crash. My MQTT is not connected. Maybe I should fix this and the crash comes from [...]

1. Which version of Tasmot are you using?
2. I will test again and write. It will be fast!

J-Mau-5, 2021-01-21:
I can reproduce it every time when i hit the buttons very fast ... but as mentioned the crash also occures randomly (not often but sometimes) when actuation the shutter just normal by pressing only one time. it then does not matter if i use the hardware or software buttons but hardware buttons seems to be more prone to produce the crash..
i also supposed that the noise from the stepper could produce fast button inputs (are gpio buttons sampled or interrupt driven ?) added capacitors on gpio and power supply with no luck... 
will also try to disable MQTT ...

stefanbode, 2021-01-21:
I am using the latest DEV version. Anyhow there where no change during the last 8 weeks that might have influence. therefore version should not be an issue. Just to ensure: You control with the button UP/DOWN you MUSTNT control the relay with the button. This should be no problem with the webinterface

stefanbode, 2021-01-21:
@J-Mau-5 : Please can you check two influences: MQTT and BUTTONDEBOUNCE

J-Mau-5, 2021-01-21:
@stefanbode shure...do i have to disable debounce for every single button or does the command ButtonDebounce 1000 affect all buttons ?

stefanbode, 2021-01-21:
It effects all buttons.

J-Mau-5, 2021-01-21:
@stefanbode did some testing with MQTT disabled and BUTTONDEBOUNCE 1000:
- hitting button down fast in webinterface ends in same reaction: sometimes motor stops and hums but not spinning for a short time, then goes on spinning and finds its position, no tasmota restart...... some other times similar behavior but followed by tasmota restart.

- hitting button up fast in webinterface: sometimes leads to motor stopping + humming followed by tasmota freezing, no automatic restart.

- operating buttons normal (no fast button hitting): not tested enough at this time...

J-Mau-5, 2021-01-21:
@stefanbode : hitting once and waiting until hit has reached position seems to work fine until now. will do further testing.
you seem to be right with your presumption about the ramp function: button inputs while motor is ramping occasionally still provokes the crash.... even though i could produce the crash with ShutterMotorDelay 0 in the past.

TrDA-hab, 2021-01-21:
I repeated the test now.
I am using Tasmtota 9.2.0 (release)

[image]
[image]

Step by step:
1. Switch on the power supply.
2. Wait 5 seconds for the software to load.
3. Quickly press 1 time on the button in the WEB interface.
4. The stepper began to spin.
5. Quick start at the button.
6. The motor has stopped.
7. The motor started to rotate.
8. Quick start at the button.
9. Tasmota - reboots.

Very rarely there is a strange behavior (bug) like in this video:
https://drive.google.com/file/d/1W_f-8oqhw3r662KdwXfpW7CK1A_atBbs/view?usp=sharing

J-Mau-5, 2021-01-21:
@TrDA-hab "Very rarely there is a strange behavior (bug) like in this video:" same with my device...

stefanbode, 2021-01-21:
ok, I got close to the root cause. with loglevel 3 or lower (0,1,2) the error does not happen. With loglevel 4 I am able to create a crash, too. the last line I get is:
`10:41:05.313 SHT: Remain steps -29, counter 29, freq 65`

and the line missing (caused by the crash is:
`10:41:04.931 SHT: Real 125000, pulsecount 74, start 125000`

The stacktrace shows that the exxor is related to the print of this line. Please can you confirm that you do NOT have the error with seriallog 2?

stefanbode, 2021-01-21:
I submitted a change request to fix the problem. Looks like the problem is in the core writing the frequency twice in a short time frame. I removed a line because I'm sure we are fine also without this. On my side now there is no crash anymore.

stefanbode, 2021-01-21:
We leave the issue open until you can confirm that the new version fixes the problem also on your side

J-Mau-5, 2021-01-21:
@stefanbode : as far as i understood the crash occures when tasmota tries to print a log message ? That would mean with loglevel off crash does not happen?
during the day i let the shutter go up and down many times by just hitting the button once. My observation:
- using web interface button Mostly it works normal without crash
- using web interface button just got a crash on ramping DOWN not UP as usual! button was pressed just once.
- using hardware buttons connected to GPIO the crash still happens often, even with ButtonDebounce 1000
 
web log level = serial log level = 2 in my setup.

stefanbode, 2021-01-21:
There is anew development version. Please test this one. No change required at all for any setting

J-Mau-5, 2021-01-21:
thanks, will do so..

ascillato2, 2021-01-24:
> thanks, will do so..

Hi, were you able to test latest fix? thanks.

J-Mau-5, 2021-01-24:
yes, i was.  The problem is not entirely solved for me:
- webinterface seems a bit more stable now, however rapid button clicks still lead to crashing / freezing of tasmota.
- hardware gpio connected buttons (that maybe bounce) lead to crash even when pushing once.
 
@TrDA-hab , can you verify the latest fix to be working for you even with hardware buttons ? ... If yes i can maybe blame my stepper motor & driver to feed too much noise into the circuit to operate properly.

TrDA-hab, 2021-01-24:
I will test it tomorrow, the new firmware and provide the results here. 
Please wait not long.

stefanbode, 2021-01-25:
Just for information: besides the button debounce I do not see a chance to fix it on the tasmota site. As long as the reboot is caused by the function frequencyChange the only thing we can do is to ensure the function is not called twice in less than 50ms.

TrDA-hab, 2021-01-25:
1. I tested the latest firmware version (for developers) and it works very stable. **I like it very much!**
2. The ESP continues to freeze and / or overload if quickly many times by pressing the button in the WEB interface. It doesn't bother me - I have a normal work schedule.
3. ButtonDebounce 1000 - doesn't help much.

I think this question can be closed.
**Tasmota possibilities are not limitless!**

closed 2021-01-25 by ascillato2

TrDA-hab, 2021-01-28:
@stefanbode 
For your information only (this is only the first test), there will be more further ...

TrDA-hab, 2021-01-28:
https://youtu.be/2r1aYv723qo

sajovicd, 2021-02-11:
Hello,
I have similar problem. 
Everytime I push the oposite button to change direction while motor is running or if I insert stop command in console while the motor is spinnig, Nodemcu crash. 
I run on development 9.2.0.7
It always crash when ramping, even if I use ShutterMotorDelay it is the same. it starts fine and also decreasing speed fine until final ramp, and then rebots.

This is my logs if any help

`18:52:53.478 CMD: Backlog Template; Module; GPIO 255
18:52:53.540 MQT: stat/zaluzije/spalnica2/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:52:53.741 MQT: stat/zaluzije/spalnica2/RESULT = {"Module":{"0":"Generic"}}
18:52:53.956 MQT: stat/zaluzije/spalnica2/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"544":"LedLink"},"GPIO3":{"0":"None"},"GPIO4":{"352":"Counter1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"416":"PWM1"},"GPIO14":{"256":"Relay_i1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

18:54:44.062 CMD: Status 0
18:54:44.070 MQT: stat/zaluzije/spalnica2/STATUS = {"Status":{"Module":0,"DeviceName":"Zaluzije Spalnica2","FriendlyName":["Zaluzije Spalnica 2","Tasmota2","zaluzijeon"],"Topic":"zaluzije/spalnica2","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:54:44.105 MQT: stat/zaluzije/spalnica2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:08:22","StartupUTC":"2021-02-11T17:46:22","Sleep":50,"CfgHolder":4617,"BootCount":83,"BCResetTime":"2021-02-10T19:36:22","SaveCount":361,"SaveAddress":"FA000"}}
18:54:44.138 MQT: stat/zaluzije/spalnica2/STATUS2 = {"StatusFWR":{"Version":"9.2.0.7(tasmota)","BuildDateTime":"2021-02-11T16:27:36","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"440/699"}}
18:54:44.160 MQT: stat/zaluzije/spalnica2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SajkoWnetAP",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","40000080","00006000","00000000"]}}
18:54:44.190 MQT: stat/zaluzije/spalnica2/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:54:44.227 MQT: stat/zaluzije/spalnica2/STATUS5 = {"StatusNET":{"Hostname":"Zaluzije-Spalnica-2","IPAddress":"192.168.1.240","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"D8:BF:C0:14:C0:07","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:54:44.248 MQT: stat/zaluzije/spalnica2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.60","MqttPort":1883,"MqttClientMask":"Zaluzije_spalnica2","MqttClient":"Zaluzije_spalnica2","MqttUser":"sajovicd","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:54:44.272 MQT: stat/zaluzije/spalnica2/STATUS7 = {"StatusTIM":{"UTC":"2021-02-11T17:54:44","Local":"2021-02-11T18:54:44","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:04","Sunset":"18:03"}}
18:54:44.294 MQT: stat/zaluzije/spalnica2/STATUS10 = {"StatusSNS":{"Time":"2021-02-11T18:54:44","COUNTER":{"C1":8000},"Shutter1":{"Position":100,"Direction":0,"Target":100}}}
18:54:44.310 MQT: stat/zaluzije/spalnica2/STATUS11 = {"StatusSTS":{"Time":"2021-02-11T18:54:44","Uptime":"0T00:08:22","UptimeSec":502,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":101,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SajkoWnetAP","BSSId":"54:E6:FC:AD:2C:B2","Channel":6,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}}
18:54:44.350 MQT: stat/zaluzije/spalnica2/STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"Exception","EPC":["4000dce5","00000000","402462b0"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40215cb8","402491ae","4026e7c8","40249df2","4026e7d5","4026e81a","4024625c","40000f49","40000f49","40000e19","40105a6d","40105a73","4010000d","40267264","40267215","40100a40","40102ae5","401026b4","4010594f","40104b2a","40104f9b","40104a63","40103928","4000050c","40103610","4000d53a","4010045e","4000d53a","401013c1","401013a9","40285196"]}}
18:54:44.394 MQT: stat/zaluzije/spalnica2/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"4"}}}
18:56:22.395 APP: Serial logging disabled`

stefanbode, 2021-02-13:
Shuttermotordelay should be a must with stepper motors to have a smooth ramp. Regarding the reboot there is nothing we can do currently. Looks like a new problem in the SDk itself. Maybe you can downgrade to a 9.1 version

J-Mau-5, 2021-02-13:
@sajovicd do you use hardware buttons or web interface to change direction ? 
for some reason web interface mostly works for me (even to change direction) while with hardware buttons i highly likely will get a crash.

sajovicd, 2021-02-14:
Hello thanks for your respond. 
@stefanbode 
I also try with shuttermotordelay it was the same, when it comes to ramping it crashes. 
will also try your suggestion on version 9.1

@J-Mau-5 
I only use web buttons for now.