arendst/Tasmota #10569: IF statement + outer statement not working

poggenpower, 2021-01-14:
### PROBLEM DESCRIPTION
Following IF Statements are not working:

rule1 ON Event#test DO IF (1 == 1) var1 1 ELSE var3 3 ENDIF; var2 2 ENDON
rule1 ON Event#test BACKLOG IF (1 == 1) var1 1 ELSE var3 3 ENDIF; var2 2 ENDON
rule1 ON Event#test DO BACKLOG IF (1 == 1) var1 1 ELSE var3 3 ENDIF; var2 2 ENDON

1st line: the if clase works as expected, but var2 wont be set
2nd line: doesn't work completely, but it is given in https://tasmota.github.io/docs/Rules/#conditional-rules
3rd line: isn't evaluating the if clause, but setting var2.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 core 
- [X] Tasmota binary firmware version number used: 9.2.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
- [X] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:54:10.374 MQT: esp/esp04/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"33":"Button2"},"GPIO5":{"227":"Relay4"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"256":"Relay_i1"},"GPIO17":{"257":"Relay_i2"},"GPIO18":{"35":"Button4"},"GPIO19":{"34":"Button3"},"GPIO20":{"0":"None"},"GPIO21":{"32":"Button1"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"}}
23:54:10.383 MQT: esp/esp04/stat/RESULT = {"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"608":"I2C SCL"},"GPIO33":{"640":"I2C SDA"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
23:55:04.623 MQT: esp/esp04/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":83,"Free":428,"Rules":"ON Event#test DO backlog var5 5; IF (1 == 0) var1 1 ELSE var3 3 ENDIF; var2 2 ENDON"}
23:55:04.851 MQT: esp/esp04/stat/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":191,"Free":320,"Rules":"ON system#boot DO var1 0 ENDON ON button3#state DO BACKLOG if (var1!=1) var1 1;var5 timer ELSE var1 0;var5 block ENDIF; publish esp/esp03/cmnd/event SetShutterBlock=%var1%; Event ReDply ENDON"}
23:55:05.102 MQT: esp/esp04/stat/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":368,"Free":143,"Rules":"ON Event#esp03ShutterMove DO backlog var2 %value%; var4 \"Delay: \"%value%; Event ReDply; ruletimer1 60 ENDON ON rules#timer=1 DO ShutterPosition1 %value% ENDON ON Event#ReDply DO BACKLOG displaytext [l1c1C0f2]%var14%[l1c1C1f2]%var4%;var14 %var4%;displaytext [l2c1C0f2]%var5%[l2c1C1f2]%var5%;var15 %var5%;displaytext [l3c1C0f2]%var16%[l3c1C1f2]%var6%;var16 %var6%; ENDON"}

```
- [X ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
23:56:26.236 MQT: esp/esp04/stat/STATUS = {"Status":{"Module":1,"DeviceName":"esp04","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"esp04","ButtonTopic":"","Power":12,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:56:26.242 MQT: esp/esp04/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"3T12:54:43","StartupUTC":"2021-01-11T10:01:43","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2021-01-03T17:26:12","SaveCount":189}}
23:56:26.249 MQT: esp/esp04/stat/STATUS2 = {"StatusFWR":{"Version":"9.2.0.3(thl-display-rule)","BuildDateTime":"2021-01-08T00:41:02","Core":"1_0_5_RC4","SDK":"v3.3.4-286-gb4c075169","CpuFrequency":80,"Hardware":"ESP32","CR":"434/699"}}
23:56:26.255 MQT: esp/esp04/stat/STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
23:56:26.264 MQT: esp/esp04/stat/STATUS4 = {"StatusMEM":{"ProgramSize":1471,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FFAC78F","00141FA1","000000CF","010013C0","40000981","00004020","00208000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,13,20,21,24,26,27,29,50","Sensors":"1,2,3,5,6,62"}}
23:56:26.273 MQT: esp/esp04/stat/STATUS5 = {"StatusNET":{"Hostname":"esp04","IPAddress":"192.168.7.105","Gateway":"192.168.7.248","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.248","Mac":"24:62:AB:D4:D3:00","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:56:26.277 MQT: esp/esp04/stat/STATUS6 = {"StatusMQT":{"MqttHost":"haus.wupp","MqttPort":1883,"MqttClientMask":"esp04","MqttClient":"esp04","MqttUser":"esp04","MqttCount":6,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:56:26.286 MQT: esp/esp04/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-01-14T22:56:26","Local":"2021-01-14T23:56:26","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:38","Sunset":"17:19"}}
23:56:26.292 MQT: esp/esp04/stat/STATUS10 = {"StatusSNS":{"Time":"2021-01-14T23:56:26"}}
23:56:26.303 MQT: esp/esp04/stat/STATUS11 = {"StatusSTS":{"Time":"2021-01-14T23:56:26","Uptime":"3T12:54:43","UptimeSec":305683,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":6,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"ON","Wifi":{"AP":1,"SSId":"iot","BSSId":"78:8A:20:49:35:67","Channel":11,"RSSI":100,"Signal":-50,"LinkCount":5,"Downtime":"0T00:14:52"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

#### 1 ######

23:57:42.868 CMD: rule1 ON Event#test DO IF (1 == 1) var1 1 ELSE var3 3 ENDIF; var2 2 ENDON
23:57:42.869 SRC: WebConsole from 192.168.11.201
23:57:42.872 CMD: Group 0, Index 1, Command "RULE", Data "ON Event#test DO IF (1 == 1) var1 1 ELSE var3 3 ENDIF; var2 2 ENDON"
23:57:42.877 RUL: Stored uncompressed, would compress from 67 to 64 (-4%)
23:57:42.886 MQT: esp/esp04/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":67,"Free":444,"Rules":"ON Event#test DO IF (1 == 1) var1 1 ELSE var3 3 ENDIF; var2 2 ENDON"}
23:57:43.504 CFG: Saved, Count 191, Bytes 4096
23:57:47.468 WIF: Checking connection...
23:57:53.279 CMD: event test
23:57:53.280 SRC: WebConsole from 192.168.11.201
23:57:53.282 CMD: Group 0, Index 1, Command "EVENT", Data "test"
[... 37 lines cut ...]
00:01:03.008 CMD: Group 0, Index 1, Command "IF", Data "(1 == 1) var1 1 ELSE var3 3 ENDIF"
00:01:03.021 MQT: esp/esp04/stat/RESULT = {"If":"Done"}
00:01:03.256 SRC: Backlog
00:01:03.259 CMD: Group 0, Index 2, Command "VAR", Data "2"
00:01:03.268 MQT: esp/esp04/stat/RESULT = {"Var2":"2"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Set the rules listed ^^^^ and then trigger them by `event test`

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
vars inside and outside of the rules should be set

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-01-15:
For 2 consecutive commands, you must use backlog first. Please, check the docs for backlog usage. Thanks.

closed 2021-01-15 by ascillato2

pestevez, 2021-11-03:
@barbudor - did you get a chance to review this? If not, does it make sense to re-open this issue or the one I created with the debug info for the latest version? That way other folks in the community can contribute too, what do you think?

I am also happy to create a new one, ensuring that I don't delete the screenshots section so the bot does not auto close it.