arendst/Tasmota #10578: [Feature Request] Binary sensor

itec78, 2021-01-16:
### PROBLEM DESCRIPTION
I'm creating a device with 4 binary sensors (reed, pir, light, obstacle)
I used pir example I found in documentation, but it's a little tricky. I had to create rules for sending mqtt messages, both for status change and for Home Assitant autodiscovery. (https://tasmota.github.io/docs/PIR-Motion-Sensors/, https://blakadder.com/pir-in-tasmota/, https://templates.blakadder.com/TY01.html)
In this way, sensor is updated only when his status changes, but it is not updated at boot or regularly. If the status changes when device is off, or there is a network issue, the status will never change (bad for door and window sensors)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1
- [x] Tasmota binary firmware version number used: 9.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
01:55:03 MQT: stat/porta/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
01:55:04 MQT: stat/porta/RESULT = {"Module":{"0":"Generic"}}
01:55:04 MQT: stat/porta/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"161":"Switch2"},"GPIO13":{"162":"Switch3"},"GPIO14":{"160":"Switch1"},"GPIO15":{"163":"Switch4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
02:02:02 MQT: stat/porta/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":2,"Free":509,"Rules":"''"}
02:02:02 RUL: Rule2 on system#boot do publish2 homeassistant/binary_sensor/%deviceid%_serratura/config {"name":"Porta serratura","state_topic":"stat/%topic%/SWITCH1","value_template":"{{ value_json.STATE }}","availability_topic":"tele/%topic%/LWT","payload_available":"Online","payload_not_available":"Offline","device_class":"lock","force_update":true,"unique_id":"%deviceid%_serratura","device":{"identifiers":["%deviceid%"]}} endon on system#boot do publish2 homeassistant/binary_sensor/%deviceid%_illu/config
02:02:02 RUL: Rule2 +{"name":"Illuminazione corridoio","state_topic":"stat/%topic%/SWITCH2","value_template":"{{ value_json.STATE }}","availability_topic":"tele/%topic%/LWT","payload_available":"Online","payload_not_available":"Offline","device_class":"light","force_update":true,"unique_id":"%deviceid%_illu","device":{"identifiers":["%deviceid%"]}} endon
02:02:02 MQT: stat/porta/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":828,"Free":187,"Rules":"on system#boot do publish2 homeassistant/binary_sensor/%deviceid%_serratura/config {\"name\":\"Porta serratura\",\"state_topic\":\"stat/%topic%/SWITCH1\",\"value_template\":\"{{ value_json.STATE }}\",\"availability_topic\":\"tele/%topic%/LWT\",\"payload_available\":\"Online\",\"payload_not_available\":\"Offline\",\"device_class\":\"lock\",\"force_update\":true,\"unique_id\":\"%deviceid%_serratura\",\"device\":{\"identifiers\":[\"%deviceid%\"]}} endon on system#boot do publish2 homeassistant/binary_sensor/%deviceid%_illu/config {\"name\":\"Illuminazi..."}
02:02:02 RUL: Rule3 on system#boot do publish2 homeassistant/binary_sensor/%deviceid%_pir/config {"name":"Corridoio pir","state_topic":"stat/%topic%/SWITCH3","value_template":"{{ value_json.STATE }}","availability_topic":"tele/%topic%/LWT","payload_available":"Online","payload_not_available":"Offline","device_class":"motion","force_update":true,"unique_id":"%deviceid%_pir","device":{"identifiers":["%deviceid%"]}} endon on system#boot do publish2 homeassistant/binary_sensor/%deviceid%_portaz/config
02:02:02 RUL: Rule3 +{"name":"Porta","state_topic":"stat/%topic%/SWITCH4","value_template":"{{ value_json.STATE }}","availability_topic":"tele/%topic%/LWT","payload_available":"Online","payload_not_available":"Offline","device_class":"door","force_update":true,"unique_id":"%deviceid%_portaz","device":{"identifiers":["%deviceid%"]}} endon
02:02:02 MQT: stat/porta/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":801,"Free":197,"Rules":"on system#boot do publish2 homeassistant/binary_sensor/%deviceid%_pir/config {\"name\":\"Corridoio pir\",\"state_topic\":\"stat/%topic%/SWITCH3\",\"value_template\":\"{{ value_json.STATE }}\",\"availability_topic\":\"tele/%topic%/LWT\",\"payload_available\":\"Online\",\"payload_not_available\":\"Offline\",\"device_class\":\"motion\",\"force_update\":true,\"unique_id\":\"%deviceid%_pir\",\"device\":{\"identifiers\":[\"%deviceid%\"]}} endon on system#boot do publish2 homeassistant/binary_sensor/%deviceid%_portaz/config {\"name\":\"Porta\",\"state_topic\"..."}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
02:03:23 MQT: stat/porta/STATUS = {"Status":{"Module":0,"DeviceName":"Porta","FriendlyName":["Porta"],"Topic":"porta","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[1,2,1,1,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
02:03:23 MQT: stat/porta/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:02:50","StartupUTC":"2021-01-16T01:00:33","Sleep":50,"CfgHolder":4617,"BootCount":48,"BCResetTime":"2021-01-11T03:09:19","SaveCount":207,"SaveAddress":"F8000"}}
02:03:23 MQT: stat/porta/STATUS2 = {"StatusFWR":{"Version":"9.2.0(sensors)","BuildDateTime":"2020-12-21T15:04:56","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"694/699"}}
02:03:23 MQT: stat/porta/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DeepThought-IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
02:03:23 MQT: stat/porta/STATUS4 = {"StatusMEM":{"ProgramSize":638,"Free":364,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
02:03:23 MQT: stat/porta/STATUS5 = {"StatusNET":{"Hostname":"tasmo_porta","IPAddress":"172.24.7.111","Gateway":"172.24.7.1","Subnetmask":"255.255.255.0","DNSServer":"172.24.7.1","Mac":"CC:50:E3:5D:63:A9","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
02:03:23 MQT: stat/porta/STATUS6 = {"StatusMQT":{"MqttHost":"172.24.7.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5D63A9","MqttUser":"mqttusr","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
02:03:23 MQT: stat/porta/STATUS7 = {"StatusTIM":{"UTC":"2021-01-16T01:03:23","Local":"2021-01-16T02:03:23","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:37","Sunset":"17:22"}}
02:03:23 MQT: stat/porta/STATUS10 = {"StatusSNS":{"Time":"2021-01-16T02:03:23","Switch1":"ON","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF"}}
02:03:23 MQT: stat/porta/STATUS11 = {"StatusSTS":{"Time":"2021-01-16T02:03:23","Uptime":"0T00:02:50","UptimeSec":170,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"DeepThought-IOT","BSSId":"18:D6:C7:95:41:C1","Channel":11,"RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
[... cut ...]
I think it could be useful to add a new peripheral "Binary sensor", with the ability to choose 1-5.
The name could be set as friendly name (in "Configure other" menu)
The device_class could be set as an option (maybe "binaryclass<x>")
An option to switch boolean state (like switchmode 1 and switchmode 2 does) could be nice

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-01-16:
>... for status change and for Home Assitant autodiscovery.

Hi. That is already implemented. 

You should try `SETOPTION19 0` for using Home Assistant Native Integration. You will see your device with its sensors under INTEGRATIONS in the configuration menu of Home Assistant. For displaying your sensors as binary sensors, check the docs at commands for detaching button from relays and you are done.

If you need further assistance, please address this to the Tasmota support chat.

closed 2021-01-16 by ascillato2

itec78, 2021-01-16:
> You should try `SETOPTION19 0` for using Home Assistant Native Integration. You will see your device with its sensors under INTEGRATIONS in the configuration menu of Home Assistant. For displaying [...]

Of course I'm using setoption19. On HA I can see the device and the entity "status", but I can't find other entities for the added sensors, so I think that native integration is not working in this way.
I tested button and detached it as you suggested. Not working. I need to know if a sensor is on or off, button has only one state.
I spent 2 nights looking for documentation and chat, but probably I'm missing something.
It's very easy to add a thermometer or every kind of sensor and it's automatically announced to HA, but seems it's impossible to do the same with a generic binary sensor.
My request was to make it simpler like adding a thermometer.

> 
> If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.
> 
> Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.
> ### Support Information
> 
> See [Docs](https://tasmota.github.io/docs/) for more information.
> See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
> See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)