arendst/Tasmota #10585: Wie: Repeat last UID when an invalid UID is received

iot-sle, 2021-01-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If an invalid UID is received the previous correct key is published again. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____Generic
- [ x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: __esptool___
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:28:33.727 MQT: stat/MiniBoard02/RESULT = {"NAME":"ESP-01S","GPIO":[1,1,1,1,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
17:28:33.940 MQT: stat/MiniBoard02/RESULT = {"Module":{"0":"ESP-01S"}}
17:28:34.197 MQT: stat/MiniBoard02/RESULT = {"GPIO0":{"6432":"Wiegand D0 (green) 5V"},"GPIO1":{"0":"None"},"GPIO2":{"6464":"Wiegand D1 (white) 5V"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
17:28:41.431 CMD: lua
17:28:41.439 MQT: stat/MiniBoard02/RESULT = {"Command":"Unknown"}

```
- [ x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
17:30:59.476 MQT: stat/MiniBoard02/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":42,"Free":469,"Rules":"on Wiegand#UID do DzSend1 58,%value% endon"}

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:32:12.923 MQT: stat/MiniBoard02/STATUS = {"Status":{"Module":0,"DeviceName":"TAS-MiniBoard2","FriendlyName":["MiniBoard2"],"Topic":"MiniBoard02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:32:12.928 MQT: stat/MiniBoard02/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T02:09:39","StartupUTC":"2021-01-16T14:22:33","Sleep":50,"CfgHolder":4617,"BootCount":80,"BCResetTime":"2020-11-09T18:13:29","SaveCount":187,"SaveAddress":"FA000"}}
17:32:12.934 MQT: stat/MiniBoard02/STATUS2 = {"StatusFWR":{"Version":"9.2.0.3(tasmota)","BuildDateTime":"2021.01.16 15:15:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"434/699"}}
17:32:12.939 MQT: stat/MiniBoard02/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["freiheit-sl-home",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
17:32:12.949 MQT: stat/MiniBoard02/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","04001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,82"}}
17:32:12.955 MQT: stat/MiniBoard02/STATUS5 = {"StatusNET":{"Hostname":"MiniBoard02-3705","IPAddress":"192.168.252.87","Gateway":"192.168.252.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.252.2","Mac":"A4:CF:12:F4:AE:79","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:32:12.959 MQT: stat/MiniBoard02/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.252.5","MqttPort":1883,"MqttClientMask":"TAS-MiniBoard2","MqttClient":"TAS-MiniBoard2","MqttUser":"xxxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:32:12.968 MQT: stat/MiniBoard02/STATUS7 = {"StatusTIM":{"UTC":"2021-01-16T16:32:12","Local":"2021-01-16T17:32:12","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"09:37","Sunset":"17:55"}}
17:32:12.974 MQT: stat/MiniBoard02/STATUS10 = {"StatusSNS":{"Time":"2021-01-16T17:32:12"}}
17:32:12.980 MQT: stat/MiniBoard02/STATUS11 = {"StatusSTS":{"Time":"2021-01-16T17:32:12","Uptime":"0T02:09:39","UptimeSec":7779,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"DC:39:6F:66:66:66","Channel":1,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:05"}}}
17:32:44.432 MQT: tele/MiniBoard02/STATE = {"Time":"2021-01-16T17:32:44","Uptime":"0T02:10:11","UptimeSec":7811,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"freiheit-sl-home","BSSId":"DC:39:6F:6A:0D:33","Channel":1,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:05"}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
receive an invalid key ( parity error)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
never send an old key a second time.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-01-16:
As the inventor of this driver I suppose you also providing the solution.

Pls use a latest pull request as I see your gpio naming is wrong.

iot-sle, 2021-01-16:
> As the inventor of this driver I suppose you also providing the solution.
> 
> Pls use a latest pull request as I see your gpio naming is wrong.

ok. I found the error on working on my copy. So I will do a pull new pull request.
And fix this issue.

ascillato2, 2021-01-16:
> I found the error on working on my copy. So I will do a pull new pull request. And fix this issue.

Thanks.

closed 2021-01-16 by ascillato2

iot-sle, 2021-01-16:
Great. it's only  a wrong if query  about bool newKey. So I will wait and not do a pull request even though I did a another small improvement.  It takes me a long time to do the pull request, because of my GIT knowledge. Sorry

ascillato, 2021-01-16:
> It takes me a long time to do the pull request

One fast way of doing a PR is just browse in Tasmota repo to the file you want to modify, click on edit, modify the code you need, click on propose changes and then click on create pull request. For a just one file modification, this way is really fast and you will not have any issue of old files or whatever. Github web will use the latest codebase for you.

PR #10589 (2021-01-17): only in case of valid key do action. Issue#10585 
…incoming signals

## Description:

**Related issue (if applicable):** fixes #10585 
only in case of valid key do action. Issue#10585 
## Checklist:
  - [ x] The pull request is done against the latest development branch
  - [ x] Only relevant files were touched
  - [x ] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ x] The code change is tested and works on Tasmota core ESP32 V.1.0.5-rc4
  - [ x] I accept the [CLA](https:// [... cut]