arendst/Tasmota #10586: Tasmota 9.2 losing contact with TH16 SI7021 Probe

JedidiahStolzfus, 2021-01-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
First, I'm rather new to the github community, so if I screw something up please forgive me.  I did search for this issue, but didn't find anything that seemed related.

I recently purchased two SONOFF TH16 devices, plus I have one that's a couple of years old running version 6.6.0.  The old one has been running flawlessly in an unheated garage for 2 years.  The two new ones were flashed with version 9.2 with no problems.  The first one flashed I started to see issues with the SI7021 temperature probe losing contact with the SONOFF.  Thinking that it might be a defective unit, I flashed the other one with the same version, only to see the exact same behavior.  On a hunch I flashed the one of them with the oldest firmware available, 6.7.1 and let it run.  It hasn't had a single loss of data since I did that. 

I then decided to conduct a test between the three units I have all with different firmware.  I setup a Node Red tab with the three graphs inserted below.  As you can see, the 9.2 version is losing contact with the sensor at random times, while the other two never lose contact.

This data leads me to believe there is something wrong in the 9.2, and possibly other versions of the firmware.  I don't know, but I suspect that the device might be polling the sensor too fast, but that is only a guess.  In the second attached image, you can see that the readings from the probe all go "null", so it's not a connectivity problem between the device and my NodeRed system..  I monitored connectivity between the devices and there was never a loss.

Hopefully this is enough information.
Thank you.

- [ ] Device used (e.g., Sonoff Basic): SONOFF TH16_____
- [ ] Tasmota binary firmware version number used: 9.2_____
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Tasmotizer for Windows_____

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
```18:33:48 MQT: stat/test/RESULT = {"NAME":"Sonoff TH","GPIO":[32,1,0,1,0,0,0,0,224,320,1248,0,0,0],"FLAG":0,"BASE":4}
18:33:48 MQT: stat/test/RESULT = {"Module":{"4":"Sonoff TH"}}
18:33:48 MQT: stat/test/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"1248":"SI7021"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
18:33:54 MQT: tele/test/STATE = {"Time":"2021-01-16T18:33:54","Uptime":"0T01:25:16","UptimeSec":5116,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"galapagos","BSSId":"60:38:E0:D3:E1:45","Channel":1,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:05"}}
18:33:54 MQT: tele/test/SENSOR = {"Time":"2021-01-16T18:33:54","SI7021":{"Temperature":26.9,"Humidity":37.8,"DewPoint":11.3},"TempUnit":"C"}
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  Status output here:
18:34:45 CMD: STATUS 0
18:34:45 MQT: stat/test/STATUS = {"Status":{"Module":4,"DeviceName":"Test","FriendlyName":["Test"],"Topic":"test","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:34:45 MQT: stat/test/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:26:07","StartupUTC":"2021-01-16T16:08:38","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2021-01-11T20:50:56","SaveCount":205,"SaveAddress":"F7000"}}
18:34:45 MQT: stat/test/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"369/699"}}
18:34:45 MQT: stat/test/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["galapagos",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
18:34:45 MQT: stat/test/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:34:45 MQT: stat/test/STATUS5 = {"StatusNET":{"Hostname":"Test","IPAddress":"172.19.100.132","Gateway":"172.19.100.254","Subnetmask":"255.255.255.0","DNSServer":"172.19.100.4","Mac":"84:CC:A8:9B:A5:1F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:34:45 MQT: stat/test/STATUS6 = {"StatusMQT":{"MqttHost":"172.19.100.2","MqttPort":1883,"MqttClientMask":"test","MqttClient":"test","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:34:45 MQT: stat/test/STATUS7 = {"StatusTIM":{"UTC":"2021-01-16T17:34:45","Local":"2021-01-16T18:34:45","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:37","Sunset":"17:22"}}
18:34:45 MQT: stat/test/STATUS10 = {"StatusSNS":{"Time":"2021-01-16T18:34:45","SI7021":{"Temperature":26.9,"Humidity":37.8,"DewPoint":11.3},"TempUnit":"C"}}
18:34:45 MQT: stat/test/STATUS11 = {"StatusSTS":{"Time":"2021-01-16T18:34:45","Uptime":"0T01:26:07","UptimeSec":5167,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"galapagos","BSSId":"60:38:E0:D3:E1:45","Channel":1,"RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

https://imagizer.imageshack.com/img923/1518/8oOQBY.jpg
https://imagizer.imageshack.com/img924/2871/U4FxeQ.jpg

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Tried to drop the screen shots directly, something didn't work.  The links are safe from my own Imageshack site.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-01-16:
My experience with Sonoff TH16 and the probe containing SI7021 is that the plug/socket connection can be fickle, and if not having good contact, null readings will occur. When that's ok, I'm seeing no fallouts on my 3 Sonoff TH16 running Tasmota 9.2 for about one month now.

JedidiahStolzfus, 2021-01-16:
> 
> 
> My experience with Sonoff TH16 and the probe containing SI7021 is that the plug/socket connection can be fickle, and if not having good contact, null readings will occur. When that's ok, I'm seeing no fallouts on my 3 Sonoff TH16 running Tasmota 9.2 for about one month now.

I did think about that.  Used QC Contact Cleaner on the jack and plug just to make sure there's no excess flux built up.  Swapped probes between the two new units.  Tried twisting the plug in the jack to try to scrape anything that might be causing a poor connection.  No luck.  The only thing that has resolved it was putting an older firmware on it.  

As it stands right now, the two new units, one with 9.2 and one with 6.7.1 have been running all day, 6.7.1 shows no drops, 9.2 shows dozens of drops.  Appreciate the input though.

When I have more time I'm probably going to start flashing versions of the firmware on the one that isn't having problem (even though it did on 9.2) until it breaks.

sfromis, 2021-01-16:
Causes of dubious connections with that hardware design includes the plug not getting properly into the socket, which can reside a bit too deep behind the plastic. This can also occur with perfectly clean metal surfaces. I've fixed one case by digging into the socket to tighten the spring, and making really sure to reduce the distance as much as practical.

JedidiahStolzfus, 2021-01-18:
A dubious connection doesn't seem to make sense that the only thing that's really changed is the firmware on the two newer devices.  As a final test, I could flash the one that's currently running 9.2 with 6.7.1 and show that it'll maintain a connection, just like the other one that's running 6.7.1 is.  I'm still graphing the data and the one that's running 9.2 is still dropping randomly while the 6.7.1 hasn't dropped once.  Both devices dropped randomly until one was downgraded to 6.7.1, nothing else changed on it other than the downgrade.

sfromis, 2021-01-18:
As I'm having rock solid stability with Sonoff TH16 SI7021 on Tasmota 9.2, odds are that "something" does not work as well in your hardware setup.

closed 2021-01-18 by ascillato2