arendst/Tasmota #10587: BME280 Sensor showing completely wrong values

flobbr, 2021-01-16:
### PROBLEM DESCRIPTION
I have a Sonoff Basic with an attached BME280 sensor which shows values which are completely wrong. It is placed outside where the temperature ist about 0°C. The sensor data shown is 21.7°C, 93.6 % humidity, 671.9 hPa pressure.
I have the BME280 connects in the right way ( VCC to 3.3V, GND to GND, RX to SDA, TX to SCL) I have Tasmota 9.2.0 Sensors, and the module is configured GPIO1 I2C SCL, GPIO3 I2C SDA.
Of course serial logging is off.
I thought the sensor was bad, so i ordered another few from a different vendor --> same problem.
I thought maybe my Sonoff is bad so i bought a new one, flashed it via the pins so those are working as well --> same problem.
The connections are good otherwise there shouldn't be data at all or -142°C.
I tried it with and without soldering VCC to CSB and SDO to GND.
I have another sonoff working with BME280 without soldering CSB to VCC in another room without issues.
I tried flashing again, i tried resetting again, i treib changing the sensor, i tried changing the sonoff, nothing is working. 
The sensor output isn't changing also, its just showing the same values for days. 
Weird thing: If i connect the sensor which works good in the other room, it doesn't work there either.
I2CScan shows: {"I2CScan":"Device(s) found at 0x76"}
I don't know what to do.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic R2
- [x] Tasmota binary firmware version number used: 9.2.0 Sensors
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:24:39 CMD: Backlog Template; Module; GPIO 255
19:24:39 MQT: stat/118/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:24:39 MQT: stat/118/RESULT = {"Module":{"1":"Sonoff Basic"}}
19:24:39 MQT: stat/118/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"608":"I2C SCL"},"GPIO2":{"0":"None"},"GPIO3":{"640":"I2C SDA"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:26:58 CMD: Status 0
19:26:58 MQT: stat/118/STATUS = {"Status":{"Module":1,"DeviceName":"118","FriendlyName":["118"],"Topic":"118","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:26:58 MQT: stat/118/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:31:01","StartupUTC":"2021-01-16T17:55:57","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2021-01-16T18:54:32","SaveCount":12,"SaveAddress":"F8000"}}
19:26:58 MQT: stat/118/STATUS2 = {"StatusFWR":{"Version":"9.2.0(sensors)","BuildDateTime":"2020-12-21T15:04:56","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"382/699"}}
19:26:58 MQT: stat/118/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GueldeBestattungen",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
19:26:58 MQT: stat/118/STATUS4 = {"StatusMEM":{"ProgramSize":638,"Free":364,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
19:26:58 MQT: stat/118/STATUS5 = {"StatusNET":{"Hostname":"118-2032","IPAddress":"192.168.2.183","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.150","Mac":"C4:4F:33:C2:27:F0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:26:58 MQT: stat/118/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.154","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C227F0","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:26:58 MQT: stat/118/STATUS7 = {"StatusTIM":{"UTC":"2021-01-16T18:26:58","Local":"2021-01-16T19:26:58","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:37","Sunset":"17:22"}}
19:26:58 MQT: stat/118/STATUS10 = {"StatusSNS":{"Time":"2021-01-16T19:26:58","BME280":{"Temperature":21.7,"Humidity":93.6,"DewPoint":20.6,"Pressure":671.9},"PressureUnit":"hPa","TempUnit":"C"}}
19:26:58 MQT: stat/118/STATUS11 = {"StatusSTS":{"Time":"2021-01-16T19:26:58","Uptime":"0T00:31:01","UptimeSec":1861,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"XXXXXXX","BSSId":"7C:FF:4D:BA:03:93","Channel":11,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
no special output

```

### TO REPRODUCE
connect a bme280 to a sonoff basic

### EXPECTED BEHAVIOUR
show the correct values

### SCREENSHOTS
[https://imgur.com/a/vM5bmYT](url)
i don't know how to add a picture here sorry :-(

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

joba-1, 2021-01-16:
just a guess: looks like you see fahrenheit despite tasmota announcing celsius?

flobbr, 2021-01-16:
sadly no, i tried SetOption8 and 24 but nothing changes. Also if i convert the values, they don’t make sense either.

sfromis, 2021-01-16:
Yeah, the very low barometric pressure displayed is a clear thing of something bad going on.

Sonoff Basic is really, really, really not a great device to add stuff to, it can barely even power itself. My experience with BME280 is that they can "freeze" at a bad reading if power is not good and stable. Running BME280 on a mix of NodeMCU and D1 Mini, with good power, is that they work excellently and very stable on all Tasmota versions I've used.

closed 2021-01-16 by ascillato2