arendst/Tasmota #10591: DisplayText [p] padding not working

gituser-rk, 2021-01-17:
### PROBLEM DESCRIPTION
I'm using a ESP32 board with SSD1306 OLED. When in DisplayMode 0 and sending repeating DisplayText commands with padding like  "DisplayText [f2p5y20x25t]" to display actual time, the old pixels will not be cleared. 
Over time more and more pixel appear and one cannot read the output...
As far as I understand, the "p5" shoud overwrite/clear the first 5 characters from left which is in this case (time with minutes) the whole string sent to the display. Tried also with higher p values. Could this be a bug in version 9.2.0(display) for ESP32 or am I doing it wrong?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Wemos ESP32 0.96' OLED
- [x ] Tasmota binary firmware version number used: 9.2.0(display) for ESP32
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: flash_download_tools_v3.6.5.exe (Espressif)
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
stat/tasmota_6BBE6C/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
stat/tasmota_6BBE6C/RESULT = {"Module":{"1":"ESP32-DevKit"}}
stat/tasmota_6BBE6C/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"640":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"1376":"WS2812"},"GPIO16":{"1312":"DS18x20"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"}}}
12:48:56 MQT: stat/tasmota_6BBE6C/RESULT = {"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}}

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
12:54:21 MQT: stat/tasmota_6BBE6C/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":121,"Free":390,"Rules":"on tele-DS18B20#Temperature do DisplayText [f1p6y0x50]%value% C endon on Time#Minute|1 do DisplayText [f2p5y20x25t] endon"}
12:54:21 MQT: stat/tasmota_6BBE6C/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:54:22 MQT: stat/tasmota_6BBE6C/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:55:05 MQT: stat/tasmota_6BBE6C/STATUS = {"Status":{"Module":1,"DeviceName":"Schlafzimmer","FriendlyName":["Schlafzimmer","Schlafzimmer2"],"Topic":"tasmota_6BBE6C","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:55:05 MQT: stat/tasmota_6BBE6C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T01:18:46","StartupUTC":"2021-01-17T10:36:19","Sleep":50,"CfgHolder":4617,"BootCount":53,"BCResetTime":"2021-01-10T12:41:16","SaveCount":140}}
12:55:05 MQT: stat/tasmota_6BBE6C/STATUS2 = {"StatusFWR":{"Version":"9.2.0(display)","BuildDateTime":"2020-12-21T15:45:21","Core":"1_0_4_2","SDK":"v3.3.4-232-g44ec7972b","CpuFrequency":80,"Hardware":"ESP32","CR":"407/699"}}
12:55:05 MQT: stat/tasmota_6BBE6C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Things",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
12:55:05 MQT: stat/tasmota_6BBE6C/STATUS4 = {"StatusMEM":{"ProgramSize":1077,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAA858E","00041FA1","000000C3","00000000","00000080","00000020","00000000"],"Drivers":"1,2,4,5,8,9,10,13","Sensors":"1,2,5,6"}}
12:55:05 MQT: stat/tasmota_6BBE6C/STATUS5 = {"StatusNET":{"Hostname":"tasmota_6BBE6C-7788","IPAddress":"192.168.22.108","Gateway":"192.168.22.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.22.1","Mac":"30:AE:A4:6B:BE:6C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:55:05 MQT: stat/tasmota_6BBE6C/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.1.91","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6BBE6C","MqttUser":"mqtt","MqttCount":78,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:55:05 MQT: stat/tasmota_6BBE6C/STATUS7 = {"StatusTIM":{"UTC":"2021-01-17T11:55:05","Local":"2021-01-17T12:55:05","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:36","Sunset":"17:23"}}
12:55:05 MQT: stat/tasmota_6BBE6C/STATUS10 = {"StatusSNS":{"Time":"2021-01-17T12:55:05","DS18B20":{"Id":"03069779B3CB","Temperature":23.1},"TempUnit":"C"}}
12:55:05 MQT: stat/tasmota_6BBE6C/STATUS11 = {"StatusSTS":{"Time":"2021-01-17T12:55:05","Uptime":"0T01:18:46","UptimeSec":4726,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":78,"POWER1":"OFF","Dimmer":62,"Color":"9E9270","HSBColor":"45,29,62","Channel":[62,57,44],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Things","BSSId":"F2:63:DA:18:95:6E","Channel":6,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
repeating DisplayText commands - writing to same position of display different characters and using [px] padding for clearing the old pixels

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
the old pixels should be cleared prior to writing the new characters

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-01-17:
display text time does not need padding since the time has always the same number of chars.
however possibly the display is in transparency text mode
Displaytext [D0] sets the drawing mode to opaque. (only needed once)
i just saw that this flag is missing in the docs
it is in bit 1 of cmd 'D'

sfromis, 2021-01-17:
As underlined on Discord, your issue is not at all padding, but that previous pixels are retained. Better not report the wrong issue :-)

gemu2015, 2021-01-17:
the default is opaque mode and auto draw. May be you did set it by accident. i have updated the docs. (this applies to esp8266 and esp32)

sfromis, 2021-01-17:
Worked with D1 (to set bit 2 to 0). To confirm, I had tested it on ESP32 in a Tasmota setup where I had not messed around with the SSD1306. Looks like the default was different, I'm pretty sure that I had not done a D command.

gemu2015, 2021-01-17:
ok, will do a pr to assure the default is always zero. transparency  is rarely needed. i introduced it to draw text over pictures in color displays. probably should force to opaque on bw displays

ascillato2, 2021-01-17:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-17 by ascillato2

gituser-rk, 2021-01-17:
Thanks for the clarification and quick response. I can confirm that it works as expected after a "DisplayText [D1]"