arendst/Tasmota #10610: Mitsubishi Electric Integration Missing Mode

maslyankov, 2021-01-18:
### PROBLEM DESCRIPTION
I am using the Mitsubishi Electric module and it is working fine - turns on the AC to each mode, sets temp, etc, but it does not show the current mode the AC is in, it only shows ON or OFF.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [x] Tasmota binary firmware version number used: 9.1.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: espflashtool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:19:56 CMD: Backlog Template; Module; GPIO 255
19:19:57 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:25:57","MiElHVAC":{"Temperature":19.0,"Operation":"OFF","Compressor":"OFF","roomtemp":"030000090001A6000000000000000000","status":"06000000000000000000000000000000","stage":"09000000000000000000000000000000"},"TempUnit":"C"}
19:19:57 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:26:27","MiElHVAC":{"Temperature":19.0,"Operation":"OFF","Compressor":"OFF","roomtemp":"030000090001A6000000000000000000","status":"06000000000000000000000000000000","stage":"09000000000000000000000000000000"},"TempUnit":"C"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:20:27 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:20:27","MiElHVAC":{"Temperature":19.0,"Operation":"OFF","Compressor":"OFF","roomtemp":"030000090001A6000000000000000000","status":"06000000000000000000000000000000","stage":"09000000000000000000000000000000"},"TempUnit":"C"}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

AC Model: Mitsubishi Electric MSZ-FH35VE
logs:
19:21:27 MQT: tele/tasmota_5F410B/HVACSETTINGS = {"Power":"ON","Temp":24.0,"FanSpeed":"auto","SwingV":"swing","SwingH":"RR","Bytes":"0200000109070007000005B000000000"}
19:21:29 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:21:29","MiElHVAC":{"Temperature":19.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000009006CA6000000000000000000","status":"06000006010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
19:21:37 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:21:37","MiElHVAC":{"Temperature":19.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000009006CA6000000000000000000","status":"06000007010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
19:21:47 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:21:47","MiElHVAC":{"Temperature":19.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000009006CA6000000000000000000","status":"06000008010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
19:21:55 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:21:55","MiElHVAC":{"Temperature":18.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000008006CA5000000000000000000","status":"06000008010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
19:21:56 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:21:56","MiElHVAC":{"Temperature":18.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000008006CA5000000000000000000","status":"06000009010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
19:21:57 MQT: tele/tasmota_5F410B/STATE = {"Time":"1970-01-01T19:21:57","Uptime":"0T19:21:59","UptimeSec":69719,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"wifiname","BSSId":"77:DD:88:00:DD:EE","Channel":11,"RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:21:57 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:21:57","MiElHVAC":{"Temperature":18.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000008006CA5000000000000000000","status":"06000009010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
19:21:57 MQT: tele/tasmota_5F410B/HVACSETTINGS = {"Power":"ON","Temp":24.0,"FanSpeed":"auto","SwingV":"swing","SwingH":"RR","Bytes":"0200000109070007000005B000000000"}
19:22:07 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:22:07","MiElHVAC":{"Temperature":18.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000008006EA5000000000000000000","status":"06000009010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
19:22:11 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:22:11","MiElHVAC":{"Temperature":18.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000008006EA5000000000000000000","status":"0600000A010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
19:22:12 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:22:12","MiElHVAC":{"Temperature":18.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000008006CA5000000000000000000","status":"0600000A010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
19:22:27 MQT: tele/tasmota_5F410B/STATE = {"Time":"1970-01-01T19:22:27","Uptime":"0T19:22:29","UptimeSec":69749,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"wifiname","BSSId":"77:DD:88:00:DD:EE","Channel":11,"RSSI":32,"Signal":-84,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:22:27 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:22:27","MiElHVAC":{"Temperature":18.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000008006CA5000000000000000000","status":"0600000A010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
19:22:27 MQT: tele/tasmota_5F410B/HVACSETTINGS = {"Power":"ON","Temp":24.0,"FanSpeed":"auto","SwingV":"swing","SwingH":"RR","Bytes":"0200000109070007000005B000000000"}
19:22:45 MQT: tele/tasmota_5F410B/SENSOR = {"Time":"1970-01-01T19:22:45","MiElHVAC":{"Temperature":18.0,"Operation":"ON","Compressor":"ON","roomtemp":"03000008006EA5000000000000000000","status":"0600000A010000000000000000000000","stage":"09000004010000000000000000000000"},"TempUnit":"C"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
I have another ESP8266 running the same binary connected to a Mitsubishi Electric MSZ-SF50VE3 with no issues.
Working AC logs:
```
20:00:24 MQT: tele/tasmota_B0FCB0/HVACSETTINGS = {"Power":"ON","Mode":"heat","HAMode":"heat","Temp":23.0,"FanSpeed":"auto","SwingV":"5","SwingH":"C","Bytes":"0200000101080005000003AE00000000"}
20:01:08 MQT: tele/tasmota_B0FCB0/SENSOR = {"Time":"1970-02-06T20:01:08","MiElHVAC":{"Temperature":22.0,"Operation":"ON","Compressor":"ON","roomtemp":"0300000C0000AD000000000000000000","status":"06000012010000000000000000000000","stage":"09000000020000000000000000000000"},"TempUnit":"C"}
20:01:19 MQT: tele/tasmota_B0FCB0/SENSOR = {"Time":"1970-02-06T20:01:19","MiElHVAC":{"Temperature":22.0,"Operation":"ON","Compressor":"ON","roomtemp":"0300000C0000AD000000000000000000","status":"06000013010000000000000000000000","stage":"09000000020000000000000000000000"},"TempUnit":"C"}
```

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

maslyankov, 2021-02-12:
#9237
Still having the issue

ascillato2, 2021-02-12:
Moving this issue to Discussions as it is a feature request

closed 2021-02-12 by ascillato2