arendst/Tasmota #10611: ZbBridge + NEO window sensor = delay 4 sec

bogorad, 2021-01-18:
### PROBLEM DESCRIPTION
ZbBridge with NEO zigbee window senros gets data from Zigbee but produces a Contact with a delay of ~4 seconds. Possibly a parsing issue. Zigbee2mqtt has no suck issue with this same sensor.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ZbBridge
- [x] Tasmota binary firmware version number used: 9.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:47:15 CMD: Backlog Template; Module; GPIO 255
13:47:15 MQT: housematix/zigbee/A4CF12D7FB80/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
13:47:15 MQT: housematix/zigbee/A4CF12D7FB80/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
13:47:15 MQT: housematix/zigbee/A4CF12D7FB80/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:47:42 CMD: Status 0
13:47:42 MQT: housematix/zigbee/A4CF12D7FB80/STATUS = {"Status":{"Module":75,"DeviceName":"sonoff-zigbee","FriendlyName":["sonoff-zigbee"],"Topic":"A4CF12D7FB80","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:47:42 MQT: housematix/zigbee/A4CF12D7FB80/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:13:05","StartupUTC":"2021-01-18T12:34:37","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2020-11-15T23:38:14","SaveCount":98,"SaveAddress":"F6000"}}
13:47:42 MQT: housematix/zigbee/A4CF12D7FB80/STATUS2 = {"StatusFWR":{"Version":"9.2.0(zbbridge)","BuildDateTime":"2020-12-17T10:08:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"435/699"}}
13:47:42 MQT: housematix/zigbee/A4CF12D7FB80/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LLULL188A","caprica"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00206080","00000000"]}}
13:47:42 MQT: housematix/zigbee/A4CF12D7FB80/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
13:47:42 MQT: housematix/zigbee/A4CF12D7FB80/STATUS5 = {"StatusNET":{"Hostname":"A4CF12D7FB80-7040","IPAddress":"192.168.86.22","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.86.1","Mac":"A4:CF:12:D7:FB:80","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:47:42 MQT: housematix/zigbee/A4CF12D7FB80/STATUS6 = {"StatusMQT":{"MqttHost":"housematix.com","MqttPort":8883,"MqttClientMask":"%012X","MqttClient":"A4CF12D7FB80","MqttUser":"zbbridge","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:47:43 MQT: housematix/zigbee/A4CF12D7FB80/STATUS7 = {"StatusTIM":{"UTC":"2021-01-18T12:47:42","Local":"2021-01-18T13:47:42","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:35","Sunset":"17:25"}}
13:47:43 MQT: housematix/zigbee/A4CF12D7FB80/STATUS10 = {"StatusSNS":{"Time":"2021-01-18T13:47:43"}}
13:47:43 MQT: housematix/zigbee/A4CF12D7FB80/STATUS11 = {"StatusSTS":{"Time":"2021-01-18T13:47:43","Uptime":"0T00:13:06","UptimeSec":786,"Vcc":3.499,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"caprica","BSSId":"D8:6C:63:F3:AB:D5","Channel":6,"RSSI":100,"Signal":-30,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
https://pastebin.com/YLaF9iYw

13:41:23 CMD: weblog 4
13:41:23 SRC: WebConsole from 192.168.86.150
13:41:23 CMD: Group 0, Index 1, Command "WEBLOG", Data "4"
13:41:23 MQT: housematix/zigbee/A4CF12D7FB80/RESULT = {"WebLog":4}
13:41:23 CFG: Saved to flash at F7, Count 97, Bytes 4096
13:41:23 WIF: Checking connection...
13:41:27 ZIG: {"ZbEZSPReceived2":"2645900145000004010005010100010000F6FFEFA75CFFFF070932011500021002"}
13:41:27 ZIG: ZbEZSPSentRaw 83
13:41:27 ZIG: {"ZbEZSPReceived":"45000004010005010100010000F6FFEFA75CFFFF070932011500021002"}
13:41:27 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x5CA7","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":183,"securityuse":0,"seqnumber":246,"fc":"0x09","manuf":"0x0000","transact":50,"cmdid":"0x01","payload":"15000210"}}
13:41:27 ZIG: ZbZCLRawReceived: {"0x5CA7":{"0500<01":"15000210","Endpoint":1,"LinkQuality":183}}
13:41:27 MQT: housematix/zigbee/A4CF12D7FB80/5CA7/SENSOR = {"ZbReceived":{"0x5CA7":{"Device":"0x5CA7","Name":"842E14FFFEE34EBF","0500<01":"15000210","Endpoint":1,"LinkQuality":183}}}
13:41:28 ZIG: {"ZbEZSPReceived2":"3645900145000004010005010100010000F7FFF3A75CFFFF070932011500021002"}
13:41:28 ZIG: ZbEZSPSentRaw 84
13:41:28 ZIG: {"ZbEZSPReceived":"45000004010005010100010000F7FFF3A75CFFFF070932011500021002"}
13:41:28 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x5CA7","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":194,"securityuse":0,"seqnumber":247,"fc":"0x09","manuf":"0x0000","transact":50,"cmdid":"0x01","payload":"15000210"}}
13:41:28 ZIG: ZbZCLRawReceived: {"0x5CA7":{"0500<01":"15000210","Endpoint":1,"LinkQuality":194}}
13:41:28 MQT: housematix/zigbee/A4CF12D7FB80/5CA7/SENSOR = {"ZbReceived":{"0x5CA7":{"Device":"0x5CA7","Name":"842E14FFFEE34EBF","0500<01":"15000210","Endpoint":1,"LinkQuality":194}}}
13:41:31 ZIG: {"ZbEZSPReceived2":"4645900145000004010005010100010000F8FFEFA75CFFFF09093300000000FF000002"}
[... 26 lines cut ...]
13:41:46 ZIG: {"ZbEZSPReceived":"45000004010005010100010000FCFFEAA75CFFFF09093500010000FF000002"}
13:41:46 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x5CA7","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":170,"securityuse":0,"seqnumber":252,"fc":"0x09","manuf":"0x0000","transact":53,"cmdid":"0x00","payload":"010000FF0000"}}
13:41:46 ZIG: ZbZCLRawReceived: {"0x5CA7":{"0500<00":"010000FF0000","ZoneStatusChange":1,"0500/FFF2":1,"Endpoint":1,"LinkQuality":170}}
13:41:46 MQT: housematix/zigbee/A4CF12D7FB80/5CA7/SENSOR = {"ZbReceived":{"0x5CA7":{"Device":"0x5CA7","Name":"842E14FFFEE34EBF","0500<00":"010000FF0000","ZoneStatusChange":1,"Contact":1,"Endpoint":1,"LinkQuality":170}}}
13:41:47 ZIG: {"ZbEZSPReceived2":"1645900145000004010005010100010000FDFFEAA75CFFFF09093500010000FF000002"}
13:41:47 ZIG: ZbEZSPSentRaw 82
13:41:47 ZIG: {"ZbEZSPReceived":"45000004010005010100010000FDFFEAA75CFFFF09093500010000FF000002"}
13:41:47 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x5CA7","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":170,"securityuse":0,"seqnumber":253,"fc":"0x09","manuf":"0x0000","transact":53,"cmdid":"0x00","payload":"010000FF0000"}}
[... cut ...]
13:41:47 MQT: housematix/zigbee/A4CF12D7FB80/5CA7/SENSOR = {"ZbReceived":{"0x5CA7":{"Device":"0x5CA7","Name":"842E14FFFEE34EBF","0500<00":"010000FF0000","ZoneStatusChange":1,"Contact":1,"Endpoint":1,"LinkQuality":170}}}
```

### TO REPRODUCE
Pair NEO zigbee and play with the magnet.

### EXPECTED BEHAVIOUR
Should get the `Contact` messages immediately.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-01-18:
Thanks for reporting. It may be caused by the device expecting a response to the command, and Tasmota does not send it (which I wll add).

Just to make sure, could you please update to the latest version from the development branch?
You can download from http://ota.tasmota.com/tasmota/

s-hadinger, 2021-01-18:
Can you try with the latest PR #10624

closed 2021-01-18 by s-hadinger

bogorad, 2021-01-18:
With `9.2.0.3`: it looks like the delay is gone.

Still sends two MQTT commands:

```
21:40:49.352 ZIG: {"ZbEZSPReceived":"45000004010005010100010000A9FFE8A75CFFFF09092C0000000001000002"}
21:40:49.354 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0500","srcaddr":"0x5CA7","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":165,"securityuse":0,"seqnumber":169,"fc":"0x09","frametype":1,"direction":1,"disableresp":0,"manuf":"0x0000","transact":44,"cmdid":"0x00","payload":"000000010000"}}
21:40:49.357 ZIG: ZbZCLRawReceived: {"0x5CA7":{"0500<00":"000000010000","ZoneStatusChange":0,"ZoneStatusChangeZone":1,"0500/FFF2":0,"Endpoint":1,"LinkQuality":165}}
21:40:49.414 MQT: housematix/zigbee/A4CF12D7FB80/5CA7/SENSOR = {"ZbReceived":{"0x5CA7":{"Device":"0x5CA7","Name":"842E14FFFEE34EBF","0500<00":"000000010000","ZoneStatusChange":0,"ZoneStatusChangeZone":1,"Contact":0,"Endpoint":1,"LinkQuality":165}}}
21:40:50.562 ZIG: {"ZbEZSPReceived":"45000004010005010100010000AAFFECA75CFFFF09092C0000000001000002"}
21:40:50.563 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0500","srcaddr":"0x5CA7","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":175,"securityuse":0,"seqnumber":170,"fc":"0x09","frametype":1,"direction":1,"disableresp":0,"manuf":"0x0000","transact":44,"cmdid":"0x00","payload":"000000010000"}}
21:40:50.566 ZIG: ZbZCLRawReceived: {"0x5CA7":{"0500<00":"000000010000","ZoneStatusChange":0,"ZoneStatusChangeZone":1,"0500/FFF2":0,"Endpoint":1,"LinkQuality":175}}
21:40:50.635 MQT: housematix/zigbee/A4CF12D7FB80/5CA7/SENSOR = {"ZbReceived":{"0x5CA7":{"Device":"0x5CA7","Name":"842E14FFFEE34EBF","0500<00":"000000010000","ZoneStatusChange":0,"ZoneStatusChangeZone":1,"Contact":0,"Endpoint":1,"LinkQuality":175}}}
21:40:59.051 ZIG: {"ZbEZSPReceived":"45000004010005010100010000ABFFF6A75CFFFF09092D0001000001000002"}
21:40:59.053 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0500","srcaddr":"0x5CA7","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":202,"securityuse":0,"seqnumber":171,"fc":"0x09","frametype":1,"direction":1,"disableresp":0,"manuf":"0x0000","transact":45,"cmdid":"0x00","payload":"010000010000"}}
21:40:59.056 ZIG: ZbZCLRawReceived: {"0x5CA7":{"0500<00":"010000010000","ZoneStatusChange":1,"ZoneStatusChangeZone":1,"0500/FFF2":1,"Endpoint":1,"LinkQuality":202}}
21:40:59.114 MQT: housematix/zigbee/A4CF12D7FB80/5CA7/SENSOR = {"ZbReceived":{"0x5CA7":{"Device":"0x5CA7","Name":"842E14FFFEE34EBF","0500<00":"010000010000","ZoneStatusChange":1,"ZoneStatusChangeZone":1,"Contact":1,"Endpoint":1,"LinkQuality":202}}}
21:41:00.262 ZIG: {"ZbEZSPReceived":"45000004010005010100010000ACFFF3A75CFFFF09092D0001000001000002"}
21:41:00.264 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0500","srcaddr":"0x5CA7","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":194,"securityuse":0,"seqnumber":172,"fc":"0x09","frametype":1,"direction":1,"disableresp":0,"manuf":"0x0000","transact":45,"cmdid":"0x00","payload":"010000010000"}}
21:41:00.267 ZIG: ZbZCLRawReceived: {"0x5CA7":{"0500<00":"010000010000","ZoneStatusChange":1,"ZoneStatusChangeZone":1,"0500/FFF2":1,"Endpoint":1,"LinkQuality":194}}
21:41:00.366 MQT: housematix/zigbee/A4CF12D7FB80/5CA7/SENSOR = {"ZbReceived":{"0x5CA7":{"Device":"0x5CA7","Name":"842E14FFFEE34EBF","0500<00":"010000010000","ZoneStatusChange":1,"ZoneStatusChangeZone":1,"Contact":1,"Endpoint":1,"LinkQuality":194}}}
```

s-hadinger, 2021-01-19:
You are right. There is an internal deduplication of received commands with same transaction id (which is the case here), but the time window is set to 200ms by default in `my_user_config.h`:

`#define USE_ZIGBEE_DEBOUNCE_COMMANDS   200`

In your case, the duplicate command arrives 1200 ms later. Can you try setting the following?

`#define USE_ZIGBEE_DEBOUNCE_COMMANDS   1500`

MattWestb, 2021-01-19:
What i can see is it one groupe command that is send to the coordinator and in the zigbee world its is done with broadcast and is being sended 3 times for being sure its is coming to the receiving device (broadcast is not acked then the sender dont knowing how many is shud receiving it). I also think its depend now many links is getting it and forwarding it to the coordinator (only one router = 1 message, 3 routers can being 3 message then its coming from 3 different links) but that is the `seqnumber` filtering out so you only getting one from the router part of the coordinator.
So we is getting the same message received with with different `seqnumber` because the routers / mesh must treating them as separate messages then forwarding them (and not forwarding them back to the sender and getting one loop).
But all messages have the same "transact":45 that i think its transaction number. I think its good doing one filtering that only one message with the same `transact` is being published and the rest is being trached in the black hole.

Is the theory worth trying Stepan ?

PS: ZHA was having the same problem then counting "clicks"from one Philips HUE dimmer switch that was counting all broadcasted "clicks" and was one big mess for the user then counting them.

s-hadinger, 2021-01-19:
Yes, the `seqnumber` is a Zigbee packet number. It is incremented since the device actually sends 2 packets.

However the `transact` number is ZCL specific and made to distinguish separate messages, disregarding resent. That's why it is correct to discard the second packet coming from the same source, same endpoint, same cluster with same transact number.

To avoid any bad side effect or device misbehaving and reusing the same `transact` (who knows what creative developers can do), I enabled the deduplication for only 200ms. This device has a slower resending scheme.

PR #10624 (2021-01-18): Zigbee send ack to command
## Description:

Send default response when a command is received without disable response.

**Related issue (if applicable):** fixes #10611

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.5-rc4
  - [x] I accept the [CLA](https://github.com/arendst/Tas [... cut]