arendst/Tasmota #10620: Erweiterung der 5x PWM Kanäle für ESP32bin

Basti149, 2021-01-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Schön guten Abend. Tasmota ist einfach Klasse und einfach zu bedienen.
Ich bräuchte für mein neues Projekt mehr als 5 PWM Kanäle. Könnte man ca. 2 weitere Kanäle (weiß Licht mit einfachen Regler würde zureichen) in die nächste Tasmota Version integrieren. Und zwar in die EP32.bin da dieser bekanntlich mehr als 5 PWM Kanäle zulassen würde.

Mit freundlichen Grüßen Basti

Good evening. Tasmota is just great and easy to use.
I would need more than 5 PWM channels for my new project. If you could integrate about 2 more channels (white light with a simple controller would be sufficient) in the next Tasmota version. Namely in the EP32.bin as it is known that this would allow more than 5 PWM channels.

Sincerely, Basti

ascillato2, 2021-01-18:
Closing this request as it is **duplicated**. Sorry.

If you want to reopen the discussion, please  do it in discussions. Github issues are for software problems reports.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-18 by ascillato2

Basti149, 2021-01-18:
> Closing this request as it is **duplicated**. Sorry.
>
> If you want to reopen the discussion, please do it in discussions. Github issues are for software problems reports.
>
> Remember to read the [...]

Gehts noch