arendst/Tasmota #10632: Tasmota and DHT12 with negative temperature

hassebjork, 2021-01-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
The DHT12 sensor measures temperature in the range -20 to +60 degrees Celsius. The code does not support negative readings.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
When the temperature goes just below zero, the readings will be +15 to +16 degrees

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Change the code in xsns_58_dht12.ino line 60 to the following:

  Dht12.temperature = ConvertTemp( ( (float) temp + (float) ( tempTenth & 0x7F ) /(float) 10.0 ) 
    * ( tempTenth & 0x80 ? -1.0 : 1.0 ) );

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-01-19:
Have you tried your provided solution? If yes why not providing a PR?
Just click on the file and hit the pencil and do your edits. 
Doing this you have generated a PR you just have to submit.

hassebjork, 2021-01-19:
I have done the changes, but not tried readings below 0 degrees. The sensor is mounted in a chicken coop and I do not expect the temperature to drop below zero until the weekend. I am not a programmer and have not read the Tasmota code...

hassebjork, 2021-01-19:
OK! Tested with a frozen pack down to -1.0 degrees and working! The code is not pretty, but working. (I never got the checksum working).

Thank you for your support!

closed 2021-01-19 by hassebjork