arendst/Tasmota #10634: SM2135: Support 15W bulbs using higher current

arendst, 2021-01-20:
I have 3 models of RGBCCT light bulbs (5W, 13W and 15W) that use SM2135 LED driver. 

Found out that I had to modify the current setting for these light bulbs in order to output the correct brigtness.

It would be good if the current setting for component SM2135 can be altered using command instead of compiling custom firmwares for each type, with the lowest current as the default.

_Originally posted by @VaThInK in https://github.com/arendst/Tasmota/discussions/10570_

closed 2021-01-20 by close-issue-app[bot]

arendst, 2021-01-20:
The fix makes limited SM2135 current selection possible by using an index to GPIO ``SM2135 DAT`` like:
- 1 = GRB 20 mA / CW 15 mA = Original LSC E14 bulb
- 2 = BGR 20 mA / CW 15 mA
- 3 = GRB 20 mA / CW 30 mA
- 4 = BGR 20 mA / CW 30 mA
- 5 = GRB 45 mA / CW 60 mA
- 6 = BGR 45 mA / CW 60 mA

Be careful NOT TO SELECT higher currents on devices which do not support this based on their power supply.

asozio, 2022-08-01:
It would appear the bulb I am working on (Arlec Smart 10W 830lm RGBCCT Bulb GLD320HA) is configured as a GBR rather than any of the options above. Is it possible to update the SM2135 options to include this configuration as well?

barbudor, 2022-08-01:
I don't think it is going to be possible to update the list of index without a breaking change
However it may be possible to implement something based on SetOption37
Something like
If SO37 == 0 (default)
   Use mapping and current from GPIO Index
Else
  Use current from GPIO Index but mapping from SO37

@arendst 
Does it makes sense ?
Would you like me to try a PR ?

asozio, 2022-09-03:
@barbudor Given that I only had one of these bulbs and it would require some modifications that use a SetOption I just rewired the connections from the G/B on the SM2135 to be switched (Pins 1/8). This way I can use the standard configuration of tasmota by @arendst above.

barbudor, 2022-09-03:
Great hack @asozio!

asozio, 2022-09-03:
Turns out I had to do another, so tested that one with the above SetOption37 suggestion. Looks like GLD130HA and potentially GLD320HA both would work with SetOption37 7 (as detailed in https://tasmota.github.io/docs/SetOption37/

barbudor, 2022-09-03:
Are these also based on SM2135 ? I was under the impression that the SM2135 was not using SetOption37
The template for [GLD320HA](https://templates.blakadder.com/arlec_GLD320HA.html) is incomplete and I see none for the GLD130HA

asozio, 2022-09-03:
I thought I had updated the GLD320HA page, but must have not committed it. 
The config I was using for the GLD320HA was:
 **{"NAME":"DMP-L1 Light","GPIO":[0,0,0,0,4067,0,0,0,0,4032,0,0,0,0],"FLAG":0,"BASE":18}**

For the GLD132HA (Edison Screw version of the GLD130HA I believe) I am using:
 **{"NAME":"GLD130HA","GPIO":[0,0,0,0,4069,0,0,0,0,4032,0,0,0,0],"FLAG":0,"BASE":18}** and **"SetOption37 7"**
although I am have not confirmed the LED current so would suggest lowering the SM2135 option from 6.