arendst/Tasmota #10644: OBIS smart meter gas reading; 2 issues

stevevanhoyweghen, 2021-01-20:
### PROBLEM DESCRIPTION
P1 smart meter (https://tasmota.github.io/docs/P1-Smart-Meter/) used in combination with a Sagemcom T211-D meter (OBIS code, from Fluvius in Belgium). All readings are correct, except the gas related readings.

1. The gas format is different from the other formats used and there is no option to parse the gas meter value.
2. If parsing the gas meter timestamp (every 10 minutes), the electricity timestamp (every few seconds) is overwritten.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X]Device used (e.g., Sonoff Basic): Wemos D1
- [X] Tasmota binary firmware version number used: Tasmota 9.2.0
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
20:16:41 MQT: stat/tasmota_14735B/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
20:16:41 MQT: stat/tasmota_14735B/RESULT = {"Module":{"0":"Generic"}}
20:16:42 MQT: stat/tasmota_14735B/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"1312":"DS18x20"},"GPIO17":{"0":"None"}}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
stat/tasmota_14735B/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_14735B","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:18:54 MQT: stat/tasmota_14735B/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:17:26","StartupUTC":"2021-01-20T19:01:28","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2021-01-16T21:39:34","SaveCount":391,"SaveAddress":"3F4000"}}
20:18:55 MQT: stat/tasmota_14735B/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2021-01-16T20:36:32","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"402/699"}}
20:18:55 MQT: stat/tasmota_14735B/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["kastanje32_low","alternative"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00206000","00000000"]}}
20:18:55 MQT: stat/tasmota_14735B/STATUS4 = {"StatusMEM":{"ProgramSize":608,"Free":2460,"Heap":11,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","07C045C6","042E0001","600207C3","180C0000","D032F989","0009200C","00001000"],"Drivers":"1,2,4,6,8,9,10,12,18,19,22,24,26,27,29,30,35,37,45","Sensors":"1,2,4,5,6,7,8,9,14,26,27,38,39,46,47,53,55,58,59,63,65,72,73,74"}}
20:18:55 MQT: stat/tasmota_14735B/STATUS5 = {"StatusNET":{"Hostname":"tasmota_14735B-4955","IPAddress":"192.168.0.242","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"D8:F1:5B:14:73:5B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:18:56 MQT: stat/tasmota_14735B/STATUS6 = {"StatusMQT":{"MqttHost":"zone55.duckdns.org","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_14735B","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:18:56 MQT: stat/tasmota_14735B/STATUS7 = {"StatusTIM":{"UTC":"2021-01-20T19:18:56","Local":"2021-01-20T20:18:56","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:34","Sunset":"17:28"}}
20:18:56 MQT: stat/tasmota_14735B/STATUS10 = {"StatusSNS":{"Time":"2021-01-20T20:18:56","P1":{"electr_time":"210120201857W","tarif1":1635.590,"tarif2":1541.157,"tariff":1,"power":0.589,"volt_l1":229.1,"volt_l2":226.3,"volt_l3":229.3,"amp_l1":1.0,"amp_l1":0.0,"amp_l1":0.0}}}
20:18:56 MQT: stat/tasmota_14735B/STATUS11 = {"StatusSTS":{"Time":"2021-01-20T20:18:56","Uptime":"0T00:17:28","UptimeSec":1048,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"kastanje32_low","BSSId":"E4:F4:C6:03:CB:49","Channel":11,"RSSI":100,"Signal":-33,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
20:20:57 MQT: stat/tasmota_14735B/RESULT = {"WebLog":4}
20:20:58 CFG: Saved to flash at 3F3, Count 392, Bytes 4096
20:21:14 WIF: Checking connection...
20:21:34 WIF: Checking connection...
20:21:37 MQT: tele/tasmota_14735B/STATE = {"Time":"2021-01-20T20:21:37","Uptime":"0T00:20:09","UptimeSec":1209,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"kastanje32_low","BSSId":"E4:F4:C6:03:CB:49","Channel":11,"RSSI":100,"Signal":-28,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:21:37 MQT: tele/tasmota_14735B/SENSOR = {"Time":"2021-01-20T20:21:37","P1":{"electr_time":"210120202140W","tarif1":1635.616,"tarif2":1541.157,"tariff":1,"power":0.562,"volt_l1":228.9,"volt_l2":225.9,"volt_l3":229.8,"amp_l1":1.0,"amp_l1":0.0,"amp_l1":0.0}}
20:21:54 WIF: Checking connection...
```

### TO REPRODUCE
**Meter output example:**
```lua
1-0:31.7.0(001*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(31235C0F39123131233335763688)
0-1:24.4.0(1)
0-1:24.2.3(210117125004W)(01524.450*m3)
```

**Remarks**

1. Meter id changed for privacy reasons '0-1:96.1.1(31235C0F39123131233335763688)'
2. Gas reading:timestamp (W->wintertime), gas meter reading '0-1:24.2.3(210117125004W)(01524.450*m3)'

**Script**
```lua
>D
>B
>M 1
+1,3,o,0,115200,P1
1,0-0:1.0.0(@#),Electr. update,,electr_time,0
1,1-0:1.8.1(@1,Tariff High,kWh,tarif1,3
1,1-0:1.8.2(@1,Tariff Low,kWh,tarif2,3
1,0-0:96.14.0(@1,Tariff,,tariff,0
1-0:1.7.0(@1,Power,kW,power,3
1,1-0:32.7.0(@1,Voltage L1,V,volt_l1,1
1,1-0:52.7.0(@1,Voltage L2,V,volt_l2,1
1,1-0:72.7.0(@1,Voltage L3,V,volt_l3,1
1,1-0:31.7.0(@1,Current L1,A,amp_l1,1
1,1-0:51.7.0(@1,Current L2,A,amp_l1,1
1,1-0:71.7.0(@1,Current L3,A,amp_l1,1
1,0-1:24.2.3(@#),Gas update,,gas_time,0
#
```

### EXPECTED BEHAVIOUR

1. '1,0-1:24.2.3(@#),Gas update,,gas_time,0' value is correct, but the '1,0-0:1.0.0(@#),Electr. update,,electr_time,0' value is overwritten by the gas time stamp. If '1,0-1:24.2.3(@#),Gas update,,gas_time,0' is commented out, then the '1,0-0:1.0.0(@#),Electr. update,,electr_time,0' value is correct in the web interface and MQTT.
2. The syntax available has no feature to read the gas meter value. I found a workaround to some extent. '1,0-1:24.2.3(@#,Gas update,,gas_time,0', so removing the closing ')' shows this: '210120204002W)(01564.809'. Proposal: maybe syntax like this to skip the first braces pair: '1,0-1:24.2.3(@!)(@1,Gas,m3,gas,3' could result in the gas meter reading.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT

1. The screenshot shows the workaround and also demonstrates that the timestamp value of the electricity meter is overwritten. If normal syntax is used the reading is 'P1 Electr. update: "210120205505W"'
2. With some scripting you could do some extra parsing as work around. However previous issue 1 is still remaining.

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-01-21:
there is only one # (string) reading possible per meter, as designed (no error)

gemu2015, 2021-01-21:
excerpt from smart meter doku:

When decoding a string result (e.g., a serial meter), use # character for this parameter (only in one line per meter).

ascillato2, 2021-01-21:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-21 by ascillato2

gemu2015, 2021-01-21:
please try this version.  the ( char shall indicate that the 1. bracket shall be skipped and the 2. open bracket is searched

0-1:24.2.3(210117125004W)(01524.450*m3)
e.g.

1,0-1:24.2.3(@(1,Gas,m3,gas_val,3

should result in 1524.450 m3

[xsns_53_sml.ino.zip](https://github.com/arendst/Tasmota/files/5851841/xsns_53_sml.ino.zip)

stevevanhoyweghen, 2021-01-21:
@ascillato2 In full 'When decoding a string result (e.g., a serial meter), use # character for this parameter (only in one line per meter). For OBIS, you need a ) termination character after the # character'. I have OBIS and this is exactly my point. '(@#)' only reads the timestamp, not (that I can find) tthe actual gas m3 reading. I tried with sml[x] in the mean time, but no luck. I will try the zip provided. However, was compiling with tasmocompiler. So need to figure out how to compile ...
So in conclusion, there is a lacking feature for OBIS meters. So if the fix works it could be considered to add to the next release. I leave it to you to reopen.
@gemu2015, thanks for your support :-)

gemu2015, 2021-01-22:
@stevevanhoyweghen 
i have on obis simulator and will evaluate this today. i will then do a pr to tasmota. so hold on.

stevevanhoyweghen, 2021-01-22:
@gemu2015: Thanks for the code change! I was able to compile the development branch and indeed the gas meter reading is working. Thanks a lot! Now I can proceed.

However, I have two remaining remarks.

1. The first problem mentioned still remains. '1,0-1:24.2.3(@#),Gas update,,gas_time,0' value is correct, but the '1,0-0:1.0.0(@#),Electr. update,,electr_time,0' value is overwritten by the gas time stamp. If '1,0-1:24.2.3(@#),Gas update,,gas_time,0' is commented out, then the '1,0-0:1.0.0(@#),Electr. update,,electr_time,0' value is correct in the web interface and MQTT.
2. I inspected the code and a Linux diff revealed the code change:
1561,1570c1561
<               if (*mp == '(') {
<                 // skip this bracket
<                 char *bp = strchr(cp, '(');
<                 if (bp) {
<                   cp = bp + 1;
<                 }
<                 dval=CharToDouble((char*)cp);
<               } else {
<                 dval=CharToDouble((char*)cp);
<               }
---
>               dval=CharToDouble((char*)cp);

Long time ago I developed in C and reading the code I see that only the first '(' brace can be skipped.
Looking at the specification used in Belgium and The Netherlands (https://www.netbeheernederland.nl/_upload/Files/Slimme_meter_15_a727fce1f1.pdf) is see for example on page 25 examples of more complicated parentheses.
E.g. 1-0:99.97.0(2)(0-0:96.7.19)(101208152415W)(0000000240*s)(101208151004W)(0000000301*s), for the Power Failure Event Log (long power failures). I am not interested in this myself, but just as a remark. A while loop could could fix this. Another approach could be to allow regular expression, but this could take too much code footprint.

In conclusion, both 'issues' are minimal, and I and the change introduced fixes my main problem. So my remarks are just suggestions for future improvements.

gemu2015, 2021-01-23:
this driver implements a universal decoder for many different meters. so no exhaustive solution for all meter applications intended. it is designed to fit for 99 percent of use cases.

remark to obis timestamps:
why do you need them ? tasmota has its own timestamp.
and when you insist why don't you read the timestamps as numbers and decompose them (you only will miss the W for winter time)

stevevanhoyweghen, 2021-01-23:
Understood and fully agree with your statements. It is probably not worth to take the effort.
In the meantime tried to calculate the m3/h. No success, scripting causes issues. However, straightforward to do in Home Assistant or Node-Red.
Thanks again for the effort. I will document my setup once done and add to the appropriate documentation as a contribution so Dutch/Belgian OBIS meter owners can benefit.

EricVdB, 2022-10-26:
@stevevanhoyweghen did you get the gas reading working? Can you share your final script? I tried everything I found but the gas is not coming through.
Thx in advance

stevevanhoyweghen, 2022-10-26:
> @stevevanhoyweghen did you get the gas reading working? Can you share your final script? I tried everything I found but the gas is not coming through. Thx in advance

Hi,
I never managed to get it working without issues. It is buggy at best ... I moved on to greener pastures. ESPhome. I also implemented my garage door opener and Ikea VINDRIKTNING Particulate Matter Sensor PM1006 in combination with home assistant and dropped Tasmota altogether. Garage was working in Tasmota, but I decided to align to one platform. If you want more info, you can contact me on https://stevevanhoyweghen.wordpress.com/contact/.

I also added a temperture sensor and build my own hardware based on a D1.

my DSMR.yaml
substitutions:
  device_name: dsmr
  friendly_name: DSMR
  device_description: "DIY P1 module to read Belgian Fluvius smart meter"

esphome:
  name: ${device_name}
  comment: "${device_description}"
  platform: ESP8266
  board: d1_mini
  esp8266_restore_from_flash: true
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "2.0"

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name}-Fallback-Hotspot"
    password: !secret wifi_ap_password
  fast_connect: true
#  manual_ip:
#    static_ip: 192.168.0.243
#    gateway: 192.168.0.1
#    subnet: 255.255.255.0

captive_portal:

# Enable logging
logger:
  level: DEBUG
  esp8266_store_log_strings_in_flash: False
  baud_rate: 0 #disable serial logger using uart

# Enable Home Assistant API
api:
  password: !secret api_password
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

web_server:
  port: 80
  ota: true
  include_internal: true
  auth:
    username: !secret web_server_user
    password: !secret web_server_password

uart:
#  - rx_pin: D7
  - rx_pin: RX
    baud_rate: 115200
    rx_buffer_size: 1700

dsmr:
  max_telegram_length: 1700
  request_interval: 30s

# Dallas temperature sensor
dallas:
  pin: D5
  update_interval: 60s

# Individual sensors
sensor:
  - platform: dallas
    index: 0
    name: "${friendly_name} temperature"
  - platform: dsmr
    energy_delivered_tariff1:
      name: "Energy Consumed Tariff 1"
    energy_delivered_tariff2:
      name: "Energy Consumed Tariff 2"
    energy_returned_tariff1:
      name: "Energy Produced Tariff 1"
    energy_returned_tariff2:
      name: "Energy Produced Tariff 2"
    power_delivered:
      name: "Power Consumed"
      accuracy_decimals: 3
#      filters:
#        - throttle_average: 20s
    power_returned:
      name: "Power Produced"
      accuracy_decimals: 3
    electricity_failures:
      name: "Electricity Failures"
      icon: mdi:alert
    electricity_long_failures:
      name: "Long Electricity Failures"
      icon: mdi:alert
    voltage_l1:
      name: "Voltage Phase 1"
    voltage_l2:
      name: "Voltage Phase 2"
    voltage_l3:
      name: "Voltage Phase 3"
    current_l1:
      name: "Current Phase 1"
    current_l2:
      name: "Current Phase 2"
    current_l3:
      name: "Current Phase 3"
    power_delivered_l1:
      name: "Power Consumed Phase 1"
      accuracy_decimals: 3
    power_delivered_l2:
      name: "Power Consumed Phase 2"
      accuracy_decimals: 3
    power_delivered_l3:
      name: "Power Consumed Phase 3"
      accuracy_decimals: 3
    power_returned_l1:
      name: "Power Produced Phase 1"
      accuracy_decimals: 3
    power_returned_l2:
      name: "Power Produced Phase 2"
      accuracy_decimals: 3
    power_returned_l3:
      name: "Power Produced Phase 3"
      accuracy_decimals: 3
    gas_delivered_be:
      name: "Gas Consumed"
  - platform: uptime
    name: "${friendly_name} uptime"
  - platform: wifi_signal
    name: "${friendly_name} WiFi Strength"
    update_interval: 60s

text_sensor:
  - platform: dsmr
    identification:
      name: "${friendly_name} identification"
    p1_version:
      name: "${friendly_name} version"
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} IP Address"
    ssid:
      name: "${friendly_name} Wi-Fi SSID"
    bssid:
      name:  "${friendly_name} Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: false

EricVdB, 2022-10-27:
> please try this version. the ( char shall indicate that the 1. bracket shall be skipped and the 2. open bracket is searched
>
> 0-1:24.2.3(210117125004W)(01524.450*m3) e.g. [...]

I have this gas string too:
0-1:24.2.3(221026113502S)(02836.295*m3)

but 1,0-1:24.2.3(@(1,Gas,m3,gas_val,3 is resulting in: 
P1 m3 | null gas_val

Does someone have any idea? 
I'm using Tasmota version 12.2.0 compiled with VS Code

Thanks in advance,
Eric

haaxema, 2022-12-19:
@EricVdB 

Did you get this operational? I have the same problem.

Martin

> I have this gas string too: 0-1:24.2.3(221026113502S)(02836.295*m3)
>
> but 1,0-1:24.2.3(@(1,Gas,m3,gas_val,3 is resulting in: P1 m3 | null gas_val
>
> Does someone have any idea? I'm using Tasmota [...]

haaxema, 2022-12-19:
That did the trick! Thanx for the assistance. 

this has been modified to skipping N brackets because there are cases with many bracket values
> 
> so try:
> 
> 
> 
> 1,0-1:24.2.3(@(1,1,Gas,m3,gas_val,3

PR #10659 (2021-01-22): Sml obis fix
## Description:

option to fetch second (val) of obis payload

**Related issue (if applicable):** fixes #10644

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.5-rc6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contri [... cut]