arendst/Tasmota #10646: Requesting senor data generates command error

ManniAT, 2021-01-20:
### PROBLEM DESCRIPTION
I have a Sonoff TH module which in general works. On MQTT I get sensor values like this:

tele/soth16/SENSOR {"Time":"2021-01-20T21:40:33","SI7021":{"Temperature":23.1,"Humidity":41.7,"DewPoint":9.4},"TempUnit":"C"}
When I publish cmnd/soth16/SENSOR I expect to get new sensor data.
Instead (in console) I get 
21:42:36 MQT: stat/soth16/RESULT = {"Command":"Error"}

The same happens when I enter SENSOR in the console.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____Sonoff TH
- [x ] Tasmota binary firmware version number used: _____9.2.ß
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: _____OTA
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

21:45:59 CMD: Backlog Template; Module; GPIO 255
21:45:59 MQT: stat/soth16/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
21:45:59 MQT: stat/soth16/RESULT = {"Module":{"4":"Sonoff TH"}}
21:45:59 MQT: stat/soth16/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"1248":"SI7021"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

21:49:00 SRC: WebConsole from 10.1.1.152
21:49:00 CMD: Group 0, Index 1, Command "STATUS", Data "0"
21:49:00 MQT: stat/soth16/STATUS = {"Status":{"Module":4,"DeviceName":"SoTH16","FriendlyName":["SoTH16"],"Topic":"soth16","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0}}
21:49:00 MQT: stat/soth16/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:36","StartupUTC":"2021-01-20T20:38:24","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2021-01-20T20:16:14","SaveCount":65,"SaveAddress":"F9000"}}
21:49:00 MQT: stat/soth16/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"403/699"}}
21:49:00 MQT: stat/soth16/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ManniAT2_24","DEVWLManni"],"TelePeriod":60,"Resolution":"55C180C0","SetOption":["80008209","2805C8000100060000005AFF000000000000","00000000","00006000","00000000"]}}
21:49:00 MQT: stat/soth16/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
21:49:00 MQT: stat/soth16/STATUS5 = {"StatusNET":{"Hostname":"So1TH16","IPAddress":"10.1.1.172","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.13","Mac":"CC:50:E3:C8:17:01","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:49:00 MQT: stat/soth16/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.80","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C81701","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:49:00 MQT: stat/soth16/STATUS7 = {"StatusTIM":{"UTC":"2021-01-20T20:49:00","Local":"2021-01-20T21:49:00","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:34","Sunset":"17:28"}}
21:49:00 MQT: stat/soth16/STATUS10 = {"StatusSNS":{"Time":"2021-01-20T21:49:00","SI7021":{"Temperature":23.0,"Humidity":41.7,"DewPoint":9.3},"TempUnit":"C"}}
21:49:00 MQT: stat/soth16/STATUS11 = {"StatusSTS":{"Time":"2021-01-20T21:49:00","Uptime":"0T00:10:36","UptimeSec":636,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ManniAT2_24","BSSId":"AC:22:0B:31:21:88","Channel":4,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

21:48:11 SRC: MQTT
21:48:11 CMD: Group 0, Index 1, Command "SENSOR", Data ""
21:48:11 MQT: stat/soth16/RESULT = {"Command":"Error"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Send cmnd/device_name/SENSOR

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Get a tele/device_name/SENSOR message

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-01-20:
Hi,

`SENSOR` is not a valid command in Tasmota.

All valid commands are explained in the docs at https://tasmota.github.io/docs/Commands/

If you want to receive the values of sensors more often, you should change the teleperiod time using command `teleperiod`.

If you need further assistance on Tasmota usage, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-20 by ascillato2

meingraham, 2021-01-20:
`SENSOR` is not a Tasmota [command](https://tasmota.github.io/docs/Commands/). It is the result output from telemetry messages which occurs every `TelePeriod`.

`STATUS 8` is the command one issues to query the sensor information (`StatusSNS`) on demand.