arendst/Tasmota #10647: BLE Scanning stops after some time

valki2, 2021-01-21:
### PROBLEM DESCRIPTION
Bluetooth Scanning stops at some point and renders eg. temperature readings to static, old values that keep beeing transmitted via MQTT. Tasmota itself remains responsive via web.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ESP32 Devkit V1, V3 and others
- [ ] Tasmota binary firmware version number used: 9.2.0.3(sensors)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: flash_download_tool_v3.8.5
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
07:55:22.446 CMD: Backlog Template; Module; GPIO 255
07:55:22.515 MQT: stat/BTGateway/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
07:55:22.732 MQT: stat/BTGateway/RESULT = {"Module":{"1":"ESP32-DevKit"}}
07:55:22.983 MQT: stat/BTGateway/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"1056":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"1120":"RFSend"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"}}
07:55:22.995 MQT: stat/BTGateway/RESULT = {"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"1152":"RFrecv"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
07:53:45.085 CMD: STATUS 0
07:53:45.099 MQT: stat/BTGateway/STATUS = {"Status":{"Module":1,"DeviceName":"BTGateway","FriendlyName":["BTGateway"],"Topic":"BTGateway","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:53:45.106 MQT: stat/BTGateway/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:18:42","StartupUTC":"2021-01-21T06:35:03","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2021-01-17T12:04:10","SaveCount":30}}
07:53:45.111 MQT: stat/BTGateway/STATUS2 = {"StatusFWR":{"Version":"9.2.0.3(sensors)","BuildDateTime":"2021-01-16T17:12:27","Core":"1_0_5_RC4","SDK":"v3.3.4-286-gb4c075169","CpuFrequency":80,"Hardware":"ESP32","CR":"417/699"}}
07:53:45.116 MQT: stat/BTGateway/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mafia_IoT",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000002"]}}
07:53:45.128 MQT: stat/BTGateway/STATUS4 = {"StatusMEM":{"ProgramSize":1462,"Free":1984,"Heap":101,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","00152005","B7FFBFCF","01DA9BC4","60360CC7","00084052","00200000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,14,17,20,21,24,29,34,50","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
07:53:45.137 MQT: stat/BTGateway/STATUS5 = {"StatusNET":{"Hostname":"BTGateway","IPAddress":"192.168.X.Y","Gateway":"192.168.X.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.101","Mac":"24:0A:C4:31:52:14","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:53:45.145 MQT: stat/BTGateway/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.111","MqttPort":1883,"MqttClientMask":"BTGateway","MqttClient":"BTGateway","MqttUser":"valki","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
07:53:45.151 MQT: stat/BTGateway/STATUS7 = {"StatusTIM":{"UTC":"2021-01-21T06:53:45","Local":"2021-01-21T07:53:45","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:33","Sunset":"17:29"}}
07:53:45.163 MQT: stat/BTGateway/STATUS10 = {"StatusSNS":{"Time":"2021-01-21T07:53:45","ATC-03b0e5":{"Temperature":19.9,"Humidity":59.0,"DewPoint":11.6,"Battery":76,"RSSI":-79},"ATC-940e79":{"Temperature":20.7,"Humidity":56.0,"DewPoint":11.6,"Battery":86,"RSSI":-61},"ATC-a3c72f":{"Temperature":18.2,"Humidity":71.0,"DewPoint":12.8,"Battery":88,"RSSI":-83},"MJ_HT_V1-398b36":{"Temperature":18.2,"Humidity":68.3,"DewPoint":12.3,"Battery":86,"RSSI":-92},"TempUnit":"C"}}
07:53:45.171 MQT: stat/BTGateway/STATUS11 = {"StatusSTS":{"Time":"2021-01-21T07:53:45","Uptime":"0T00:18:42","UptimeSec":1122,"Heap":97,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":42,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Mafia_IoT","BSSId":"4A:8F:5A:C5:17:AC","Channel":7,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash 9.2.0.3(sensors) to an ESP32, let it fetch sensor data from LYWSD03MMC (ATC custom firmware) or MJ_HT_V1 and let it transmit its data to mqtt for a day

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect this to work longer than 1 or 2 days :)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
```lua
This is the console output after running all night. Note that the temperature of ATC-940e79 is not 20.8 but 19.8... Tasmota sends old, freezed values via mqtt and is "fixed" instantly after the reboot of esp32.

 07:28:09.461 MQT: tele/BTGateway/STATE = {"Time":"2021-01-21T07:28:09","Uptime":"0T15:08:43","UptimeSec":54523,"Heap":103,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Mafia_IoT","BSSId":"4A:8F:5A:C5:17:AC","Channel":7,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}
07:28:09.474 MQT: tele/BTGateway/SENSOR = {"Time":"2021-01-21T07:28:09","ATC-03b0e5":{"Temperature":20.8,"Humidity":68.0,"DewPoint":14.7,"Battery":76,"RSSI":-74},"ATC-940e79":{"Temperature":21.6,"Humidity":63.0,"DewPoint":14.2,"Battery":86,"RSSI":-57},"MJ_HT_V1-398b36":{"Temperature":18.9,"Humidity":77.2,"DewPoint":14.8,"Battery":86,"RSSI":-91},"ATC-a3c72f":{"Temperature":17.5,"Humidity":66.0,"DewPoint":11.1,"Battery":88,"RSSI":-82},"TempUnit":"C"}
07:29:09.467 MQT: tele/BTGateway/STATE = {"Time":"2021-01-21T07:29:09","Uptime":"0T15:09:43","UptimeSec":54583,"Heap":103,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Mafia_IoT","BSSId":"4A:8F:5A:C5:17:AC","Channel":7,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:04"}}
07:29:09.479 MQT: tele/BTGateway/SENSOR = {"Time":"2021-01-21T07:29:09","ATC-03b0e5":{"Temperature":20.8,"Humidity":68.0,"DewPoint":14.7,"Battery":76,"RSSI":-74},"ATC-940e79":{"Temperature":21.6,"Humidity":63.0,"DewPoint":14.2,"Battery":86,"RSSI":-57},"MJ_HT_V1-398b36":{"Temperature":18.9,"Humidity":77.2,"DewPoint":14.8,"Battery":86,"RSSI":-91},"ATC-a3c72f":{"Temperature":17.5,"Humidity":66.0,"DewPoint":11.1,"Battery":88,"RSSI":-82},"TempUnit":"C"}
[... 21 lines cut ...]
07:35:08.149 HTP: Web server active on BTGateway with IP address 192.168.X.Y
07:35:09.909 MQT: Attempting connection...
07:35:09.948 MQT: Connected
07:35:09.955 MQT: tele/BTGateway/LWT = Online (retained)
[... cut ...]
07:35:13.982 MQT: tele/BTGateway/SENSOR = {"Time":"2021-01-21T07:35:13","ATC-03b0e5":{"Temperature":19.8,"Humidity":59.0,"DewPoint":11.5,"Battery":76,"RSSI":-77},"ATC-940e79":{"Temperature":20.6,"Humidity":56.0,"DewPoint":11.5,"Battery":86,"RSSI":-62},"ATC-a3c72f":{"Temperature":18.3,"Humidity":70.0,"DewPoint":12.7,"Battery":88,"RSSI":-88},"TempUnit":"C"}

```

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
What WORKS:
- Wifi
- MQTT Connection
- ESP32 itself
- Tasmota itself as the website still is working... no issues with reboots etc.
- BT Sensors (tested with bluetooth scanner)

**(Please, remember to close the issue when the problem has been addressed)**

valki2, 2021-01-22:
BTW:  There is no core dump occuring like in this issue: https://github.com/espressif/esp-idf/issues/4196

There is also an open issue at espressif but I am unsure about the environment how the precompiled binaries were made...
https://github.com/espressif/esp-idf/issues/4001

dpritt, 2021-01-25:
I can confirm similar results with my BLE gateway using the standard release of v9.2, same hardware and setup as valki2.
This is a new install which stopped receiving messages from four LYWSD03MMC / ATC devices that have been flashed with the custom software from Aaron Christophel. Restarting the ESP32 in the morning restores the system, but it fails overnight.

valki2, 2021-01-27:
Fun Fact: in parallel, I got another esp32 running with openmqttgateway flashed. 
It seems to suffer from the same problem (stops refreshing the values) but last significantely longer until failure.

Again: sensors are fine and a bluetoothscanner proofs that.

So maybe its really a bug in the esp32 arduino core...

Milbert-R, 2021-01-27:
I have a similar problem. I use a ESP32 with built-in BT for scanning my Beacons for presence. With self compiled 9.0.0.3 all works fine for some days. After 2 or 3 days BLE-scanning stopps and time in TELE-output counts up. For workaround I restart the device every night. After OTA-Update to self compiled 9.2.0.3 BLE-scanning stopps after some hours. So I went back to 9.0.0.3 and my workaround.

homer66, 2021-01-31:
This is the same problem for me. It works for 24 hours, after which it only transmits the old value. (ESP32 Tasmota 9.2, MJ_HT_V1)

cmadsen, 2021-02-01:
Same problem here, works for ~12h then transmits stale values from my LYWSD03MMC's.

Eguar-rc, 2021-02-02:
The same problem, after a variable time, sends the same values, after restarting the ESP32 it works again for another period.

amargo, 2021-02-04:
Same problem for me, I have a workaround restart the ESP32 device by HA automations every 2 hours.

ryanjjennings, 2021-02-05:
I am also seeing this after about 2 days.

Eguar-rc, 2021-02-07:
It's been three days since everything works ok, although I don't know if it is related, it coincided with two changes:
I had two ESP32s, each for two sensors, I have removed one of them.
The Tasmota integration is no longer beta.
If I discover more clues I will share it.

valki2, 2021-02-07:
Funny... I removed one of my instances as well and since then it is much more stable. But that makes little sense....

Eguar-rc, 2021-02-07:
I am monitoring the other sensors using the custom-components / ble_monitor and they work fine, I use the bluethooth built into the raspberry pi.
I will wait a few more days to confirm the results.

jafrfc, 2021-02-13:
Same problem , circa 12h working & then transmits stale values from LYWSD03MMC's.

spayder0, 2021-02-16:
The same problem, after a variable time, sends the same values.

ascillato2, 2021-03-02:
Hi, please, can you test with latest version from the development branch?  Thanks.

dpritt, 2021-03-04:
Just tried using the latest tasmota32-sensors development version,  but it does not detect any of my BLE devices, also issuing the MI32xxxx commands gave errors. I did reissue the setoption115 1 just in case the setting had been lost.
Reverted back to 9.2.0.6(sensors) and it bust back into life.

dpritt, 2021-03-04:
Done some further tests, and used the production version 'Kenneth' 9.3.1, and this too is failing.

jafrfc, 2021-03-04:
Has someone tried tasmota32-bluetooth.bin 9.3.1? Does it works with LYWSD03MMC's ?
--

jafrfc, 2021-03-04:
Sorry for the typo

dpritt, 2021-03-05:
I cannot get any of the 9.3 releases to capture the messages from my LYWSD03MMC devices.

My production ESP32 is running 'Julie' and this runs OK showing all devices, but I have to monitor the 'last_changed' attribute from the devices it and restart if messages are not being updated after 15 minutes as BLE module appears to stop about once a day, as reported previously.

I use Home Assistant so have an automation to monitor the last_changed and send a MQTT command to restart the ESP32 if it stops updating.

Milbert-R, 2021-03-08:
Ok, I got some new ESP32-devices for testing the actual Tasmota-Firmware...

Until now I used 9.0.0.3 for scaning my BLE-Beacons via MI32Beacon1-3.
The problem was, that BLE-scanning stoped sometimes, so I used restarting the ESP32-device every day via ioBroker

Now I was able to test the actual version 9.3.1.1 with one of my new devices. I recognised that "MI32Beacon" will not work with this Firmware anymore. 

I recognised new information from: https://btsimonh.github.io/docs/ble/Bluetooth-Esp32/

So I will have to completly rewrite my presence-scaning via BLE and Tasmota with ioBroker.

I tell the results after it's done...

dpritt, 2021-03-24:
I flashed another ESP32 device with the development build 9.3.1.1(bluetooth) compiled on the 2021-03-11 at 08:05:44 later that day. 
This has been up and running correctly since then, all looks stable and it is still receiving and forwarding messages from my Bluetooth LYWSD03MMC / ATC modules.
The device was supplied via AZ-Delivery and is described as a ESP32 NodeMCU Module WLAN WiFi Development Board with CP2102.

Jason2866, 2021-04-02:
Is the issue solved?

dpritt, 2021-04-04:
I think it is, been stable for over a week now.

Regards

Dave

ascillato2, 2021-04-07:
Thanks for reporting.

closed 2021-04-07 by ascillato2

cctweaker, 2021-12-26:
I had the same problem with the latest version 10.1.0 Noelle, my NodeMCU32 would stop randomly after 12-36 hours.
Flashed 9.3.1 and it's been stable for the last 5 days.

barbudor, 2021-12-26:
@cctweaker 
Hijacking an old issue doesn't make any sense has many things have changes since the version it was reported on.
If you have an issue with BLE32, please upgrade to latest dev version 10.1.0.3 (from http://ota.tasmota.com/tasmota32).
If you still have a problem, open a new issue without forgetting to fill-up the required information with as much details as you can.