arendst/Tasmota #10651: Tasmota ZBBridge send no more events from Zigbee groups added IKEA remotes

heimgeist, 2021-01-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:06:58.461 CMD: Backlog Template; Module; GPIO 255
15:06:58.517 MQT: stat/ZBBridge/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:06:58.725 MQT: stat/ZBBridge/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
15:06:58.980 MQT: stat/ZBBridge/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:08:03.029 CMD: Status 0
15:08:03.034 MQT: stat/ZBBridge/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"ZBBridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:08:03.037 MQT: stat/ZBBridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-zbbridge.bin.gz	","RestartReason":"Software/System restart","Uptime":"1T06:42:57","StartupUTC":"2021-01-20T07:25:06","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2021-01-18T15:46:57","SaveCount":20,"SaveAddress":"FD000"}}
15:08:03.039 MQT: stat/ZBBridge/STATUS2 = {"StatusFWR":{"Version":"9.2.0.3(zbbridge)","BuildDateTime":"2021-01-19T15:24:56","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"355/699"}}
15:08:03.041 MQT: stat/ZBBridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AP2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
15:08:03.046 MQT: stat/ZBBridge/STATUS4 = {"StatusMEM":{"ProgramSize":572,"Free":1216,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
15:08:03.049 MQT: stat/ZBBridge/STATUS5 = {"StatusNET":{"Hostname":"ZBBridge","IPAddress":"192.168.32.40","Gateway":"192.168.32.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.33.3","Mac":"C8:2B:96:04:B8:CE","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:08:03.052 MQT: stat/ZBBridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.32.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_04B8CE","MqttUser":"DVES_USER","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:08:03.058 MQT: stat/ZBBridge/STATUS7 = {"StatusTIM":{"UTC":"2021-01-21T14:08:03","Local":"2021-01-21T15:08:03","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:33","Sunset":"17:29"}}
15:08:03.061 MQT: stat/ZBBridge/STATUS10 = {"StatusSNS":{"Time":"2021-01-21T15:08:03"}}
15:08:03.066 MQT: stat/ZBBridge/STATUS11 = {"StatusSTS":{"Time":"2021-01-21T15:08:03","Uptime":"1T06:42:57","UptimeSec":110577,"Vcc":3.471,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"Wifi":{"AP":1,"SSId":"AP2","BSSId":"FE:EC:DA:1A:EC:C8","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
# -> IKEA Remote 0x77BD is joined to Tasmoto and sent all button events....
11:04:38.838 MQT: tele/ZBBridge/SENSOR = {"ZbReceived":{"0x77BD":{"Device":"0x77BD","0005!07":"00010D00","ArrowClick":0,"Endpoint":1,"LinkQuality":157}}}
11:04:44.138 MQT: tele/ZBBridge/SENSOR = {"ZbReceived":{"0x77BD":{"Device":"0x77BD","0005!07":"01010D00","ArrowClick":1,"Endpoint":1,"LinkQuality":160}}}

# -> The Bindstate is empty....
11:08:45.669 CMD: ZbBindstate 0x77BD
11:08:45.675 MQT: stat/ZBBridge/RESULT = {"ZbBindState":"Done"}
11:08:47.640 MQT: tele/ZBBridge/RESULT = {"ZbBindState":{"Device":"0x77BD","Status":0,"StatusMessage":"SUCCESS","Total":0,"Start":1,"Bindings":[]}}

# -> Now I bind the IKEA remote to Zigbee group 100 where a light bulb already belongs to
11:09:51.820 CMD: ZbBind {"device":"0x77BD","ToGroup":100,"Endpoint":1,"Cluster":6}
11:09:51.828 MQT: stat/ZBBridge/RESULT = {"ZbBind":"Done"}
11:09:59.549 ZIG: {"ZbEZSPReceived":"800042BD77"}
11:10:02.835 MQT: tele/ZBBridge/RESULT = {"ZbBind":{"Device":"0x77BD","Status":0,"StatusMessage":"SUCCESS"}}

# -> The binding was successfull and the light bulb react to the button commands. 
# -> But after this point I didn't receive any longer button pressed events from the IKEA remote. 
# -> Only the light bulb send "Power":0/1 events after status changes.  

# -> The remote is still available. It answer to the ZbBindstate command. But the button pressed events are gone...
11:13:42.945 CMD: ZbBindstate 0x77BD
11:13:42.951 MQT: stat/ZBBridge/RESULT = {"ZbBindState":"Done"}
11:13:43.771 MQT: tele/ZBBridge/RESULT = {"ZbBindState":{"Device":"0x77BD","Status":0,"StatusMessage":"SUCCESS","Total":1,"Start":1,"Bindings":[{"Cluster":"0x0006","Endpoint":1,"ToGroup":100}]}}

### EXPECTED BEHAVIOUR
I expect button pressed events from the remote after a Zigbee group binding. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

alexhk, 2021-01-25:
I used `ZbListen1 100` to get the events back in the log. Not sure if there is a more elegant way.

Be aware that this doesn't survive reboots as far as I can tell. As workaround I added the following rule for my 4 groups. Another workaround is that zigbee starts up last (long after system#boot or power1#boot) so I added a 60 second delay. 

```
Rule1 
 ON System#Boot DO Ruletimer8 60 ENDON
 ON Rules#Timer=8 DO Backlog ZbListen1 101; ZbListen2 102; ZbListen3 103; ZbListen4 104 ENDON
  
rule1 1

```

After `restart 1`  you should see the rules being executed and your events should be back.

s-hadinger, 2021-01-25:
+1. EZSP needs you to explicitly declare which groups you are listening to.

closed 2021-02-24 by github-actions[bot]

moodyblue, 2021-11-04:
> EZSP needs you to explicitly declare which groups you are listening to

Where is this documented ?

s-hadinger, 2021-11-04:
What do you mean? It's the behavior of EZSP. You want a link to EZSP documentation?

moodyblue, 2021-11-04:
The only documentation that I've found about Z2T is this and it does not mention that EZSP bahaviour
https://taillieu.info/index.php/articles/category-blog/46-my-projects/tasmota
Is there any official Z2T documentation ? If not, maybe one can be developed using this link as a base.

moodyblue, 2021-11-04:
> What do you mean?

The only documentation that I've found about Z2T is this and it does not mention that EZSP behaviour

https://taillieu.info/index.php/articles/category-blog/46-my-projects/tasmota

Is there any oficial Z2T documentation ? If not, maybe one can be developed using the above link as a base.

s-hadinger, 2021-11-04:
It is briefly mentioned here https://tasmota.github.io/docs/Commands/#cmnd_zblisten

You are welcome to propose improvements to the documentation

moodyblue, 2021-11-05:
> You are welcome to propose improvements to the documentation

ZbListen docs start with the following remark `(EZSP ZBBridge only)Listens to a multicast group address. By default EZSP will not report group messages unless you subscribe to the group.`

I propose that in ZbBind you also include a remark, such as `(EZSP ZBBridge only) If you bind devices to groups you should also use ZbListen to that group, otherwise MQTT messages will not be published`

s-hadinger, 2021-11-05:
Thanks, added to doc.