arendst/Tasmota #10660: Home Assitant autodiscovery does not work for binary sensors

Gifford47, 2021-01-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The docs [(HA Supported Entities)](https://tasmota.github.io/docs/Home-Assistant/#supported-entities) say that the auto discovery function also works for binary sensors, depending on _switchmode_ and _setoption114_.
However, I have connected a PIR to gpio5 on a Wemos D1 Mini and I do not get a homeassistant topic for automatic discovery.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: __tasmotizer___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
 

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00 CFG: Loaded from flash at F4, Count 40
00:00:00 QPC: Count 1
00:00:00 Project tasmota tasmota1_dht22 Version
9.2.0(tasmota)-2_7_4_9(2020-12-21T15:03:40)
00:00:00 WIF: Connecting to AP1 nodes_land Channel 11 BSSId F2:9F:C2:3A:CD:0A in mode 11n as tasmota1_dht22-4720...
00:00:01 WIF: Connected
00:00:02 HTP: Web server active on tasmota1_dht22-4720 with IP address
192.168.1.83
19:48:13 MQT: Attempting connection...
19:48:13 MQT: Connected
19:48:13 MQT: tele/tasmota1_dht22/LWT = Online (retained)
19:48:13 MQT: cmnd/tasmota1_dht22/POWER =
19:48:13 MQT: tele/tasmota1_dht22/INFO1 = {"Module":"Generic","Version":"9.2.0(tasmota)","FallbackTopic":"cmnd/DVES_EFF270_fb/","GroupTopic":"cmnd/tasmotas/"}
19:48:13 MQT: tele/tasmota1_dht22/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota1_dht22-4720","IPAddress":"192.168.1.83"}
19:48:13 MQT: tele/tasmota1_dht22/INFO3 = {"RestartReason":"Software/System restart"}
19:48:16 QPC: Reset
19:48:17 MQT: tele/tasmota1_dht22/STATE = {"Time":"2021-01-21T19:48:17","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"nodes_land","BSSId":"F2:9F:C2:3A:CD:0A","Channel":11,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}

19:48:17 MQT: tele/tasmota1_dht22/SENSOR = {"Time":"2021-01-21T19:48:17","Switch1":"OFF","AM2301":{"Temperature":20.5,"Humidity":55.0,"DewPoint":11.1},"TempUnit":"C"}

19:48:22 MQT: homeassistant/sensor/EFF270_AM2301_Temperature/config = {"name":"tasmota1_dht22 AM2301 Temperature","stat_t":"tele/tasmota1_dht22/SENSOR","avty_t":"tele/tasmota1_dht22/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"EFF270_AM2301_Temperature","dev":{"ids":["EFF270"]},"unit_of_meas":"°C","dev_cla":"temperature","frc_upd":true,"val_tpl":"{{value_json['AM2301']['Temperature']}}"} (retained)
19:48:22 MQT: homeassistant/sensor/EFF270_AM2301_Humidity/config = {"name":"tasmota1_dht22 AM2301 Humidity","stat_t":"tele/tasmota1_dht22/SENSOR","avty_t":"tele/tasmota1_dht22/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"EFF270_AM2301_Humidity","dev":{"ids":["EFF270"]},"unit_of_meas":"%","dev_cla":"humidity","frc_upd":true,"val_tpl":"{{value_json['AM2301']['Humidity']}}"} (retained)
19:48:22 MQT: homeassistant/sensor/EFF270_AM2301_DewPoint/config = {"name":"tasmota1_dht22 AM2301
DewPoint","stat_t":"tele/tasmota1_dht22/SENSOR","avty_t":"tele/tasmota1_dht22/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"EFF270_AM2301_DewPoint","dev":{"ids":["EFF270"]},"unit_of_meas":"°C","ic":"mdi:weather-rainy","frc_upd":true,"val_tpl":"{{value_json['AM2301']['DewPoint']}}"} (retained)
19:48:22 MQT: homeassistant/sensor/EFF270_status/config = {"name":"tasmota1_dht22 status","stat_t":"tele/tasmota1_dht22/HASS_STATE","avty_t":"tele/tasmota1_dht22/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"tele/tasmota1_dht22/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"EFF270_status","dev":{"ids":["EFF270"],"name":"tasmota1_dht22","mdl":"Generic","sw":"9.2.0(tasmota)","mf":"Tasmota"}} (retained)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
You can configure a switch (160) and use it as binary sensor, using _switchmode1 1_. Detach switches from relays and send MQTT messages instead _SetOption114 1_. Restart and look into console do find _homeassistant/*_ auto discovery topic.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
A auto discovery topic should be send to mqtt broker, like:
_homeassistant/binary_sensor/switch1/config = ..._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
a similar issue was created here: #10578

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-01-22:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2021-01-22:
The template is incomplete so I will be guessing. Are you using `setoption19 1`? If so, that is **MQTT DISCOVERY**. For **Home Assistant NATIVE Discovery** (and the ability to use setoption114) you have to use `setoption19 0`. Your entities will be automatically populated under INTEGRATIONS in Home Assistant.

Gifford47, 2021-01-22:
sorry, i updated the template. yes, i am using setoption19 1. I need the option for openhab3, that the things will be created automatically. but like i described the switch topic is missed ...

ascillato, 2021-01-22:
Ok so you are using OpenHab and not Home Assistant!

setoption19 1 is MQTT DISCOVERY that can be used for OpenHab and others
setoption19 0 is Home Assistant Discovery.

As the title said Home Assistant, I assume you were using Home Assistant. Ok, let me do some tests.

effelle, 2021-01-23:
https://tasmota.github.io/docs/Home-Assistant/#supported-entities_1

```
To have switches discovered SwitchTopic must be set to a custom name and it will automatically start 
to listen and publish using stat/%topic%/SWITCH<x> (binary sensor) or stat/%topic%/SWITCH<x>T (trigger) topics.
```

marc-gist, 2021-01-23:
> https://tasmota.github.io/docs/Home-Assistant/#supported-entities_1
> 
> ```
> To have switches discovered SwitchTopic must be set to a custom name and it will automatically start 
> to listen and publish using stat/%topic%/SWITCH<x> (binary sensor) or stat/%topic%/SWITCH<x>T (trigger) topics.
> ```

This isn't working for me. I set a `SwitchTopic door`, and did so19 (switch is `SwitchMode1 15`), and still no binary sensor set in HA.

effelle, 2021-01-23:
Switchmode 15 is not supported by the discovery since is an switchmode made just for MQTT messages and the discovery use its customized topic.
As a test use another switchmode and please report.

marc-gist, 2021-01-23:
> Switchmode 15 is not supported by the discovery since is an switchmode made just for MQTT messages and the discovery use its customized topic.
> As a test use another switchmode and please report.

That worked, set to `switchmode1 1`, however, setup a device named `<tasmota_name>_switch1`
Rather than using the SwitchTopic for the name.  
Is there a way to name the switch and device type prior to enabling so19?

Gifford47, 2021-01-24:
> https://tasmota.github.io/docs/Home-Assistant/#supported-entities_1
> 
> ```
> To have switches discovered SwitchTopic must be set to a custom name and it will automatically start 
> to listen and publish using stat/%topic%/SWITCH<x> (binary sensor) or stat/%topic%/SWITCH<x>T (trigger) topics.
> ```

thanks for the answer!

I have now set "switchtopic pir" and a switch channel is now automatically created in openhab.
however, the D1 mini does not send a mqtt message as soon as the pir sensor value changes. no MQTT log appears in the tasmota log.
I can't create a rule either, because I created the switch topic...

Gifford47, 2021-01-25:
ok, now it works. to use mqtt autodiscovery with switches in openhab you have to set the following:
"switchmode1 1" -> binary sensor (ON/OFF)
"switchtopic <mytopic>" -> must be set to a custom value, to use mqtt autodiscovery for switches (acc. to. [supported entities)](https://tasmota.github.io/docs/Home-Assistant/#supported-entities_1)
"setoption19 1" -> enable homeassistant auto discovery

I still have one question, though:
by using switchtopic, i can no longer use a rule on the Switch#State trigger. however, i would like to "switch on" or "switch off" the switch (in my case the pir sensor) when it is not needed. how can i now use a rule to prevent the switch from sending its status every time?

effelle, 2021-01-25:
There is a possible workaround but is a bit of a hit and miss:
```py
on System#Boot do backlog switchtopic pir; delay 200;  switchtopic 2 endon
```
And then use the switch JUST with a rule using `ON SWITCH<x>#STATE=n`. 
Pay attention to the delay value, since the discovery will be updated +/- 10 seconds after a full reboot. Also consider using a rule to manage the switches will introduce a little delay in action.

closed 2021-01-25 by ascillato2