arendst/Tasmota #10661: ShutterButton "Toggle" not toggling up/down direction

rbswift, 2021-01-22:
### PROBLEM DESCRIPTION
When setting a `ShutterButton` as single button `toggle` control the expected behaviour on repeated presses is up, stop, down, stop, up, stop, down, stop etc. The actual control behaviour is, for example: up, stop, up, stop, up, stop etc.

This means that, using a single "toggle" control button, it is not possible to go from 0% (fully closed) to 10% open then back to 0% without going via 100% (fully open).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R2
- [x] Tasmota binary firmware version number used: 9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP easy
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
09:18:06 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
09:18:06 SRC: Backlog
09:18:06 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
09:18:06 MQT: stat/bedroomshutter/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:18:07 SRC: Backlog
09:18:07 CMD: Group 0, Index 1, Command "MODULE", Data ""
09:18:07 MQT: stat/bedroomshutter/RESULT = {"Module":{"39":"Sonoff Dual R2"}}
09:18:07 SRC: Backlog
09:18:07 CMD: Group 0, Index 1, Command "GPIO", Data "255"
09:18:07 MQT: stat/bedroomshutter/RESULT = {"GPIO0":{"160":"Switch1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"33":"Button2"},"GPIO4":{"0":"None"},"GPIO5":{"225":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"32":"Button1"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
n/a
```
- [x] Provide the output of this command: `Status 0`:
```lua
09:20:44 MQT: stat/bedroomshutter/STATUS = {"Status":{"Module":39,"DeviceName":"Bedroom Shutter","FriendlyName":["Bedroom Shutter","Bedroom Shutter 2"],"Topic":"bedroomshutter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
09:20:44 MQT: stat/bedroomshutter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"7T08:21:04","StartupUTC":"2021-01-14T23:59:40","Sleep":50,"CfgHolder":4617,"BootCount":66,"BCResetTime":"2020-04-08T10:42:08","SaveCount":4023,"SaveAddress":"F6000"}}
09:20:44 MQT: stat/bedroomshutter/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-17T10:05:31","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"463/699"}}
09:20:44 MQT: stat/bedroomshutter/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Teh Interwebz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C029","2805C8000100060000005A64000000000000","C00000C0","00006000","00000000"]}}
09:20:44 MQT: stat/bedroomshutter/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
09:20:44 MQT: stat/bedroomshutter/STATUS5 = {"StatusNET":{"Hostname":"bedroomshutter","IPAddress":"192.168.188.63","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.188.12","Mac":"C4:4F:33:9E:8A:73","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:20:44 MQT: stat/bedroomshutter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.188.12","MqttPort":1883,"MqttClientMask":"bedroomshutter","MqttClient":"bedroomshutter","MqttUser":"bedroomshutter","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:20:44 MQT: stat/bedroomshutter/STATUS7 = {"StatusTIM":{"UTC":"2021-01-22T08:20:44","Local":"2021-01-22T09:20:44","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:32","Sunset":"17:31"}}
09:20:44 MQT: stat/bedroomshutter/STATUS10 = {"StatusSNS":{"Time":"2021-01-22T09:20:44","Switch1":"ON","Shutter1":{"Position":0,"Direction":0,"Target":0}}}
09:20:44 MQT: stat/bedroomshutter/STATUS11 = {"StatusSTS":{"Time":"2021-01-22T09:20:44","Uptime":"7T08:21:04","UptimeSec":634864,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Teh Interwebz","BSSId":"08:96:D7:8A:94:84","Channel":6,"RSSI":76,"Signal":-62,"LinkCount":3,"Downtime":"0T00:02:07"}}}
09:20:44 MQT: stat/bedroomshutter/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":240,"Close":190,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
09:37:57 SHT: Settings.shutter_button[i] 0, shutter_index 0, Button.press_counter[i] 0, min_shutterbutton_press_counter -1, i 0
09:37:57 SHT: Settings.shutter_button[i] -2147483440, shutter_index 0, Button.press_counter[i] 1, min_shutterbutton_press_counter -1, i 1
09:37:57 SHT: Settings.shutter_button[i] 0, shutter_index 0, Button.press_counter[i] 0, min_shutterbutton_press_counter -1, i 2
09:37:57 SHT: Settings.shutter_button[i] 0, shutter_index 0, Button.press_counter[i] 0, min_shutterbutton_press_counter -1, i 3
09:37:57 SHT: shutter 1, button 2 = 1 (single=1, double=2, tripple=3, hold=4)
09:37:57 SHT: Payload toggle: 1, i 1, dir 0
09:37:57 SHT: Pos. in: payload  (0), payload 100, idx 1, src 7
09:37:57 SHT: lastsource 7:, real 0, target 480000, payload 100
09:37:57 SHT: Delay Start. var0 <99>=<>, max10s?
09:37:57 SHT: Delay Start. Done
09:37:57 SRC: Shutter
09:37:57 SHT: Switched relay: 1 by Shutter
09:37:57 SHT: Shutter 1: source: Shutter, powerstate_local 1, ShutterGlobal.RelayCurrentMask 1, manual change 0
09:37:57 MQT: stat/bedroomshutter/RESULT = {"POWER1":"ON"}
09:37:57 MQT: stat/bedroomshutter/POWER1 = ON (retained)
09:37:57 MQT: stat/bedroomshutter/RESULT = {"Shutter1":{"Button2":1}}
09:37:57 SHT: Shutter 1 Real 0, Start 0, Stop 480000, Dir 1, Delay 0, Rtc 0.05 [s], Freq 0, PWM 0
[... 134 lines cut ...]
09:38:14 SHT: Stop Shutter 0. Switchmode 0
09:38:14 SRC: Shutter
09:38:14 SHT: Switched relay: 1 by Shutter
09:38:14 SHT: Shutter 1: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
09:38:14 MQT: stat/bedroomshutter/RESULT = {"POWER1":"OFF"}
09:38:14 MQT: stat/bedroomshutter/POWER1 = OFF (retained)
```

### TO REPRODUCE

1. set `shutterbutton1 2 t` to configure the toggling shutter button
2. press the button every few seconds
3. observe the movement pattern is up, stop, up, stop, up, stop.
4. there is no way to 'toggle' the direction to down until the shutter is fully opened.

### EXPECTED BEHAVIOUR
successive presses of the toggle button should toggle the direction i.e. up, stop, down, stop, up, stop, down, stop

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
n/a

**(Please, remember to close the issue when the problem has been addressed)**

Sofian34, 2021-01-25:
wqjrejszgx

stefanbode, 2021-01-27:
Thanks for bringing this up. I have not invented the buttons functionality, but will take a look at it today.

rbswift, 2021-02-08:
@stefanbode thanks, let me know if I can do anything to help. cheers

stefanbode, 2021-02-18:
@rbswift : would be breat if you can check the newest development branch and provide feedback. It work on my side

rbswift, 2021-02-22:
Thanks @stefanbode I tried the latest development branch 9.3.0.1 and the behaviour is still down, stop, down, stop unfortunately. Here's my log...

```
00:00:04.539 WIF: Checking connection...
00:00:04.541 WIF: Attempting connection...
00:00:05.753 WIF: Checking connection...
00:00:05.756 WIF: Connected
00:00:06.005 HTP: Web server active on bedroomshutter with IP address 192.168.188.63
00:00:06.498 QPC: Reset
00:00:06.767 RTC: UTC 2021-02-22T03:19:23, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
04:19:24.272 MQT: Attempting connection...
04:19:24.286 MQT: Connected
04:19:24.291 MQT: tele/bedroomshutter/LWT = Online (retained)
04:19:24.296 MQT: cmnd/bedroomshutter/POWER = 
04:19:24.300 MQT: Subscribe to cmnd/bedroomshutter/#
04:19:24.305 MQT: Subscribe to cmnd/tasmotas/#
04:19:24.310 MQT: Subscribe to cmnd/bedroomshutter_fb/#
04:19:24.314 MQT: Unsubscribe from homeassistant/status
04:19:24.323 MQT: tele/bedroomshutter/INFO1 = {"Module":"Sonoff Dual R2","Version":"9.3.0.1(tasmota)","FallbackTopic":"cmnd/bedroomshutter_fb/","GroupTopic":"cmnd/tasmotas/"}
04:19:24.334 MQT: tele/bedroomshutter/INFO2 = {"WebServerMode":"Admin","Hostname":"bedroomshutter","IPAddress":"192.168.188.63"}
04:19:24.345 MQT: tele/bedroomshutter/INFO3 = {"RestartReason":"Software/System restart"}
04:19:24.351 MQT: stat/bedroomshutter/RESULT = {"POWER1":"OFF"}
04:19:24.356 MQT: stat/bedroomshutter/POWER1 = OFF (retained)
04:19:24.361 MQT: stat/bedroomshutter/RESULT = {"POWER2":"OFF"}
04:19:24.367 MQT: stat/bedroomshutter/POWER2 = OFF (retained)
04:19:25.423 APP: Boot Count 70
04:19:25.888 CFG: Saved to flash at F9, Count 4306, Bytes 4096
04:19:28.403 MQT: tele/bedroomshutter/STATE = {"Time":"2021-02-22T04:19:28","Uptime":"0T00:00:13","UptimeSec":13,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Teh Interwebz","BSSId":"08:96:D7:8A:94:84","Channel":1,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:07"}}
04:19:28.434 MQT: tele/bedroomshutter/SENSOR = {"Time":"2021-02-22T04:19:28","Switch1":"ON","Shutter1":{"Position":100,"Direction":0,"Target":100}}
04:19:33.417 MQT: tasmota/discovery/C44F339E8A73/config = {"ip":"192.168.188.63","dn":"Bedroom Shutter","fn":["Bedroom Shutter","Bedroom Shutter 2",null,null,null,null,null,null],"hn":"bedroomshutter","mac":"C44F339E8A73","md":"Sonoff Dual R2","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.3.0.1","t":"bedroomshutter","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[3,3,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
04:19:33.426 MQT: tasmota/discovery/C44F339E8A73/sensors = {"sn":{"Time":"2021-02-22T04:19:33","Switch1":"ON","Shutter1":{"Position":100,"Direction":0,"Target":100}},"ver":1} (retained)
04:19:43.044 WIF: Checking connection...
04:20:03.020 WIF: Checking connection...
04:20:23.005 WIF: Checking connection...
04:20:43.018 WIF: Checking connection...
04:20:55.177 HTP: Main Menu
04:21:00.514 HTP: Console
04:21:03.002 WIF: Checking connection...
04:21:11.547 SHT: Time 16, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 461055, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:11.597 SHT: Time 17, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 459792, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:11.698 SHT: Time 19, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 457266, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:11.797 SHT: Time 21, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 454740, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:11.847 SHT: Time 22, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 453477, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:11.897 SHT: Time 23, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 452214, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:11.948 SHT: Time 24, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 450951, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:12.047 SHT: Time 26, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 448425, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:12.097 SHT: Time 27, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 447162, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:12.147 SHT: Time 28, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 445899, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:12.247 SHT: Time 30, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 443373, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:12.297 SHT: Time 31, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 442110, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:12.347 SHT: Time 32, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 440847, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:12.398 SHT: Time 33, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 439584, nStop 0, Trgt 0, mVelo 0, Dir -1
04:21:12.402 SHT: Shtr1 Real 439584, Start 480000, Stop 0, Dir -1, Delay 0, Rtc 1.65 [s], Freq 0, PWM 0
04:21:12.408 MQT: stat/bedroomshutter/RESULT = {"Shutter1":{"Position":92,"Direction":-1,"Target":0}}
04:21:12.449 SHT: Time 34, toBA 0, cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 438321, nStop 0, Trgt 0, mVelo 0, Dir -1
[... cut ...]
04:21:16.909 MQT: stat/bedroomshutter/RESULT = {"POWER2":"OFF"}
04:21:16.914 MQT: stat/bedroomshutter/POWER2 = OFF (retained)
04:21:17.418 SHT: Shtr1 Real 389064, Start 389064, Stop 390327, Dir 0, Delay 0, Rtc 1.60 [s], Freq 0, PWM 0
04:21:17.424 MQT: stat/bedroomshutter/SHUTTER1 = 81 (retained)
04:21:17.429 MQT: stat/bedroomshutter/RESULT = {"Shutter1":{"Position":81,"Direction":0,"Target":81}}
04:21:17.436 MQT: stat/bedroomshutter/RESULT = {"Shutter1":{"Button2":1}}
04:21:17.057 CFG: Saved to flash at F5, Count 4310, Bytes 4096
```

stefanbode, 2021-02-24:
I double-checked the code and the gentlemen that implemented the button toogle functionality had the following behavior in mind:
if you move down 0-->100 and stop at 22. then the next toogle move on to 100. No direction change.
If you stop after 50%. e.g. 62%. Then the next toogle change the direction and the shutter moves to 0. 
I checked and see that it behaves exactly like this. Now the question: Why? and can we change it. I would say no. If you want to change it on your side I can guide you wich line to change.

stefanbode, 2021-02-24:
Line ~850
OLD:
```
              AddLog_P(LOG_LEVEL_DEBUG_MORE, PSTR("SHT: Shtr%d -> %d"), shutter_index+1, position);
              if (102 == position) {
                XdrvMailbox.payload = XdrvMailbox.index;
                CmndShutterToggle();
              } else {
                CmndShutterPosition();
              }
```
NEW
```
              AddLog_P(LOG_LEVEL_DEBUG_MORE, PSTR("SHT: Shtr%d -> %d"), shutter_index+1, position);
              if (102 == position) {
                XdrvMailbox.payload = XdrvMailbox.index;
                **CmndShutterToggleDir();**
              } else {
                CmndShutterPosition();
              }
```

rbswift, 2021-02-24:
Thanks so much for checking @stefanbode . I don't really see how that behaviour can be correctly described as toggle if the direction does not change. Would it be possible to give that behaviour a more accurate description and implement an actual toggle?

My use case is that the blind is rarely fully closed just toggle between 100% open and about 80% open every day. Currently I have to fully close the blind before I can open it.

stefanbode, 2021-02-24:
Let me check if there is a simple solution with normal button and a rule. We already have a toggle command doing what you want, but this is not connected to the Shutterbutton1 function.

stefanbode, 2021-02-25:
Here we go:
I suggest to disconnect the button from the relays first: `setoption71 1`
then defining a single press event on button 2
```
rule1 ON button2#state=10 DO shutterstoptoggledir1 ENDON
rule1 on
```
For me this works as you expected. To make the execution faster you can limit the buttonfunctionality to SINGLE press only. 
`SetOption13 ON - Allow immediate action on single button press`

rbswift, 2021-02-25:
great - thanks, I will try it tomorrow

rbswift, 2021-02-26:
Great - it works!

Just in case someone finds this later, I suspect `setoption71 1` above was that a typo for `setoption73 1` (?) I also had to set `shutterbutton1 0` first which took me a little time to figure out. After that it's working as expected.

Thanks so much for your help @stefanbode !!

closed 2021-02-26 by rbswift