arendst/Tasmota #10669: Device Groups not working in different BSSIDs (but same Network / SSID)

mrbluebrett, 2021-01-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Device Groups not working for devices in different BSSIDs (but same Network / SSID), but when changing all the devices to the same BSSID it's working fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Moes WS-EUY1 1 Gang Switch 
- [X] Tasmota binary firmware version number used: 9.2.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: tuya-convert
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:20:22 MQT: light/pasillo_slave/RESULT = {"NAME":"WS-EUY1-W","GPIO":[544,0,0,32,224,0,0,0,0,0,288,0,0,0],"FLAG":0,"BASE":18}
22:20:23 MQT: light/pasillo_slave/RESULT = {"Module":{"0":"WS-EUY1-W"}}
22:20:23 MQT: light/pasillo_slave/RESULT = {"GPIO0":{"544":"LedLink"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"224":"Relay1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"288":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
  STATUS 0 output here:
22:21:13 MQT: light/pasillo_slave/STATUS = {"Status":{"Module":0,"DeviceName":"Pasillo Slave","FriendlyName":["Pasillo Slave"],"Topic":"pasillo_slave","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:21:13 MQT: light/pasillo_slave/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:20:47","StartupUTC":"2021-01-20T21:00:26","Sleep":50,"CfgHolder":4617,"BootCount":34,"BCResetTime":"2021-01-14T18:39:38","SaveCount":308,"SaveAddress":"F8000"}}
22:21:13 MQT: light/pasillo_slave/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"469/699"}}
22:21:13 MQT: light/pasillo_slave/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xERICx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","2805C8000100060000005A00000000000000","00000000","00006008","00000000"]}}
22:21:13 MQT: light/pasillo_slave/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":0,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
22:21:13 MQT: light/pasillo_slave/STATUS5 = {"StatusNET":{"Hostname":"pasillo_slave-2708","IPAddress":"192.168.13.160","Gateway":"192.168.13.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"BC:DD:C2:60:EA:94","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
22:21:13 MQT: light/pasillo_slave/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.13.13","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_60EA94","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:21:13 MQT: light/pasillo_slave/STATUS7 = {"StatusTIM":{"UTC":"2021-01-20T21:21:13","Local":"2021-01-20T22:21:13","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:34","Sunset":"17:28"}}
22:21:13 MQT: light/pasillo_slave/STATUS10 = {"StatusSNS":{"Time":"2021-01-20T22:21:13"}}
22:21:13 MQT: light/pasillo_slave/STATUS11 = {"StatusSTS":{"Time":"2021-01-20T22:21:13","Uptime":"0T00:20:47","UptimeSec":1247,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xERICx","BSSId":"80:3F:5D:C1:D5:F9","Channel":13,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Sender:
22:18:47 SRC: WebGui from 192.168.13.3
22:18:47 DGR: Sending Pasillo message to network: seq=9, flags=0, 128=16777217
22:18:47 MQT: light/pasillo_slave/RESULT = {"POWER":"ON"}
22:18:47 MQT: light/pasillo_slave/POWER = ON
22:18:48 HTP: Command
22:18:48 SRC: WebCommand from 192.168.13.13
22:18:48 CMD: Group 0, Index 1, Command "STATUS", Data "0"
22:18:48 MQT: light/pasillo_slave/STATUS = {"Status":{"Module":0,"DeviceName":"Pasillo Slave","FriendlyName":["Pasillo Slave"],"Topic":"pasillo_slave","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:18:48 MQT: light/pasillo_slave/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:48","StartupUTC":"2021-01-21T20:59:00","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2021-01-14T18:39:38","SaveCount":327,"SaveAddress":"F5000"}}
22:18:48 MQT: light/pasillo_slave/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"469/699"}}
22:18:48 MQT: light/pasillo_slave/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xERICx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","2805C8000100060000005A00000000000000","00000040","00006008","00000000"]}}
22:18:48 MQT: light/pasillo_slave/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":0,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
22:18:48 MQT: light/pasillo_slave/STATUS5 = {"StatusNET":{"Hostname":"pasillo_slave-2708","IPAddress":"192.168.13.160","Gateway":"192.168.13.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"BC:DD:C2:60:EA:94","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
22:18:48 MQT: light/pasillo_slave/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.13.13","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_60EA94","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:18:48 MQT: light/pasillo_slave/STATUS7 = {"StatusTIM":{"UTC":"2021-01-21T21:18:48","Local":"2021-01-21T22:18:48","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:33","Sunset":"17:29"}}
22:18:48 MQT: light/pasillo_slave/STATUS10 = {"StatusSNS":{"Time":"2021-01-21T22:18:48"}}
22:18:48 MQT: light/pasillo_slave/STATUS11 = {"StatusSTS":{"Time":"2021-01-21T22:18:48","Uptime":"0T00:19:48","UptimeSec":1188,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"xERICx","BSSId":"94:6A:B0:C8:F1:76","Channel":1,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:09"}}}
22:18:48 CFG: Saved to flash at F4, Count 328, Bytes 4096
22:18:49 WIF: Checking connection...

[... 12 lines cut ...]
[... cut ...]
Switching ON and OFF on each switch shoud change the other one with no dependancy of the BSSID the switch is connected to.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I tried 2 different MQTT servers (Mosquitto package from synology NAS, and MQTT broker on home assistant supervisor) and both perform the same...
I can't see any message log in MQTT servers even I have activated debug log.

**(Please, remember to close the issue when the problem has been addressed)**

pcdiem, 2021-01-22:
Device Groups use multicast UDP packets, not MQTT. As documented, all
devices in a device group must be in the same multicast domain.

Paul C Diem
PCDiem@FoxValley.net

mrbluebrett, 2021-01-22:
Sorry i thought it was MQTT.
So how can i check the multicast domain? 
I have my ISP wifi router, and another AP connected by ethernet cable to the router.
Or how can I force the device to use only one AP even if the signal is weaker, and even if the AP restarts, dont try to connect to the other...

pcdiem, 2021-01-22:
Check the configuration on the router and AP. Any setting that block multicasts should be disabled.

mrbluebrett, 2021-01-23:
I dont have option to change any of these or activate igmp snooping... I’m closing because this might be my network setup and not an issue then...

closed 2021-01-23 by mrbluebrett