arendst/Tasmota #10670: OTA upgrade sequence broken when using a TLS binary - multiple devices bricked

gkvirkvia, 2021-01-23:
### PROBLEM DESCRIPTION
Device[s] get bricked when I try OTA-updating them with a large binary, in my case a MQTT-TLS binary built on Gitpod. The only diff in my workspace is `+#define USE_MQTT_TLS   ` line.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic(two of them), Sonoff S26, Hyleton-311
- [x] Tasmota binary firmware version number used: 9.2
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Esptool

Logging output unavailable because the devices are bricked.

### TO REPRODUCE
1. Erase the chip completely: `$ ./Library/Python/2.7/bin/esptool.py --port=/dev/tty.usbserial-A4Z9PRO1 erase_flash`
2. Flash official, 9.2 tasmota.bin: `$ ./Library/Python/2.7/bin/esptool.py --port=/dev/tty.usbserial-A4Z9PRO1 write_flash 0x0 Downloads/tasmota.bin`
3. Connect the device to local WiFi.
4. Open the Firmware Update page and flash official minimal firmware from OTA URL: `http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz`
5. Build and download the MQTT TLS binary `tasmota-tls.bin.gz` from Gitpod.
6. Once the device reboots to the tasmota-minimal, upload the custom built `tasmota-tls.bin.gz` to it.
7. Witness the "Upload Successful" message. The status LED turns on for a brief period of time and then turns off. From now on the device is bricked, no power cycles, holding down the button for 30 seconds or quickly power-cycling the device 7 times do anything. The status LED stays off, button clicks do nothing.
 

Verified on three different devices, 100% reproducibility on all of them multiple times.

### EXPECTED BEHAVIOUR
A successful update and a non-bricked device.

### ADDITIONAL CONTEXT
 Unzipping the `tasmota-tls.bin.gz` and flashing resulting `tasmota-tls.bin` directly via Esptool **does** work so this isn't a broken binary issue.
If in the step 6 above, instead of using the custom built `tasmota-tls.bin.gz`, I use `tasmota-minimal.bin.gz` or classic `tasmota.bin.gz`, either via OTA URL or via direct upload, the OTA update **does** succeed.

It seems like the culprit might be bigger size of the custom `tasmota-tls.bin.gz`.
In the past, I've been successful using this workflow to upgrade my devices to the custom TLS binary. This seems to be a new issue affecting 9.2 version.

**(Please, remember to close the issue when the problem has been addressed)**

gkvirkvia, 2021-02-22:
Ping. This should be prioritized above other issues. The bug defeats the selling functionality of the firmware - The "OTA" part of TasmOTA is broken :)

jaymemaurice, 2021-02-26:
@gkvirkvia does this also happen when you use the uncompressed binary?

gkvirkvia, 2021-02-26:
I'm unable to squeeze in the uncompressed binary via OTA as it's too big.

I mention in my original report that flashing the uncompressed binary *directly* to the chip *does* work. This is how I have them all up and running nowadays.

jaymemaurice, 2021-02-26:
I have a bunch of esp4 modules coming from ebay - when they get here I'll attempt to repro and root cause.
Maybe we got to the point where the fw is too big to be staged and flashed?
If you create another bricked module before I do, can you use esptool.py to read the flash for binary comparison?

gkvirkvia, 2021-02-26:
Yeah totally, I do have another bricked Sonoff Basic that I'll have to FTDI to unbrick, I'll try to do what you're asking over the weekend.

patrickdk77, 2021-02-27:
Not sure exactly how your bricking them. I have 60 devices all running tls, some since 7.x and others just flashed this week and have never had a single issue.
But just enabling tls is not a good thing, and the docs state several times that tls uses a lot of memory and you need to disable other things (enabling tls automatically disables some of the web gui stuff to make room), so you don't run out of ram/stack space. All my flash updates have been via ota using compressed binaries since 8.2 release. My initial flashes are normally using tuya convert with a tls enabled tasmota binary, or direct flash upload using esptool. I don't see the larger size of a tasmota-tls.bin being an issue, as I'm running some huge.
All my devices have been upgraded to 9.2.0.2 9.2.0.4 and 9.2.0.7 and I just upgraded them all to 9.3.1.1

586256	tasmota.bin
419227	tasmota.bin.gz
386464	tasmota-minimal.bin
276576	tasmota-minimal.bin.gz
621408	tasmota-tls.bin
444838	tasmota-tls.bin.gz
483600	tasmota-tls-convert.bin
347270	tasmota-tls-convert.bin.gz
626672	tasmota-tls-ir.bin
434822	tasmota-tls-ir.bin.gz
556176	tasmota-tls-lite.bin
398150	tasmota-tls-lite.bin.gz
673168	tasmota-tls-sensors.bin
476729	tasmota-tls-sensors.bin.gz

I disable these on all my builds since I don't use them:
#undef USE_DOMOTICZ
#undef USE_KNX_WEB_MENU
#undef USE_EMULATION_HUE                      // Enable Hue Bridge emulation for Alexa (+14k code, +2k mem common)
#undef USE_EMULATION_WEMO                     // Enable Belkin WeMo emulation for Alexa (+6k code, +2k mem common)
#undef USE_DISCOVERY                            // Enable mDNS for the following services (+8k code or +23.5k code with core 2_5_x, +0.3k mem)
#undef WEBSERVER_ADVERTISE                    // Provide access to webserver by name <Hostname>.local/
#undef MQTT_HOST_DISCOVERY                    // Find MQTT host server (overrides MQTT_HOST if found)
#undef USE_ARDUINO_OTA                          // Add optional support for Arduino OTA (+13k code)

ascillato2, 2021-03-02:
Please, when you have time, tell us if you could solve your issue as suggested. Thanks.

closed 2021-03-02 by ascillato2

rapi3, 2021-03-23:
I just had a similar failure with few devices: tuya smart pulgs, sonoff rf bridge and wemos d1 and for me I think the cause was my long custom name inside the gz archive because I compile my own variant for all devices and disable all I can and I don't use. After I keep the firmware name short all it is good with the same firmware compiled with USE_MQTT_TLS that bricked the devices.

gkvirkvia, 2021-03-23:
> Not sure exactly how your bricking them. I have 60 devices all running tls, some since 7.x and others just flashed this week and have never had a single issue.
> But just enabling tls is not a good thing, and the docs state several times that tls uses a lot of memory and you need to disable other things (enabling tls automatically disables some of the web gui stuff to make room), so you don't run out of ram/stack space. All my flash updates have been via ota using compressed binaries since 8.2 release. My initial flashes are normally using tuya convert with a tls enabled tasmota binary, or direct flash upload using esptool. I don't see the larger size of a tasmota-tls.bin being an issue, as I'm running some huge.
> All my devices have been upgraded to 9.2.0.2 9.2.0.4 and 9.2.0.7 and I just upgraded them all to 9.3.1.1
> 
> 586256 tasmota.bin
> 419227 tasmota.bin.gz
> 386464 tasmota-minimal.bin
> 276576 tasmota-minimal.bin.gz
> 621408 tasmota-tls.bin
> 444838 tasmota-tls.bin.gz
> 483600 tasmota-tls-convert.bin
> 347270 tasmota-tls-convert.bin.gz
> 626672 tasmota-tls-ir.bin
> 434822 tasmota-tls-ir.bin.gz
> 556176 tasmota-tls-lite.bin
> 398150 tasmota-tls-lite.bin.gz
> 673168 tasmota-tls-sensors.bin
> 476729 tasmota-tls-sensors.bin.gz
> 
> I disable these on all my builds since I don't use them:
> #undef USE_DOMOTICZ
> #undef USE_KNX_WEB_MENU
> #undef USE_EMULATION_HUE // Enable Hue Bridge emulation for Alexa (+14k code, +2k mem common)
> #undef USE_EMULATION_WEMO // Enable Belkin WeMo emulation for Alexa (+6k code, +2k mem common)
> #undef USE_DISCOVERY // Enable mDNS for the following services (+8k code or +23.5k code with core 2_5_x, +0.3k mem)
> #undef WEBSERVER_ADVERTISE // Provide access to webserver by name .local/
> #undef MQTT_HOST_DISCOVERY // Find MQTT host server (overrides MQTT_HOST if found)
> #undef USE_ARDUINO_OTA // Add optional support for Arduino OTA (+13k code)

You might be onto something! So far I've been ignoring the memory warnings in the source code because I **do** use mDNS for MQTT discovery. This hasn't been an issue prior but it makes sense that as the feature code grows, this might be the issue that I'm bumping into. I'll try to use your config to see if it fixes my problem.

In the meantime though, it seems like it's too easy to shoot yourself in the foot and end up with a bricked device due to compiling too many memory hungry features into the binary. I wonder if it's possible to come up with a tool that checks the memory consumption of a binary.
There seems to be an esp8266 simulator out there - https://github.com/OSLL/qemu-xtensa/tree/xtensa-esp8266
We could potentially make a tool that checks if a firmware does indeed run out of memory during the boot sequence - (x simulated seconds or something like that).

rapi3, 2021-03-23:
**It is a problem with length name of archive + file inside archive, probably similar with password to long problem.**
I just made another test with this wemos d1:
Program Version 9.3.1.2(tasmota)
Build Date & Time 2021-03-23T12:18:11
Core/SDK Version 2_7_4_9/2.2.2-dev(38a443e)
...
MQTT Port 8883
MQTT TLSEnabled
...
Flash Size  4096 kB
Program Flash Size 4096 kB
**Program Size 528 kB**
Free Program Space 2540 kB
**Free Memory 21.7 kB**

step to reproduce:
1 I renamed firmware to: **tasmota9.3.1.2-wemosd1-ping-temp-displayi2c-v7a-160mhz.bin**
2 create archive: **tasmota9.3.1.2-wemosd1-ping-temp-displayi2c-v7a-160mhz.tar.gz**
**_3 flash OTA -> succes and device bricked no boot_**

Fixed with tasmotizer flashed: tasmota9.3.1.2-wemosd1-ping-temp-displayi2c-v7a-160mhz.bin
wemos booted ok
1 renamed to flash.bin
2 create archive: flash.gz
3 flash OTA - > succes 100%

**It is a problem with length name of archive + file inside archive, probably similar with password to long problem.**
I did not read the flash to see what was written... but the config was there and intact because I flash with tasmotizer without erase and my rules and setting are good.

patrickdk77, 2021-03-23:
So the problem is not that the name is too long, and the problem isn't due to free memory issue, and the deivces are not bricked at all.

The only thing I have seen so far, after we confirmed there is enough memory, is for some reason your packing the bin file into a tar archive? I'm not sure why, tar is not supported.

If you flashed your device with a .tar file, yes it will not boot, is it bricked? no. Can you OTA flash it? no.

Don't tar your flash image, it is not a supported format for the esp8266

gkvirkvia, 2021-03-23:
Hmm I forgot the details of my bugreport. The memory doesn't seem to be the issue as flashing the resulting binary directly with FTDI solves the issue.

Re: comment above, @ patrickdk77 
that doesn't sound consistent with what I've experienced.
I'm 100% sure I haven't packed anything myself, I used `platformio run -e tasmota` in Gitpod to build the .gz artifact I was flashing. Also, I'm 90% sure that I've actually extracted the `.gz` file and flashed it directly with FTDI, after which the devices worked fine.

patrickdk77, 2021-03-23:
As for your origional issue, I cannot really comment at all. We don't know how big the bin files and gz files your attempting to flash are, expecially since they are custom builds. If it's over 600kb it gets hairy if it's too large, and then it's just not supported, unless you use the newer 2m or 4m images, assuming your device has enough flash space.

But rapi3 hijacked your issue and that is what I'm commenting on. in his step 2 to reproduce he used a .tar.gz and not a .bin.gz file. and tar files are unsupported.

It would have been preferred if people would open their own bugs for unrelated issues.

gkvirkvia, 2021-03-23:
I've put exact replication steps for building the exact firmware I'm messing with. It takes 5 minutes tops to obtain the artifacts if you follow the instructions. I'm happy to build them myself and attach them here if necessary.

> If it's over 600kb it gets hairy if it's too large.
What do you mean exactly by "hairy" ? There's a bunch of file size checks around https://github.com/arendst/Tasmota/blob/a127771aba6aa0e5c64ad36223933ca4c3a3103b/tasmota/xdrv_01_webserver.ino#L2496

If the file is too big, the error message *musht* be clear on that and the result shouldn't be a bricked device.

patrickdk77, 2021-03-23:
I don't understand how the device is bricked. You even say you unbricked it, so it wasn't bricked in the first place.
Bricked is a device that cannot be fixed. It cannot be flashed again with working firmware, it will not boot again, ever, not matter what is done to it.

These are just sounding like failed flash, why I'm not sure. I haven't had a single issue with any of my tls builds being flashed ota. Some of my devices do take >10min to flash though, if you powercycle it, start pushing buttons and stuff, before it's done, yes, you will have a failed flash. Why do some of my devices flash so slow? I'm not sure, but even directly soldered up, it's slow, so it's something to do with the flash chip itself being slow where the others that came in the same box did not have this issue. Other than being slow to flash it gives me no issues.

But I cannot reproduce the issues you had. I do not have any of the devices you said failed for you, no sonoff devices here.
I have a bunch of treatlife, mj, gosund, d1mini, esp-01s, esp-m3, esp-m3. I have not had to serial flash a single one of them due to an ota issue, but any issues that happened where recovered from another ota flash (normally bad wifi signal)

Your origional issue was you believed the larger bin size wasn't flashable over ota when using gzip. The stock tasmota + tls is smaller than some builds I use, and have no issues, so wanted to just point out the idea of looking that that route is likely not the correct source of the problem.

Stock tasmota with tls enabled
621408 tasmota-tls.bin
444838 tasmota-tls.bin.gz

My custom builds with tls, where ir is slightly smaller, and sensors is much larger
626672 tasmota-tls-ir.bin
434822 tasmota-tls-ir.bin.gz
673168 tasmota-tls-sensors.bin
476729 tasmota-tls-sensors.bin.gz

So I'm not saying you didn't have an issue, but the suggestion that the larger size is causing the problem and it to be *bricked* is unlikely to be the issue.

sschirr, 2021-03-30:
I think i have a similar issue. I tried to upgrade the following device to 9.3.1:
[https://templates.blakadder.com/onestyle_SD-WL-02.html](url)

I build the firmware in gitpod with mqtt tls enabled:

> #ifndef USE_MQTT_TLS
#define USE_MQTT_TLS
  #define USE_MQTT_TLS_CA_CERT
#endif
#define MQTT_PORT 8883

I flashed the firmware as described here except i did use the compressed firmware:
[https://tasmota.github.io/docs/Gitpod/#ota-flash](url)

Flashing `tasmota-minimal.bin` went fine. After i flashed `tasmota.bin.gz` the device reported success but did not come back online.
I did not touch the device for at least half an hour. This happened to both of my devices.

A previous update to an older tasmota version also with mqtt tls enabled was successful.

Do you think there is any chance to get the device back online without opening the device? As far as i can tell it is impossible to open the device without destroying the plastic case.

gkvirkvia, 2021-06-03:
Why did this get closed, especially with others reproducing the issue?

rpcai, 2026-02-04:
Bump. @giorgi1324  I've just encountered the same issue. Some of the responses you've gotten seem unhelpful and combative. 

- Compile firmware, try to OTA upload from minimal `tasmota.bin` -> Not enough space
- Notice that all of the firmwares on the OTA update server are ...bin.gz
- Being a noob try `tar -zcvf tasmota.bin tasmota.bin.gz`
- OTA upgrade with tasmota.bin.gz appears to work but then bricks device. _GET WRECKED NOOB_
- Re-Flash via serial, find this discussion, realize that tar should not have been used.
- `gzip -k tasmota.bin` 
- OTA upgrade with this tasmota.bin.gz (not a tarball) -> Works. 

I blindly used claude code to compile my firmware, so there may still be a bug in the workflow that's producing your .gz - However an update to documentation, for noobs like me 'DONT TAR THE .bin,' could be usefull....